AllskyTeam/allsky #2610: Module manager periotic jobs unchecks.

BobW55, 2023-05-06:
I have my main router set to reeboot every night at 3:00AM.
this is the second day in a row I have woken up and found my overlay information on the live screen not updating.  Both time I have found the RUN PERIOTIC JOB un-checked.
Both times I have had the live vew open in the WEBUI, although my computer was asleep.
My Allsky is POE powered so it is hard wired to my home network.   Both the POE switch and the router the POE is hooked up to DO NOT reboot.
[image]

BobW55, 2023-05-06:
Will try again tonight, but close the WEBUI before I go to bed

BobW55, 2023-05-07:
OK, last night I closed the active WEBUI window.   This morning I opened the WEBUI and the it was happlily imaging away overlays and all.  Something with keeping the WEBUI window open while the router reboots causes it to uncheck???

Alex-developer, 2023-05-08:
I suspect the custom script is taking too long to run, what watchdog timeout do you have set in the module settings?

Paste the contents of the periodic flow here as that has some debug info in it that may help

BobW55, 2023-05-08:
just incase, here is the script it runs:  remove the .txt
[getweather.sh.txt](https://github.com/thomasjacquin/allsky/files/11417147/getweather.sh.txt)

Alex-developer, 2023-05-08:
> just incase, here is the script it runs: remove the .txt [getweather.sh.txt](https://github.com/thomasjacquin/allsky/files/11417147/getweather.sh.txt)

Thanks. I need the json file when its been disabled, in the one you posted its enabled. Sorry my fault I should have explained that

BobW55, 2023-05-09:
OK.. will do that next time it fails.   Left it running last night and it stayed enabled.   I haver also power cycled my laptop, not sure if that didn't have an infulence on it as well.

BobW55, 2023-05-20:
**PLEASE CLOSE** 
Believe the issue was caused by leaving the live view active at night.  The reboot of the main router causing issues.
The Allsky is plugged into a hard wired 2nd router that does NOT reboot at night, only my main router does.
Been having issues with the main router dropping or not assigning IP addresses.

EricClaeys, 2023-05-20:
Closing per user request.