AllskyTeam/allsky #2615: [BUG]

JosWennmacker, 2023-05-07:
### Environment

* Camera:  ZWO  ASI178MC
* OS: Ubuntu 22.04
* Allsky version: v2023.05.01
* Pi model and amount of memory (512 MB, 1, 2, 4, or 8 GB) Pi 4B 4GB

### Bug Description

* Error during install: pip3 on Ubuntu does not support --build option
* Can it be reproduced?  yes, by running a new install.
* Did anything change?  No
* Please include the exact messages or a screenshot.
allsky@allsky:~/allsky$ ./install.sh
****************************************
*** Welcome to the Allsky Installer! ***
****************************************

* Using prior ZWO camera.
* Size of current swap (1 MB) is sufficient; no change needed.
* /home/allsky/allsky/tmp is currently in memory; no change needed.

The following steps can take up to 1 - 3 HOURS depending on the speed of your Pi
and how many of the necessary dependencies are already installed.
* Installing the web server.
* Installing dependencies.
* Preparing Allsky commands.
* Modifying locations for WebUI.
* Setting up WebUI options for ZWO cameras.
* Keeping 'C.UTF-8' locale.
* Setting permissions on /var/log/allsky.log and /var/log/allskyperiodic.log.
* Installing PHP Modules.
* Installing other PHP dependencies.
* Installing Python dependencies.
        This may take a LONG time if the packages are not already installed.
*    === Package # 1 of 15: [opencv_python>=4.5.3.56]

*** ERROR:
**********
Python dependency [opencv_python>=4.5.3.56] failed
**********

The full log file is in /home/allsky/allsky/config/installation_logs/Python_dependencies.1.log
The end of the file is:
  pip3 install [options] [-e] <vcs project url> ...
  pip3 install [options] [-e] <local project path> ...
  pip3 install [options] <archive url/path> ...

no such option: --build

The Python_dependencies.1.log contains:

Usage:   
  pip3 install [options] <requirement specifier> [package-index-options] ...
  pip3 install [options] -r <requirements file> [package-index-options] ...
  pip3 install [options] [-e] <vcs project url> ...
  pip3 install [options] [-e] <local project path> ...
  pip3 install [options] <archive url/path> ...

no such option: --build

### Log / configuration files

[Python_dependencies.1.log](https://github.com/thomasjacquin/allsky/files/11415265/Python_dependencies.1.log)

Alex-developer, 2023-05-08:
Thanks for the report. I will take a look at the issue.

If you are comfortable editing the install script then change this line

pip3 install --no-warn-script-location --build "${PIP3_BUILD}" -r /tmp/package > "${L}" 2>&1

To

pip3 install --no-warn-script-location -r /tmp/package > "${L}" 2>&1

And see if that helps

JosWennmacker, 2023-05-09:
Yes, it helps. All (15) pip3 actions succeed now.
But I ran immediately into another problem. The install of the true type fonts freezes the Pi. When I install the fonts manually I have to agree to an EULA. I guess that the display of the EULA breaks the script. 
I solved it by running the apt-get cpmmand manually and comenting it out in install.sh

Alex-developer, 2023-05-09:
> Yes, it helps. All (15) pip3 actions succeed now. But I ran immediately into another problem. The install of the true type fonts freezes the Pi. When I install the fonts manually I have to agree to [...]

Thanks for that. I'll to a full test install on Ubuntu this weekend and fix any issues

Alex

candrakhunter, 2023-05-11:
So after this bugfix the installation should run properly up to package 15?

EricClaeys, 2023-05-12:
@candrakhunter   All 15 pip3 packages installed for @JosWennmacker so they should for you as well.  You may run into the same issue he did with font installation.
@Alex-developer  Will test this weekend with Ubuntu.  All the developers and testers have Pi OS so we never encountered this problem.

EricClaeys, 2023-06-03:
@JosWennmacker,
@Alex-developer looked into what it would take to support Ubuntu, and it's more effort than we can commit to right now, especially given the limited audience.
Hence, we have updated the documentation and installation script to stae only Pi OS from Raspberry Pi is supported.  If you are able to get Allsky working on your Ubuntu Pi, please let us know as we would be interested in incorporating your changes into Allsky.

closed 2023-06-03 by EricClaeys

JosWennmacker, 2023-06-04:
Thank you , @EricClaeys and @Alex-developer .  I already saw some more problems. I don't feel confident enough to try and get Allsky working om my Ununtu. Moreover I don't have a spare Pi left. The last one got into production for home automation a few weeks ago. I switched already to bullseye a few days ago.