AllskyTeam/allsky #2620: Unable to change config, unable to add latitude and longitude coordinates

wj7106, 2023-05-09:
* Camera:   RPi - include the model
* OS: raspberry 4b 4GB Bullseye
* Allsky version:
 
 Editor

Failed to save '/home/pi/allsky/config/config.sh': Unable to update settings: We trust you have received the usual lecture from the local System Administrator. It usually boils down to these three things: #1) Respect the privacy of others. #2) Think before you type. #3) With great power comes great responsibility. sudo: a terminal is required to read the password; either use the -S option to read from standard input or configure an askpass helper sudo: a password is required

wj7106, 2023-05-09:
> @wj7106 which version of Allsky? Did you reboot after installing Allsky? Please run `id` and post the results.
v2023.05.01， After installation, I have restarted several times and it has always been like this. I had to manually give the file the highest permission to save longitude and latitude data, but I still couldn't enable it  
I have to reinstall back to the old version

EricClaeys, 2023-05-22:
@wj7106    Any update?

Stewaro, 2023-05-29:
I have the same problem.

Stewaro, 2023-05-29:
i@allsky:~/allsky $ id pi
uid=1000(pi) gid=1000(pi) Gruppen=1000(pi),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),104(input),106(render),108(netdev),999(spi),998(i2c),997(gpio),117(lpadmin)

pi@allsky:~/allsky $ id   
uid=1000(pi) gid=1000(pi) Gruppen=1000(pi),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),104(input),106(render),108(netdev),117(lpadmin),997(gpio),998(i2c),999(spi)

EricClaeys, 2023-05-29:
@Stewaro   Henry,
which version of Allsky? Did you reboot after installing Allsky? Please run the following and post the results:
~~~
ls -l /etc/sudoers.d/allsky
~~~
Then ATTACH (do not copy/paste) `/etc/sudoers.d/allsky`.

I have also seen this problem in the past but don't remember what I did to fix it.

Thanks - Eric

Stewaro, 2023-05-29:
pi@allsky:~ $ ls -l /etc/sudoers.d/allsky
ls: Zugriff auf '/etc/sudoers.d/allsky' nicht möglich: Datei oder Verzeichnis nicht gefunden

file is not found in the directory?!

Stewaro, 2023-05-29:
pi@allsky:~ $ ls -l /etc/sudoers.d
insgesamt 16
-r--r----- 1 root root  36 29. Apr 2019  010_at-export
-r--r----- 1 root root  27  3. Mai 02:09 010_pi-nopasswd
-r--r----- 1 root root 211 18. Mär 2022  010_proxy
-r--r----- 1 root root 958 14. Jan 14:29 README
pi@allsky:~ $

Stewaro, 2023-05-29:
Version: v2023.05.01 
yes, reboot several times

EricClaeys, 2023-05-29:
@Stewaro. Henry,
Would you please ATTACH `~/allsky/config/installation_logs/install.sh.log` ?
/etc/sudoers.d/allsky should exist.

Also, please run `sudo date`.  It will tell us if `sudo` is working.

Stewaro, 2023-05-30:
pi@allsky:~ $ sudo date
Di 30. Mai 15:55:38 CEST 2023
pi@allsky:~ $ ls -l
insgesamt 40
drwxr-xr-x 15 pi pi 4096 29. Mai 15:57 allsky
drwxr-xr-x  2 pi pi 4096  3. Mai 02:16 Bookshelf
drwxr-xr-x  2 pi pi 4096 29. Mai 15:34 Desktop
drwxr-xr-x  2 pi pi 4096 29. Mai 15:35 Documents
drwxr-xr-x  2 pi pi 4096 29. Mai 15:35 Downloads
drwxr-xr-x  2 pi pi 4096 29. Mai 15:35 Music
drwxr-xr-x  2 pi pi 4096 29. Mai 15:35 Pictures
drwxr-xr-x  2 pi pi 4096 29. Mai 15:35 Public
drwxr-xr-x  2 pi pi 4096 29. Mai 15:35 Templates
drwxr-xr-x  2 pi pi 4096 29. Mai 15:35 Videos
pi@allsky:~ $ cd allsky
pi@allsky:~/allsky $ ls -l
insgesamt 176
-rwxr-xr-x 1 pi pi       11773 29. Mai 15:47 allsky.sh
drwxr-xr-x 2 pi pi        4096 29. Mai 15:47 assets
drwxr-xr-x 2 pi pi        4096 29. Mai 20:03 bin
drwxr-xr-x 2 pi pi        4096 30. Mai 15:52 config
drwxr-xr-x 4 pi pi        4096 29. Mai 20:03 config_repo
drwxr-xr-x 9 pi pi        4096 29. Mai 15:47 html
-rwxr-xr-x 1 pi pi       78807 29. Mai 15:47 install.sh
-rw-r--r-- 1 pi pi        1070 29. Mai 15:47 LICENSE
-rw-r--r-- 1 pi pi        1246 29. Mai 15:47 Makefile
drwxr-xr-x 2 pi pi        4096 29. Mai 15:47 notification_images
drwxr-xr-x 2 pi pi        4096 29. Mai 15:57 pip3.build
-rw-r--r-- 1 pi pi        8697 29. Mai 15:47 README.md
drwxr-xr-x 3 pi pi        4096 29. Mai 15:47 scripts
drwxr-xr-x 5 pi pi        4096 29. Mai 15:56 src
drwxrwxr-x 2 pi www-data   160 30. Mai 15:52 tmp
-rwxr-xr-x 1 pi pi        1256 29. Mai 15:47 uninstall.sh
-rwxr-xr-x 1 pi pi        6989 29. Mai 15:47 upgrade.sh
-rw-r--r-- 1 pi pi        5796 29. Mai 15:47 variables.sh
-rw-r--r-- 1 pi pi          12 29. Mai 15:47 version
drwxr-xr-x 2 pi pi        4096 29. Mai 15:47 website
pi@allsky:~/allsky $ cd config
pi@allsky:~/allsky/config $ ls -l
insgesamt 60
-rw-r--r-- 2 pi pi  4087 29. Mai 20:03 cc.json
-rw-r--r-- 2 pi pi  4087 29. Mai 20:03 cc_ZWO_ASI178MC.json
-rw-r--r-- 1 pi pi  5856 29. Mai 20:03 config.sh
-rw-r--r-- 1 pi pi  2008 29. Mai 15:56 ftp-settings.sh
-rw-r--r-- 1 pi pi   370 30. Mai 15:52 messages.txt
-rw-r--r-- 1 pi pi 26906 29. Mai 20:03 options.json
-rw-r--r-- 1 pi pi  1988 29. Mai 20:03 settings.json
-rw-r--r-- 1 pi pi  1967 29. Mai 20:03 settings_ZWO_ASI178MC.json
pi@allsky:~/allsky/config $ 

file is not found in the directory?!

EricClaeys, 2023-05-30:
@Stewaro, You may have deleted that directory after seeing the message to do so in the WebUI.

I suggest doing g the following:
~~~
cd ~/allsky/config
rm settings_ZWO_ASI178MC.json
ln settings.json settings_ZWO_ASI178MC.json
cd
sudo systemctl stop allsky 
mv allsky allsky-OLD
~~~
Then run the installation again, starting with the "git clone ..." command.  If it fails, attach `~/allsky/config/installation_logs/install.sh.log`.

Stewaro, 2023-05-31:
Then ATTACH (do not copy/paste) /etc/sudoers.d/allsky

www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli scan
www-data ALL=(ALL) NOPASSWD:/sbin/wpa_cli reconfigure
www-data ALL=(ALL) NOPASSWD:/bin/cp /tmp/hostapddata /etc/hostapd/hostapd.conf
www-data ALL=(ALL) NOPASSWD:/etc/init.d/hostapd start
www-data ALL=(ALL) NOPASSWD:/etc/init.d/hostapd stop
www-data ALL=(ALL) NOPASSWD:/etc/init.d/dnsmasq start
www-data ALL=(ALL) NOPASSWD:/etc/init.d/dnsmasq stop
www-data ALL=(ALL) NOPASSWD:/bin/cp /tmp/dhcpddata /etc/dnsmasq.conf
www-data ALL=(ALL) NOPASSWD:/sbin/shutdown -h now
www-data ALL=(ALL) NOPASSWD:/sbin/reboot
www-data ALL=(ALL) NOPASSWD:/bin/systemctl
www-data ALL=(ALL) NOPASSWD:/bin/chgrp
www-data ALL=(ALL) NOPASSWD:/bin/chown
www-data ALL=(ALL) NOPASSWD:/bin/chmod
www-data ALL=(ALL) NOPASSWD:/bin/rm
www-data ALL=(ALL) NOPASSWD:/bin/cp
www-data ALL=(ALL) NOPASSWD:/bin/mv
www-data ALL=(ALL) NOPASSWD:/usr/bin/vcgencmd
www-data ALL=(ALL) NOPASSWD:/home/pi/allsky/scripts/postData.sh
www-data ALL=(ALL) NOPASSWD:/usr/sbin/ifconfig
www-data ALL=(ALL) NOPASSWD:/usr/bin/truncate
www-data ALL=(ALL) NOPASSWD:/home/pi/allsky/scripts/upload.sh
www-data ALL=(ALL) NOPASSWD:/home/pi/allsky/scripts/makeChanges.sh
www-data ALL=(ALL) NOPASSWD:/usr/bin/date

Stewaro, 2023-05-31:
Then run the installation again, starting with the "git clone ..." command. If it fails, attach ~/allsky/config/installation_logs/install.sh.log

STARTING INSTALLATON AT Di 30. Mai 18:24:17 CEST 2023.

PRIOR_ALLSKY_VERSION=v2023.05.01
PRIOR_CAMERA_TYPE=ZWO
PRIOR_SETTINGS_FILE=/home/pi/allsky-OLD/config/settings.json
Using the 'master' branch.
  DEBUG: TEMP_LOCALE=LANG=de_DE
LANGUAGE=de_DE
LC_ALL=de_DE, CURRENT_LOCALE=de_DE
Using CAMERA_TYPE 'ZWO' from prior config.sh
  DEBUG: RAM_SIZE=4096, SUGGESTED_SWAP_SIZE=0.
* Size of current swap (100 MB) is sufficient; no change needed.
* /home/pi/allsky/tmp is currently in memory; no change needed.
  DEBUG: Existing IMAGES=/home/pi/allsky/tmp/image.jpg
* Installing the web server.
* Installing dependencies.
* Preparing Allsky commands.
* Modifying locations for WebUI.
* Setting up WebUI options for ZWO cameras.
* Restoring settings files:
* 	'settings_ZWO_ASI178MC.json'
Executing makeChanges.sh   --cameraTypeOnly   'cameraType' 'Camera Type' 'ZWO' 'ZWO'

Settings files:
-rw-r--r-- 2 pi pi 1988 29. Mai 20:03 /home/pi/allsky/config/settings.json
-rw-r--r-- 2 pi pi 1988 29. Mai 20:03 /home/pi/allsky/config/settings_ZWO_ASI178MC.json
* Keeping 'de_DE' locale.
Info: Settings file '/home/pi/allsky/config/settings.json' contained .locale = 'de_DE'.
* Setting permissions on /var/log/allsky.log and /var/log/allskyperiodic.log.
* Installing PHP Modules.
* Installing other PHP dependencies.
* Installing Python dependencies.
	This may take a LONG time if the packages are not already installed.
*    === Package # 1 of 15: [opencv_python>=4.5.3.56]
*    === Package # 2 of 15: [Pillow]
*    === Package # 3 of 15: [pyephem]
*    === Package # 4 of 15: [skyfield]
*    === Package # 5 of 15: [astral]
*    === Package # 6 of 15: [pytz]
*    === Package # 7 of 15: [scipy]
*    === Package # 8 of 15: [scikit-image]
*    === Package # 9 of 15: [paho-mqtt]
*    === Package # 10 of 15: [numpy>=1.23.3]
*    === Package # 11 of 15: [photutils]
*    === Package # 12 of 15: [astropy]
*    === Package # 13 of 15: [suncalc]
*    === Package # 14 of 15: [Adafruit-Blinka]
*    === Package # 15 of 15: [vcgencmd]
* Installing Trutype fonts.
* Setting up modules and overlays.
* Copying '/home/pi/allsky/config/overlay/config/overlay-ZWO.json' to '/home/pi/allsky/config/overlay/config/overlay.json'.
* No prior 'endOfNight_additionalSteps.sh' so can't restore.
No prior 'images' directory so can't restore; This unusual.
* No prior 'darks' directory so can't restore.
* No prior 'modules' directory so can't restore.
* No prior 'overlay' directory so can't restore.
* No prior 'extra' directory so can't restore.
* No prior 'WebUI security settings' so can't restore.
No prior remote Website configuration so can't restore.
  DEBUG: CONFIG_SH_VERSION=1, PRIOR=1
* Restoring prior 'config.sh' file.
* Updating ALLSKY_VERSION in 'config.sh' to 'v2023.05.01'.
  DEBUG: FTP_SH_VERSION=1, PRIOR=1
* Restoring prior 'ftp-settings.sh' file.
* Updating '/home/pi/allsky/config/config.sh'
* Setting permissions on web-related files.
* Adding pi to www-data group.
* Creating/updating sudoers file.
*** NOTICE: 
**********
The old WebUI location '/var/www/html' exists and is empty.
You can safely delete it after installation:  sudo rmdir '/var/www/html'
**********

Pi will reboot after installation completes.
After you've configured Allsky, run:
   cd ~/allsky;  scripts/check_allsky.sh
to check for any issues.  You can also run it whenever you make changes.

ENDING INSTALLATON AT Di 30. Mai 22:47:54 CEST 2023.

Stewaro, 2023-05-31:
After resetting the Allsky Settings to default same problem.
*** ERROR: Latitude not specified! (May 31, 04:25:32 )
*** ERROR: Longitude not specified! (May 31, 04:25:32 )
can not write

EricClaeys, 2023-05-31:
Closing this Issue.  Original poster never responded.

@Stewaro   Henry, if you have additional issues after installing the point release when it's out, please create a new Issue.

closed 2023-05-31 by EricClaeys