AllskyTeam/allsky #2621: Cannot enable overlay

ea1ii, 2023-05-10:
**Overlay Method** is set to **_'module'_**, that saves as `"overlayMethod": "1"` in `allsky/config`.

Overlays is Enabled in Module manager both for day and night capture.

But...

1.  All variables Tab in Overlay manager warns about enabling Overlay module (that it is) and Enabling debug mode within it (that I cannot find)
2.  Does not render the overlay, just the legacy mode
3.  Debug info in module manager shows allskyoverlay.py as 0.00 Run Time and External OVerlay Disabled as Result.

EricClaeys, 2023-05-10:
@ea1ii. Try stopping Allsky, then `sudo truncate -s 0 /var/log/allsky.log` then set Debug Level to 4 in the WebUI. After a couple pictures ATTACH var/log/allsky.log and I will take a look.

ea1ii, 2023-05-10:
This line brought my attention:

`May 10 22:26:39 CGallsky allsky[14058]: ERROR: unsupported operand type(s) for -: 'str' and 'int'`

There was a change: it now renders nothing, while before restarting it was rendering as per the legacy config.

The whole log:

```
May 10 22:25:54 CGallsky allsky[13656]:      ***** Starting AllSky *****
May 10 22:25:55 CGallsky allsky[13714]: copy_notification_image.sh: Uploading StartingUp.jpg
May 10 22:25:56 CGallsky allsky[13757]: Upload output from '/home/pi/allsky/tmp/image.jpg:
May 10 22:25:56 CGallsky allsky[13757]:    quote: PWD: Operation not supported
May 10 22:26:03 CGallsky allsky[13967]:  >>> Parameter [config                   ]  Value: [/home/pi/allsky/tmp/capture_args.txt]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [debuglevel               ]  Value: [4]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [aggression               ]  Value: [85]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [alwaysshowadvanced       ]  Value: [1]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [angle                    ]  Value: [-8]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [autousb                  ]  Value: [1]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [camera                   ]  Value: [ZWO ASI224MC]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [cameramodel              ]  Value: [ASI224MC]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [cameranumber             ]  Value: [0]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [cameratype               ]  Value: [ZWO]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [computer                 ]  Value: [Raspberry Pi 4B, 8Gb]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [config                   ]  Value: [[none]]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [consistentdelays         ]  Value: [1]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [dayautoexposure          ]  Value: [1]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [dayautogain              ]  Value: [1]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [dayawb                   ]  Value: [0]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [daybin                   ]  Value: [1]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [daybrightness            ]  Value: [110]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [daydelay                 ]  Value: [10000]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [dayexposure              ]  Value: [1000]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [daygain                  ]  Value: [100]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [daymaxautoexposure       ]  Value: [1000]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [daymaxautogain           ]  Value: [300]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [dayskipframes            ]  Value: [5]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [daywbb                   ]  Value: [95]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [daywbr                   ]  Value: [52]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [debuglevel               ]  Value: [4]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [displaysettings          ]  Value: [1]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [extratext                ]  Value: []
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [extratextage             ]  Value: [0]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [filename                 ]  Value: [image.jpg]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [flip                     ]  Value: [0]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [fontcolor                ]  Value: [255 0 0]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [fontline                 ]  Value: [1]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [fontname                 ]  Value: [0]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [fontsize                 ]  Value: [7]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [fonttype                 ]  Value: [0]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [gaintransitiontime       ]  Value: [5]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [height                   ]  Value: [0]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [histogrambox             ]  Value: [500 500 50 50]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [imageurl                 ]  Value: [https://cgastrophoto.co.uk/CG_allsky/image.jpg]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [lastchanged              ]  Value: [2023-05-10 12:12:48]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [latitude                 ]  Value: [50.928333N]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [lens                     ]  Value: [CCTVLens 2.1mm]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [locale                   ]  Value: [en_GB.UTF-8]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [location                 ]  Value: [Southampton, Hants]
May 10 22:26:03 CGallsky allsky[13967]:      >>> Parameter [longitude                ]  Value: [1.3513888W]
[... cut ...]
May 10 22:27:49 CGallsky allsky[13967]: STARTING EXPOSURE at: 2023-05-10 22:27:49   @ 29.6 sec
May 10 22:27:49 CGallsky allsky[13967]:     > Camera set auto-exposure to 29.6 sec
May 10 22:27:49 CGallsky allsky[14377]: upload.sh: FTP '/home/pi/allsky/tmp/image-20230510222717.jpg' finished
May 10 22:27:49 CGallsky allsky[14377]: Upload output from '/home/pi/allsky/tmp/image-20230510222717.jpg:
May 10 22:27:49 CGallsky allsky[14377]:    quote: PWD: Operation not supported
May 10 22:27:49 CGallsky allsky[14377]: upload.sh: Also copying to /home/pi/allsky/html/allsky/image.jpg
```

EricClaeys, 2023-05-10:
@ea1ii.  Please ATTACH your allsky/config/overlay/config/overlay.json (I think that’s where it is).
Going forward, if there are a lot of lines in a file, like the log file, please ATTACH the file instead of copy/paste.  Thanks.

ea1ii, 2023-05-10:
```
{
    "fields": [
        {
            "label": "${DATE}  ${TIME}",
            "x": 110,
            "y": 16,
            "id": "oe-field-0",
            "format": "%d-%m-%Y,%H:%M:%S",
            "strokewidth": 1,
            "sample": "DATE,TIME",
            "tlx": 0,
            "tly": 0,
            "fontsize": 32,
            "fill": "#ff0000",
            "font": "ds-digib",
            "stroke": "#f4cccc"
        }
    ],
    "images": [
        {
            "image": "CGastrophoto_logo.jpg",
            "x": "1265",
            "y": "935",
            "tlx": -401265,
            "tly": -40935,
            "id": "oe-field-1"
        }
    ],
    "settings": {
        "defaultdatafileexpiry": "550",
        "defaultincludeplanets": false,
        "defaultincludesun": true,
        "defaultincludemoon": true,
        "defaultimagetopacity": 0.63,
        "defaultimagerotation": 0,
        "defaulttextrotation": 0,
        "defaultfontopacity": 1,
        "defaultfontcolour": "white",
        "defaultfont": "Arial",
        "defaultfontsize": 52,
        "defaultimagescale": 1,
        "defaultnoradids": "25544",
        "defaultstrokecolour": ""
    },
    "fonts": {
        "moon_phases": {
            "fontPath": "fonts\/moon_phases.ttf",
            "fonttcolour": "red",
            "backgroundcolor": "white",
            "fontopacity": 0.5
        },
        "ds-digi": {
            "fontPath": "\/fonts\/DS-DIGI.TTF"
        },
        "ds-digib": {
            "fontPath": "\/fonts\/DS-DIGIB.TTF"
        },
        "ds-digii": {
            "fontPath": "\/fonts\/DS-DIGII.TTF"
        },
        "ds-digit": {
            "fontPath": "\/fonts\/DS-DIGIT.TTF"
        }
    }
}
```

ea1ii, 2023-05-10:
Noted image x and y values quoted as strings. Removed the double quotes and and here's my very simple (at the moment) overlay
[image]

Also
[image]

And
[image]

ea1ii, 2023-05-10:
The image is not rendered though

ea1ii, 2023-05-10:
I was slightly off the frame. Moved fully inside and now coordinates a stored as numbers, not strings. 
[image]

And is rendered
[image]

EricClaeys, 2023-05-11:
@Alex-developer   Alex, please note that the image x and Y were initially stored as strings, not numbers, which gave the `ERROR: unsupported operand type(s) for -: 'str' and 'int'` message and caused the overlay to not display.
Also, when the image was partially off screen it didn't display or give an error.

Alex-developer, 2023-05-11:
> @Alex-developer Alex, please note that the image x and Y were initially stored as strings, not numbers, which gave the `ERROR: unsupported operand type(s) for -: 'str' and 'int'` message and caused [...]

I fixed the string x and y on text fields but obviously forgot on images, I'll sort that when back.

If an overlayed image would display outside the main image then it will not be displayed and an error will be logged. I could look at changing this but it should only be an issue if the co ordinates are manually set. When using the overlay manager the image will have a red rectangle around it if its out of bounds.

Open to suggestions on what the best fix for this is

EricClaeys, 2023-05-11:
Thanks @Alex-developer.
Can the red border also appear if the coordinates were manually set?

Maybe in the next release a place for textual errors/warnings to appear in cases where the field is completely off the image?  Kinda like messages in the WebUI.

Alex-developer, 2023-05-11:
> Thanks @Alex-developer.
> 
> Can the red border also appear if the coordinates were manually set?
> 
> 
> 
> Maybe in the next release a place for textual errors/warnings to appear in cases where the field is completely off the image?  Kinda like messages in the WebUI. 

That's a good idea Eric. I'll have a think about the best way to do it

EricClaeys, 2023-05-12:
@Alex-developer,  That "error/warning" area in the Overlay Editor (and I suppose the Module Manager if appropriate) could be used for all sorts of messages, even "success" ones, like in the WebUI.

Alex-developer, 2023-05-22:
Fixed in Dev branch

Alex-developer, 2023-05-23:
Issue with images not being highlighted when using the spinners fixed in dev branch so closing

closed 2023-05-23 by Alex-developer