AllskyTeam/allsky #2625: [BUG] Variable in variable manager not correct

jcauthen78, 2023-05-12:
### Environment

* Camera:  ZWO
* OS: Bullseye
* Allsky version: 2023.05.01
* Pi model: RPI 4b, 8gb
* SD Card: 128gb

### Bug Description

I use python to dump various data to JSON files to be read by AllSky into the Variables Manager.  A singular value is not being associated correctly, but the others are just fine
* In the `Variable Manager` --> `All Variables`, it is listed correctly & has its associated value (see screenshots below). 
* After Adding it to the `AllSky Variables`, it shows correct. 
* When I click the orange `edit` button, the WRONG variable info is displayed
* When I add it as an overlay, the same WRONG variable is displayed
`~/allsky/tmp/overlaydebug.txt` has the value in question listed CORRECTLY

`All Variables`
[image]

`AllSky Variables`
[image]

`EDIT button`
[image]

`overlaydebug.txt`
[image]

### Log / configuration files

`git log | head`
pi@allsky:~/allsky $ cd ~/allsky; git log | head
commit c645775e28cba6c9a8b7e0b52ab6b0b10309a494
Author: Eric Claeys <83164203+EricClaeys@users.noreply.github.com>
Date:   Tue May 2 10:38:30 2023 -0600

    changeLog.html: update version (#2602)

commit 841cc2bcbc84cf62b7549605de6fcc49261ead82
Merge: ff357bb8 fe9f0346
Author: Eric Claeys <83164203+EricClaeys@users.noreply.github.com>
Date:   Tue May 2 10:10:22 2023 -0600

`allsky.log clip` (line 156 shows the offending)
[allsky_log.txt](https://github.com/thomasjacquin/allsky/files/11458502/allsky_log.txt)

### Troubleshooting
* Deleted JSON file containing variable in question & re-loaded the editor, variable was successfully removed from the list. 
* Ran my python script again to create the JSON file. Adding the variable worked as expected, but the placement or editing is incorrect again - displaying the SAME offending variable. 
* Rebooted, did not fix.

EricClaeys, 2023-05-12:
@jcauthen78    Jim,  We recommend not prefixing your own variables with "AS_" since that's what the Allsky variables use, and I believe the overlay code treats them differently.  Try using a different prefix.

@Alex-developer, Alex, FYI

jcauthen78, 2023-05-12:
I have no say in the AS_ prefix. They get added in that way by default, or at-least when pulled from the JSON files.

jcauthen78, 2023-05-12:
[image]
 this is what the JSON file is written as - the AS_ is prefixed by AllSky

jcauthen78, 2023-05-12:
[image]

sorry, this is the RPM json file - but from this and the previous above, the `AS_` isn't anywhere in my pipeline, it seems to be added by AllSky when it queries the files in the `extra` folder.

jcauthen78, 2023-05-12:
Additional Troubleshooting: 
- Happens in both FireFox & Chrome, in MacOS
- cleared both caches, no fix.

jcauthen78, 2023-05-15:
just in case it's helpful - here's the entire `overlaydebug.txt`. not sure if it helps to know how many variables it's parsing. 
[overlaydebug.txt](https://github.com/thomasjacquin/allsky/files/11479551/overlaydebug.txt). I've noticed it having an issue with the last made overlay variable having the hiccup (not the last in the file debug), meaning when I had a new variable created for a new piece of data, the previous problem variable works fine when re-overlaying it, but the new variable now has the hiccup.. hopefully that makes sense. I can try to make a short video of it if y'all think it'll help.

I'd like to suggest the user variables be given a different prefix by AllSky. Looking at the list right now there are 2 main prefixes listed: `allsky_` & `as_`, and a bunch of options without any prefixes.  Maybe have the imported variables prefixed with `usr_` or `custom_`something like that?

EricClaeys, 2023-05-15:
I had the same thing happen to me during testing.  `saveImage.sh` exported a new variable and I was able to see it in the "All Variables" and copy it to "Allsky Variables" by clicking on the "+" icon, but when I added it to the overlay it was a different variable.
I THINK I got it to work by copying the variable to "Allsky Variables" (which if I recall correctly did NOT cause the "save" icon to turn green), then changing some existing field to force the "save" icon to turn green so I could save, then adding the new variable to the overlay.

jcauthen78, 2023-05-15:
Hmm.  I'll give that process a try a little later today and see what it does on my end.
1. Add to AllSky Variables
2. Change existing field to force Save icon
3. Save
4. Add new variable to overlay.

Alex-developer, 2023-05-22:
I think this is the same as https://github.com/thomasjacquin/allsky/issues/2649 

I reproduced the problem as follows

- Add a new variable to the list from the 'all variables' tab
- switch back to the 'Allsky variables' tab and select the last field, this will be the one added above
- Instead of adding the last field its adding the one above it. 

If you refresh the overlay editor then you can add the vairable ok

Alex-developer, 2023-05-22:
Fixed in PR https://github.com/thomasjacquin/allsky/pull/2676 which is currently in the DEV branch. The fix will be in the next minor release

closed 2023-05-22 by EricClaeys