AllskyTeam/allsky #2629: [BUG]

sonny486, 2023-05-12:
* Camera:    RPiHQ
* OS: Bullseye 64bit
* Allsky version: v2023.05.01
* Pi model Raspberry Pi 4 with 4 gigs of memory

### Bug Description

ERROR: Unable to process settings file '/home/allsky/allsky/config/settings.json'.
File '/home/allsky/allsky/config/settings.json' missing!

New install - Installed per instruction in Documentation section. 

https://htmlpreview.github.io/?https://raw.githubusercontent.com/thomasjacquin/allsky/master/html/documentation/installations/Allsky.html

paulwwerner, 2023-05-12:
dito, but 32bit works

EricClaeys, 2023-05-12:
@paulwwerner would you please attach that file as well?

sonny486, 2023-05-13:
Even 32 bit didn't work.

[image]

sonny486, 2023-05-13:
[Installation Logs.zip](https://github.com/thomasjacquin/allsky/files/11468461/Installation.Logs.zip)

EricClaeys, 2023-05-13:
@sonny486    Sonny, the log file says it can't find the RPi camera. Did you have it working prior to installing Allsky?
Please run `libcamera-hello` and paste the output here.   If the camera is working you should see an image appear for a few seconds, then disappear.

EricClaeys, 2023-05-13:
@sonny486    Please also run `vcgencmd get_camera`.

wj7106, 2023-05-13:
I have also encountered this issue. Previously, I encountered the issue of not being able to save longitude and latitude. I reformatted the tf card and reinstalled the system, but the result kept prompting for this. Both 32-bit and 64 Bulleye systems have been tried, and the same prompt still shows that settings. json is missing

EricClaeys, 2023-05-13:
@wj7106   Please ATTACH `~/allsky/config/installation_logs/install.sh.log`, and run
~~~
libcamera-hello
vcgencmd get_camera
~~~

In Sonny's case, the message about "settings.json not found" should not have appeared - there should have been a message about a missing camera.  I fixed that bug but haven't released it yet.

wj7106, 2023-05-13:
After the libcamera hello command, the camera is fine. You can connect the RPI camera and the screen can display photos

wj7106, 2023-05-13:
I have now given up using the latest version of allsky. Once it is fixed, I will reinstall and upgrade again

sonny486, 2023-05-13:
Yeah looks like lightning took out my camera.  Lol.  I replaced it with another RPiHQ, have a working libcamera-hello (Shows live video on the desktop).
vcgencmd get_camera gives this response: supported=1 detected=0, libcamera interfaces=1

libcamera-hello output:
[0:02:35.558153324] [2615]  INFO Camera camera_manager.cpp:299 libcamera v0.0.4+22-923f5d70
[0:02:35.595009142] [2616]  INFO RPI raspberrypi.cpp:1476 Registered camera /base/soc/i2c0mux/i2c@1/imx477@1a to Unicam device /dev/media4 and ISP device /dev/media0
[0:02:35.595955445] [2615]  INFO Camera camera.cpp:1028 configuring streams: (0) 2028x1520-YUV420
[0:02:35.596291267] [2616]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1520-SBGGR12_1X12 - Selected unicam format: 2028x1520-pBCC

On Allsky I get a Allsky starting up, but never get an image.

EricClaeys, 2023-05-13:
@sonny486 What a bummer.
Is the output above with the broken camera or the good one?

Look in /var/log/allsky.log - I suspect images are being deleted for being too dark, i.e.  under the 0.1 threshold. If that's the case then adjust the LOW threshold in config.sh.

EricClaeys, 2023-05-13:
@wj7106 Please ATTACH ~/allsky/config/installation_logs/install.sh.log.  It will help me debug your problem.

wj7106, 2023-05-13:
STARTING INSTALLATON AT Sat 13 May 22:53:26 CST 2023.

Using the 'master' branch.
  DEBUG: TEMP_LOCALE=LANG=en_GB.UTF-8, CURRENT_LOCALE=en_GB.UTF-8
* Using RPi camera.
  DEBUG: RAM_SIZE=512, SUGGESTED_SWAP_SIZE=3584.
* Size of current swap (3584 MB) is sufficient; no change needed.
* /home/pi/allsky/tmp is currently in memory; no change needed.
  DEBUG: /home/pi/allsky/tmp already mounted
* Installing the web server.
* Installing dependencies.
* Preparing Allsky commands.
* Modifying locations for WebUI.
* Setting up WebUI options and default settings for RPi cameras.
Executing makeChanges.sh --force  --cameraTypeOnly   'cameraType' 'Camera Type' '' 'RPi'
WARNING: Sensor 'imx219' connected to Pi but not known by Allsky
ERROR: Could not find information on camera 'imx219'.
[0;31mERROR: Unable to create cc file '/home/pi/allsky/config/cc.json'.[0m
*** ERROR: 
**********
Unable to save camera capabilities.
**********

ENDING INSTALLATON AT Sat 13 May 22:54:32 CST 2023.

EricClaeys, 2023-05-13:
@wj7106. What kind of camera do you have?  Allsky doesn't support the imx219 sensor, at least not yet.

sonny486, 2023-05-13:
@EricClaeys Yeah, I will take a look.  That is what it looks like, it just continuously says Allsky is starting up.

I did notice however if I look at the images panel that the date for the day was written, however no images in the listbox.

sonny486, 2023-05-13:
You were right, image is below the threshold.  However when I set the threshold to .1 it still shows removed due to threshold at 1.

May 13 12:02:12 allsky allsky[3711]: STARTING EXPOSURE at: 2023-05-13 12:02:12   @ 1.4 sec
May 13 12:02:16 allsky allsky[3711]: STARTING EXPOSURE at: 2023-05-13 12:02:16   @ 540.03 ms (0.54 sec)
May 13 12:02:19 allsky allsky[3711]: STARTING EXPOSURE at: 2023-05-13 12:02:19   @ 206.22 ms (0.21 sec)
May 13 12:02:21 allsky allsky[3711]: STARTING EXPOSURE at: 2023-05-13 12:02:21   @ 78.75 ms (0.08 sec)
May 13 12:02:24 allsky allsky[3711]:   > Saving DAY image 'image-20230513120221.jpg'
May 13 12:02:26 allsky allsky[3833]: removeBadImages.sh: File is bad: removed 'image-20230513120221.jpg' (below threshold: MEAN=0.993559, threshold = 1)
May 13 12:02:39 allsky allsky[3711]: STARTING EXPOSURE at: 2023-05-13 12:02:39   @ 30.07 ms (0.03 sec)
May 13 12:02:42 allsky allsky[3711]:   > Saving DAY image 'image-20230513120239.jpg'
May 13 12:02:44 allsky allsky[3921]: removeBadImages.sh: File is bad: removed 'image-20230513120239.jpg' (below threshold: MEAN=0.717188, threshold = 1)

sonny486, 2023-05-13:
Found the setting for threshold in the /EDITOR/ .  

All is working as it should now.  Thanks!

closed 2023-05-13 by sonny486

EricClaeys, 2023-05-13:
@sonny486 do you remember what the LOW and HIGH values were before you changed the, and what they are now?

sonny486, 2023-05-13:
Low was 1
High was 90

Moved it to .1 and kept 90 on high

EricClaeys, 2023-05-13:
@sonny486 Thanks.  I found a bug that we'll fix in a release in a couple weeks.  At that point you can set the low value back to "1".