AllskyTeam/allsky #2635: New installation hangs installing 'scikit-image'

EricClaeys, 2023-05-14:
### Discussed in https://github.com/thomasjacquin/allsky/discussions/2607

<div type='discussions-op-text'>

<sup>Originally posted by **nigelg49** May  5, 2023</sup>
Just trying to do a fresh install with the new software.
Raspberry pi 4 8GB, RPiHQ camera. after waiting 1 hour I got to this stage

* Installing Python dependencies.
        This may take a LONG time if the packages are not already installed.
*    === Package # 1 of 15: [opencv_python>=4.5.3.56]
*    === Package # 2 of 15: [Pillow]
*    === Package # 3 of 15: [pyephem]
*    === Package # 4 of 15: [skyfield]
*    === Package # 5 of 15: [astral]
*    === Package # 6 of 15: [pytz]
*    === Package # 7 of 15: [scipy]
*    === Package # 8 of 15: [scikit-image]
Waited another hour and nothing has changed. Opened the WebUI which said I had 2 errors.

ERROR: 2 required fields are missing (Latitude, Longitude) - see highlighted entries below
--
I added the Lat Long, but when I save changes the 2 fields no longer contain anything.
Any help would be appreciated.
May  5 13:45:56 willows allsky[3986]: *** ERROR: Latitude not specified!
May  5 13:45:56 willows allsky[3986]: *** ERROR: Longitude not specified!
May  5 13:45:59 willows allsky[3986]:      ***** Stopping AllSky *****
May  5 13:45:59 willows allsky[3714]: ***
May  5 13:45:59 willows allsky[3714]: *** After fixing, restart the allsky service. ***
May  5 13:45:59 willows allsky[3714]: ***
May  5 13:45:59 willows allsky[3714]:      ***** AllSky Stopped *****
May  5 13:18:31 willows allsky[894]:      ***** Starting AllSky *****
May  5 13:18:44 willows allsky[1393]: **************************************************
May  5 13:18:44 willows allsky[1393]: *** Allsky Camera Software Version v2023.05.01 ***
May  5 13:18:44 willows allsky[1393]: **************************************************
May  5 13:18:44 willows allsky[1393]: Capture images of the sky with a Raspberry Pi and an RPi camera
May  5 13:18:44 willows allsky[1393]: Add -h or --help for available options
May  5 13:18:44 willows allsky[1393]: Author: Thomas Jacquin - <jacquin.thomas@gmail.com>
May  5 13:18:44 willows allsky[1393]: Contributors:
May  5 13:18:44 willows allsky[1393]:  -Knut Olav Klo
May  5 13:18:44 willows allsky[1393]:  -Daniel Johnsen
May  5 13:18:44 willows allsky[1393]:  -Robert Wagner
May  5 13:18:44 willows allsky[1393]:  -Michael J. Kidd - <linuxkidd@gmail.com>
May  5 13:18:44 willows allsky[1393]:  -Rob Musquetier
May  5 13:18:44 willows allsky[1393]:  -Eric Claeys
May  5 13:18:44 willows allsky[1393]:  -Andreas Lindinger
May  5 13:18:44 willows allsky[1393]: *** ERROR: Latitude not specified!
May  5 13:18:44 willows allsky[1393]: *** ERROR: Longitude not specified!
May  5 13:18:47 willows allsky[1393]:      ***** Stopping AllSky *****
May  5 13:18:48 willows allsky[894]: ***
May  5 13:18:48 willows allsky[894]: *** After fixing, restart the allsky service. ***
May  5 13:18:48 willows allsky[894]: ***
May  5 13:18:48 willows allsky[894]:      ***** AllSky Stopped *****

Nigel

</div>

EricClaeys, 2023-05-14:
Python_dependencies.8.log contained:
~~~
DEPRECATION: The -b/--build/--build-dir/--build-directory option is deprecated and has no effect anymore. pip 21.1 will remove support for this functionality. A possible replacement is use the TMPDIR/TEMP/TMP environment variable, possibly combined with --no-clean. You can find discussion regarding this at https://github.com/pypa/pip/issues/8333.
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting scikit-image
Using cached scikit_image-0.20.0.tar.gz (22.9 MB)
Installing build dependencies: started
Installing build dependencies: finished with status 'done'
Getting requirements to build wheel: started
Getting requirements to build wheel: finished with status 'done'
Installing backend dependencies: started
Installing backend dependencies: finished with status 'done'
Preparing wheel metadata: started
Python_dependencies.8.log (END)
~~~

EricClaeys, 2023-05-14:
@Alex-developer   Will look into this as it appears the installation is still running.

@EricClaeys thinks someone mentioned that one of the packages took a couple hours to install, so it's possible that is the case here as well.

petmos, 2023-05-16:
I saw the same issue on Raspberry pi 4 from a collegue who used the 32bit OS.
After flashing the 64bit OS installation was sucessful.

EricClaeys, 2023-05-17:
@Alex-developer what was the issue?  Does the fix also fix the --build option?

Alex-developer, 2023-05-17:
If you want to try the fix ahead of the release then edit the config_repo/requirements.txt and replace its contents with

opencv-python-headless<=4.6.0.66
Pillow
pyephem
skyfield
astral
pytz
scipy<=1.8.1
scikit-image<=0.19.3
paho-mqtt
numpy<=1.24.2
photutils
astropy
suncalc
Adafruit-Blinka
vcgencmd

@EricClaeys The code is in my dev banch but will not commit it until I have sorted the --build option

EricClaeys, 2023-05-22:
Closing issue - fixed in point release.