AllskyTeam/allsky #2637: [BUG]

HeavyJohnny, 2023-05-15:
* Camera:  ZWO ASI178
* OS: Bullseye
* Allsky version: v2023.05.01
* Pi4 8GB

### Bug Description

Fresh install new rig. When outside and disconnected the USB from the camera to the pi and conected it to a laptop so I could get focus. After achieving focus using the asi software I reconnected the pi to the camera and began getting an error in my log during image capture: 

"May 15 23:32:09 allsky allsky[31406]:    cp: cannot create regular file '/var/www/html/allsky//image.jpg': No such file or directory"  

After verifying the folder exists in the allsky .dir on the PI, I clicked the button on the bottom of the Allsky Settings page to reset to default values. Now I am unable to input my Long and Lat as well. 

I restarted the Pi, and the UI but the issue prevails

### Log / configuration files

[image]

[config.txt](https://github.com/thomasjacquin/allsky/files/11479274/config.txt)
[ftp_Settings.txt](https://github.com/thomasjacquin/allsky/files/11479275/ftp_Settings.txt)
[config.jason.txt](https://github.com/thomasjacquin/allsky/files/11479276/config.jason.txt)
[Log1.txt](https://github.com/thomasjacquin/allsky/files/11479277/Log1.txt)
[Log2.txt](https://github.com/thomasjacquin/allsky/files/11479278/Log2.txt)

EricClaeys, 2023-05-15:
@HeavyJohnny, Kevin, did you install the new Allsky Website by running `website/install.sh`?  If so, you need to change the IMAGE_DIR in ftp-settings.sh to point to `/home/pi/allsky/html/allsky`, and change the `videos`, `startrails`, and `keograms` settings accordingly.  And spell them correctly.  That will fix the "cp" Error.

What happens when you enter the latitude and longitude and press the "Save changes" button?

HeavyJohnny, 2023-05-16:
Nothing would happen It would refuse to save and kick an error at the top of the scree stating long and lat could not me left blank. I formatted my pi and reinstalled.

HeavyJohnny, 2023-05-16:
Also, Changing the directory helped. Thank you. I used the folder that Chad gave on his youtube tutorial, which I know is out of date. Thanks for the help!

EricClaeys, 2023-05-16:
Chad will be updating his video now that the new release is out.  The Allsky Documentation via the WebUI has all the information you should need and is up-to-date. 

Is everything working now?

HeavyJohnny, 2023-05-16:
Looking forward to seeing him back on youtube. I hope everything works out
for him.

closed 2023-05-17 by HeavyJohnny