AllskyTeam/allsky #2717: Requested changes to flow timer

EricClaeys, 2023-05-30:
Requested changes to the flow timer change:

- Allow the number of images to process to optionally be passed as an argument (to flow_runner.py ?).  The default can remain at 10.
- Add a variable to allsky/variables.sh to define the directory, e.g., ALLSKY_FLOWTIMINGS.  This will make it easy for check_allsky.sh to find the directory.
- How will check_allsky.sh know the names of the files?  Should other variables be added, e.g., ALLSKY_FLOWTIMINGS_XXX ?
- allsky/allsky.sh should `rm -fr "${ALLSKY_FLOWTIMINGS}"` so we know any files are from the current instance of Allsky.
- Please make times stored in the files in integer milliseconds.

Alex-developer, 2023-05-31:
- Allow the number of images to process to optionally be passed as an argument (to flow_runner.py ?). The default can remain at 10.

>Added --flowtimerframes option to flow-runney.py Default is 10

- Add a variable to allsky/variables.sh to define the directory, e.g., ALLSKY_FLOWTIMINGS. This will make it easy for check_allsky.sh to find the directory.

>Added the following variables
>ALLSKY_FLOWTIMINGS="${ALLSKY_TMP}/flowtimings"
>ALLSKY_FLOWTIMINGS_DAY="${ALLSKY_FLOWTIMINGS}/day-average"
>ALLSKY_FLOWTIMINGS_NIGHT="${ALLSKY_FLOWTIMINGS}/night-average"

- How will check_allsky.sh know the names of the files? Should other variables be added, e.g., ALLSKY_FLOWTIMINGS_XXX ?

> See above

- allsky/allsky.sh should rm -fr "${ALLSKY_FLOWTIMINGS}" so we know any files are from the current instance of Allsky.

> New option --cleartimings added to flow-runner.py. This is called from allsky.sh to clear all timing data

- Please make times stored in the files in integer milliseconds.

> Values are now ints

EricClaeys, 2023-05-31:
Closing Issue.  Changes will be in upcoming point release.
Thanks Alex!!

closed 2023-05-31 by EricClaeys