AllskyTeam/allsky #2813: Install instructions followed - Asking for username and password on the git clone command line

thearduinoguy, 2023-06-19:
### Environment

* Camera:  Pi HQ Camera
* OS: Bullseye
* Allsky version:
      Latest
* Pi model and amount of memory (512 MB, 1, 2, 4, or 8 GB)
* Pi 4 4Gb

### Bug Description

* Include a clear and concise description of what the bug is.
* 
Install instructions followed - Asking for username and password on the git clone command line

### Log / configuration files

EricClaeys, 2023-06-19:
Did you use `sudo git clone ...` ?
If so, that means the user you are logged in as (usually "pi") does not have permission to execute `sudo`.
Please run the following:
~~~
id
groups
~~~

Has Allsky run on this Pi using the current operating system before?

EricClaeys, 2023-07-04:
@thearduinoguy    Mike, any update?

EricClaeys, 2023-12-09:
Closing Issue - no response from user, and this user is the only one experiencing the problem.

closed 2023-12-09 by EricClaeys