AllskyTeam/allsky #2834: [BUG]

snowformatics, 2023-06-24:
### Environment

* Camera:  ZWO 120MC
* OS: Bullseye
* Allsky version:latest
* Pi 4 (4, GB)

### Bug Description
Hi,

I am trying to configure the ftp setting file but I have problems to upload my images. 

1) I created folders (images, kerogram, startrail) under ~/allsky/html
-> Is this correct? 

2) I edited the ftp-settings.sh file and set the paths like:

IMAGE_DIR="~/allsky/html/images"

3) When I check the log file 
~~~
Upload output from '/home/x/allsky/tmp/image-20230624205054.jpg:
Jun 24 20:50:57 allsky allsky[1194]:    cp: cannot create regular file '~/allsky/html/images/image-20230624205054.jpg': No such file or directory
~~~

I have only a local WebUI running, I can see the images in the live view but they don't appear in the image folder.

Any help would be appreciated
Stefanie

EricClaeys, 2023-06-24:
Stephanie,
Please delete the folders you created, then run:
~~~
cd
website/install.sh
~~~
It will create everything you need in the correct places.

Then look at the bottom of the "ftp-settings.sh" section of the [Allsky Website Settings](https://htmlpreview.github.io/?https://raw.githubusercontent.com/thomasjacquin/allsky/master/html/documentation/settings/allskyWebsite.html)  page to see what to enter.
Let me know how it goes.

snowformatics, 2023-06-25:
Hi Eric,

I installed the website again and I entered the following path settings:
```
IMAGE_DIR=""
VIDEOS_DIR="${ALLSKY_WEBSITE}/videos"
KEOGRAM_DIR="${ALLSKY_WEBSITE}/keograms"
STARTRAILS_DIR="${ALLSKY_WEBSITE}/startrails"
```

Unfortunately this raises again " No such file or directory". 
Then I set the IMAGE_DIR to:

`IMAGE_DIR="${ALLSKY_WEBSITE}/"`

Not sure whether this correct but the images are stored now in the website folder. But I still can not see them inside the website "Images" menu.

[image]

Thanks
Stefanie

EricClaeys, 2023-06-25:
@snowformatics Stefanie, sorry I misspelled your name in my last message.

There are two different websites.  Your screenshot is from the WebUI which runs on the Pi and gets all its images from the Pi.  The `Images` page has an "All" line that always appears as well as one line for each day in ~/allsky/images.  The "All" line never has an icon for the images because there could be tens of thousands of them.
Has Allsky been running? If so, there should be at least one additional line.  Are there any directories in ~/allsky/images ?

The other website is the optional Allsky Website which uses the ftp-settings.sh file to tell it where to copy the images to - either locally on the Pi (as in your case) or on a remote server, or both.  You can get to your local Allsky Website by clicking on the "Local Website" link at the top of any WebUI page.

snowformatics, 2023-06-25:
No worries Eric.

I have no directory called images in ~/allsky, should I create it manually? Allsky WebUI is running and creating images but bthe local Allsky Website just displays loading...

Thanks for your support,

[image]
[image]

EricClaeys, 2023-06-26:
Getting those lines above the image usually indicates a missing file.
Since there appears to be some missing and/or corrupted files, I suggest re-installing Allsky and the Allsky Website.  It should be easy and go quickly.

See the Allsky Documentation --> Installing Allsky page in the WebUI for details.
If there are any issues, attach `~/allsky/config/installation_logs/install.sh.log`.

EricClaeys, 2023-06-26:
You shouldn't need to create any files or directories  - the installations do that.

snowformatics, 2023-06-26:
I deletes allsky and run the installation again (went smoothly this time) but I still don't have an images folder. I attached the installation log file.

[install.log](https://github.com/thomasjacquin/allsky/files/11865234/install.log)

EricClaeys, 2023-06-27:
Let's first make sure the core Allsky is working, then we will work on the Allsky Website in `~/allsky/html/allsky`, ok?

If you go to the WebUI's "Liveview" page, do you see the most recent image?  On the "Allsky Settings" page, are you saving daytime images?  If not, the current day's folder in allsky/images won't be created until nighttime.
If there is nothing in `~/allsky/images` at the end of the day, please attach `/var/log/allsky.log`

snowformatics, 2023-06-27:
Hi Eric,

it missed the advanced settings for daytime saving, it looks like it's working now. I see the daily folder on the WebUI and inside ~/allsky/images.

Thanks for your patience, I will try this night.

EricClaeys, 2023-07-04:
@snowformatics    Stefanie, any update?  If everything's working would you please close this item?
Thanks - Eric

snowformatics, 2023-07-05:
Hi Eric,
I am still waiting for my parts and the weather is to nasty to try without waterproof case. Indoor everything works, so I will close this item. Thanks again!
Stefanie

closed 2023-07-05 by snowformatics