AllskyTeam/allsky #2835: [ENHANCEMENT] Update the WebUI

Alex-developer, 2023-06-24:
Update the WebUI to be

1) Responsive
2) Provide a dashboard
3) Improve the login system
4) Make it API first
5) More configurable by the user

EricClaeys, 2024-11-28:
Suggestion from @jsalsburg:
* Add a monitor and system alarm/message for `/dev/sda*` memory value, a display in the image file overlay, something like that generated by the `df -`P command line.

EricClaeys, 2024-11-28:
Suggestion from @elwell:
* Show IPv6 addresses to "LAN Dashboard" WebUI page (from [Issue #2579](https://github.com/AllskyTeam/allsky/issues/2579).

EricClaeys, 2024-12-06:
The overlay editor needs to be more tablet friendly for example, selecting a field is difficult because the double tap tends to move the field.