AllskyTeam/allsky #2838: [BUG]: Default stroke color doesn't work

EricClaeys, 2023-06-25:
I have black (the top left color) as the default stroke color.
Whenever I add a new field the stroke color is as shown in the screenshot below.
The default font color (white in my case) works fine as can be seen in the screenshot.

[image]

Alex-developer, 2023-06-28:
@EricClaeys A few things

1) Currently the stroke colour against the field will not reflect the default colour, this is by design and intended to reflect the fact that the field is using the default. After the field has been saved it will change to the default.
2) In the screenshot above there is no 'Stroke Size' so nothing will appear

Cheers

Alex

EricClaeys, 2023-07-02:
I request a change to the design based on an inconsistency which confuses me:

* The "Colour" field shows the default colour, if one is set, otherwise the gray/black boxes.  This makes perfect sense.
* The "Stroke Colour" field ALWAYS shows the gray/black boxes, regardless of whether or not a default Stroke Color is set.

This is inconsistent.  I suggest having BOTH fields show the default color if one is set, otherwise the gray/black boxes.

After some more testing I noticed the default stroke color does in fact work, but the "Stroke Colour" drop down doesn't show the default until AFTER a stroke size it set AND the field is saved.
 
[image]

Alex-developer, 2023-07-06:
Fix added - In testing will be committed with other changes

closed 2023-09-02 by Alex-developer