AllskyTeam/allsky #2854: Overlay issue

bribon2, 2023-07-03:
Hello, good afternoon. I have a problem with the overlay.
The altitude of the Moon does not correspond to the real thing, and the coordinates entered in the webui are correct.
For example, right now, the moon is at -53º (real) and the overlay shows an altitude of -25º.
Then I have the problem that when I activate the "SUN" option, all the overlay of the image disappears. If I deactivate it, it appears again.
The website is: https://observatoridepujalt.cat/allsky/
All the best.

Alex-developer, 2023-07-03:
I will have a look at these issues

Cheers

Alex

Alex-developer, 2023-07-04:
Added fix - Will test for a few days before commiting

remidone, 2023-07-09:
> Hello, good afternoon. I have a problem with the overlay. The altitude of the Moon does not correspond to the real thing, and the coordinates entered in the webui are correct. For example, right now, the moon is at -53º (real) and the overlay shows an altitude of -25º.

 **Me too**

 >Then I have the problem that when I activate the "SUN" option, all the overlay of the image disappears. If I deactivate it, it appears again. 

**Maybe because you have an error like "sun never go under 6 degrees below horizon"**

EricClaeys, 2023-07-09:
This will be fixed in the 3rd point release which will be out in a week or two.  We're testing the fixes now.  There are several other overlay-related fixes that will be included.

remidone, 2023-07-09:
> This will be fixed in the 3rd point release which will be out in a week or two. We're testing the fixes now. There are several other overlay-related fixes that will be included.

Thank you guys! Good job!

shinigami-de, 2023-08-06:
Do the fixes also cover locale issues? When using a locale with ',' instead of '.' for decimals, there seem to be multiple issues with the lat and lon settings. The "allsky settings" expects the locale version (here with , ) to accept anything after the decimal sign -> 48.4123 would end up as 48.0 otherwise. Putting it in correctly with ,  will instead break the overlay for e.g. sun/moon altitudes, since _convertLatLon in line 1158 is using "float" instead of "s.float", hence the overlay will disappear due to a conversion error.

EricClaeys, 2023-08-07:
@shinigami-de, the program that takes pictures will allow commas in latitude and longitude.

@Alex-developer  Alex, would you please take a look at the module-related lat/long issue.

@shinigami-de, which file on line 1158 are you referring to?  Thanks for investigating.

shinigami-de, 2023-08-07:
@EricClaeys thank you for the reply. Sorry that I did not provide all needed information.
The "issues" arise, when I use the + or - notation for the GPS coordinates instead of the N/S/W/E notation. So e.g. +48.4123 instead of 48.4123N. Then on startup the config output at the beginning of allsky.log will show a latitude of 48,00000N and in this case, sometimes even trigger the error "Sun never reaches 6 degrees below the horizon". Using N notation, the log will show the exact number and everything seems to work. Not sure, if this is intended, but a degree resolution seems a bit coarse if that is also returned to other modules.
The line 1158 I mentioned was from the allsky_overlay.py in allsky/scripts/modules. Might be different now, if you already have a merged patch of course. The function you are looking for is _convertLatLon(self, input).
Also mentioning locales, using the + notation seems to also throw off at least the openweathermap module. Here, when using the +/- notation, requests will be sent with inverted signs and sometimes screw up the coordinates, when using "." in a locale where this char denotes thousands. E.g. in this case entering +48.4123 will end up getting a request to lat -48304.0 (note that the config output showed 48,00000). But that is a different topic I guess.

EricClaeys, 2023-08-07:
@shinigami-de
When @Alex-developer returns in a few days he'll need to look at the overlay error and hopefully be able to fix it (if not already fixed) for Point Release 3.

For Point Release 3 I tested the capture program that takes pictures using the US and German locales with both periods and commas in the latitude/longitude, and it worked.  So did +/- and NSEW.
Allsky passes the angle, latitude, and longitude to the `sunwait` command which determines if it's currently DAY or NIGHT, and what time the day/night transitions are.  I've tested it with periods and commas and it works with both, so I'm not sure why it outputs "Sun never reaches 6 degrees below the horizon".  If you can get that message, please set the Debug Level to 4, then look for the message.  Right above it should be the `sunwait` command that was executed.  I'm curious what it was since it's possible there's another bug.

In the meantime, I wouldn't worry about dropping numbers after the decimal point - the difference isn't usually enough to make a big difference in when nighttime and daytime are, maybe a couple minutes.

shinigami-de, 2023-08-07:
@EricClaeys That is strange indeed. Not sure, if there is something else going on. My locale for reference is de_DE-UTF-8, if that is of any help.
I can reliably reproduce the "Sun never reaches 6 degrees below the horizon, at this location." by setting 48.somethingN , 8.somethingE . Sunwait is run before the "STARTING EXPOSURE" message and resolves to "Executing sunwait poll exit angle -7.0000 48.somethingN 8.somethingE > /dev/null". So the numbers match exactly what was set in Allsky settings. By the way I have also set "Sun elevation" or "Angle" as it is called in the settings to -7. No other info that I can see in level4 that would point to something more obvious.
Not a big issue, since everything is running fine when I use "," and NSEW. Just took me a bit to figure out why my overlay disappeared :)

EricClaeys, 2023-08-18:
@shinigami-de please run `sunwait poll exit angle -7.0000 48.123N 8.123E` and let me know what it says.  I just ran it (at 2 am) and it said "DAY".  My locale is en_GB.UTF-8.

shinigami-de, 2023-08-29:
@EricClaeys Sorry for the late reply due to me being on vacation. Just ran the command. Output:
x@allsky:~ $ sunwait poll exit angle -7.0000 48.123N 8.123E
DAY
x@allsky:~ $ date
Di 29. Aug 15:33:20 CEST 2023

EricClaeys, 2023-08-29:
@Alex-developer were you able to look into this?

Alex-developer, 2023-08-29:
> @Alex-developer were you able to look into this?

Eric

Yes

See https://github.com/thomasjacquin/allsky/pull/2862

EricClaeys, 2023-08-29:
@Alex-developer.  Were they also merged into point release 3?

Alex-developer, 2023-08-29:
> @Alex-developer. Were they also merged into point release 3?

Yes all of the issues in the issues list, which includes this one, were merged into the point release

bribon2, 2023-09-01:
The planet data has been fixed in the new version but the Moon continues to show me an incorrect altitude.
Also, when I check the option to include the Sun in the overlay, all the image information keeps disappearing (date, time,...)

Alex-developer, 2023-12-22:
Fixes for this are in the PR4 release due out in the next day or so.

Closing this issue

closed 2023-12-22 by Alex-developer