AllskyTeam/allsky #2866: [BUG] Consistent delays doesn't work at night with ZWO

EricClaeys, 2023-07-09:
Version v2023.05.01_01 added ability to enable the Allsky auto-exposure mode at night which is a big improvement.  However, doing so broke the Consistent Delays at night.

EricClaeys, 2023-08-02:
Fixed in v2023.05.01 point release 3.

closed 2023-08-02 by EricClaeys

Apennine2, 2023-08-19:
Hi @EricClaeys , the new exposure algorithm has definitely addressed the flickering. I still have the non-consistent delay as described in this fixed issue. How do I upgrade to point release 3? Or, do I just wait until its officially out?

I downloaded the latest, but it says Version: v2023.05.01_02.

Thanks!

EricClaeys, 2023-08-19:
You can get point release 3 by adding `--branch v2023.05.01_03` to the `git clone` command.  If you upgrade now, please let me know how it went since we are still testing it.

commit 2221520a (2023-07-09): Update allsky_common.cpp: fix consistent delays bug

Fixes #2866.

Also:
* Better align settings output.
* Ignore some future settings.