AllskyTeam/allsky #2892: [BUG] Show on map disabled and my camera is on the map

Rapid1981, 2023-07-24:
I never turned on the option to show my camera on the map, but it was there, until I enabled, then disabled the Show on Map option.

EricClaeys, 2023-07-24:
@Rapid1981,
To make sure I understand, with "Show On Map" OFF your camera appears on the map whenever you add the two URLs in the map section of the WebUI.  Is that correct?

Would you please do the following?
~~~
# set "Show on Map" to No
# clear the URLs
# click "Save changes" button
sudo systemctl stop allsky
sudo truncate -s 0 /var/log/allsky.log
# in the WebUI:
#     set the Debug Level to 4
#     enter the URLs
#     click "Save changes"
~~~
When that finishes, attach /var/log/allsky.log and if there are any messages in the WebUI attach a screenshot of them.
You can then restore Debug Level to whatever it was.

Thanks - Eric

Rapid1981, 2023-07-25:
I did what you asked, but the allsky.log was 0 bytes after this.
Here's the messages the WebUI displayed after I clicked Save Changes.

[image]

EricClaeys, 2023-07-29:
This fix will be in point release 3.

commit 2a038a82 (2023-07-29): Update changeLog.html: add map fix

Fixes #2892
closed 2023-07-29 by EricClaeys