AllskyTeam/allsky #2893: Add SSD temperature module

EricClaeys, 2023-07-24:
### Discussed in https://github.com/thomasjacquin/allsky/discussions/2067
Add a module that gets SSD temperature and places it in an "extra" file.

Alex-developer, 2023-07-25:
Got the basics working just need to test with a few more drives to ensure the output from smartctl is consistent

Alex-developer, 2023-07-26:
Module is now written. I need to test it with multiple drives but am awaiting another USB to SATA convertor that support SMART.

Module Options

[image]

You can set colours for temp limits if you wish

[image]

As soon as I have a second convertor and tested it I will add it to the extra modules repo.

closed 2023-08-13 by Alex-developer

Rapid1981, 2023-08-23:
You've missed one dependency, the pySMART. After I installed it, it works flawlessly. Thank you!

Alex-developer, 2023-08-23:
> You've missed one dependency, the pySMART. After I installed it, it works flawlessly. Thank you!

Hmm thats odd the requirement is there. I assume you used the module installer to install it - Did it throw any errors?

Rapid1981, 2023-08-23:
Yes, it did, but the module appeared in the module manager, just the running time was 0.00s when I used it and in the logs it said the pySMART module is missing.

```
Installing allsky_hddtemp
=========================
INFO: Installing dependencies
sh: 1: cannot create /home/pi/allsky-modules/allsky_hddtemp/dependencies.log: Permission denied
INFO: Installing allsky_hddtemp module
SUCCESS: Module allsky_hddtemp Installed
ERROR: Installation of allsky_hddtemp Failed, Check /home/pi/allsky-modules/allsky_hddtemp/dependencies.log for any errors

```

Alex-developer, 2023-08-23:
How are you running the module installer - Looks like there are permission issues

Rapid1981, 2023-08-24:
I tried to run it with sudo, but the installer said "Don't run it with sudo", so normally, `./install.sh`