AllskyTeam/allsky #2895: [BUG] Latitude/Longitude do not match

remidone, 2023-07-30:
the coordinates in the allsky settings are not the same in allsky.log
 
[image]
[image]

I also tried to submit coordinates with "N" and "E" but in allsky.log I have "ERROR: Sun never reaches 6 degrees below the horizon, at this location."
[image]

EricClaeys, 2023-07-30:
Please do the following:
~~~
sudo systemctl stop allsky
sudo truncate -s 0 /var/log/allsky.log
# In the WebUI, set Debug Level to 4
# Click on Save Changes
~~~
after about 30 seconds, attach /var/log/allsky.log
Reset the Debug Level to whatever it was before.

Thanks - Eric

remidone, 2023-07-30:
Now there's no error but latitude and longitude haven't decimals

remidone, 2023-07-30:
It does not change anything

EricClaeys, 2023-07-31:
@remidone  Please attach the log file as requested.  It will help me troubleshoot the problem. 
I didn't expect it to change anything.

Eric

remidone, 2023-07-31:
Jul 31 07:39:50 vstallsky allsky[6832]:      ***** Starting AllSky *****
Jul 31 07:39:52 vstallsky allsky[6903]: copy_notification_image.sh: Uploading StartingUp.jpg
Jul 31 07:39:53 vstallsky allsky[6935]: *** upload.sh: ERROR: '/allsky' directory not found!
Jul 31 07:39:53 vstallsky allsky[6944]:  >>> Parameter [config                   ]  Value: [/home/pi/allsky/tmp/capture_args.txt]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [cmd                      ]  Value: [libcamera-still]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [debuglevel               ]  Value: [4]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [takedaytimeimages        ]  Value: [1]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [savedaytimeimages        ]  Value: [0]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [dayautoexposure          ]  Value: [1]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [daymaxautoexposure       ]  Value: [10000]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [dayexposure              ]  Value: [500]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [daymean                  ]  Value: [0.5]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [daymeanthreshold         ]  Value: [0.1]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [daybrightness            ]  Value: [0]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [daydelay                 ]  Value: [10000]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [dayautogain              ]  Value: [0]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [daymaxautogain           ]  Value: [4]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [daygain                  ]  Value: [1]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [daybin                   ]  Value: [1]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [dayawb                   ]  Value: [1]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [daywbr                   ]  Value: [2.5]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [daywbb                   ]  Value: [2]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [dayskipframes            ]  Value: [5]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [daytuningfile            ]  Value: []
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightautoexposure        ]  Value: [0]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightmaxautoexposure     ]  Value: [60000]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightexposure            ]  Value: [60000]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightmean                ]  Value: [0.2]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightmeanthreshold       ]  Value: [0.1]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightbrightness          ]  Value: [0]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightdelay               ]  Value: [5000]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightautogain            ]  Value: [0]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightmaxautogain         ]  Value: [16]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightgain                ]  Value: [6]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightbin                 ]  Value: [1]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightawb                 ]  Value: [0]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightwbr                 ]  Value: [2.5]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightwbb                 ]  Value: [2]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nightskipframes          ]  Value: [1]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [nighttuningfile          ]  Value: []
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [extraargs                ]  Value: []
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [saturation               ]  Value: [1]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [contrast                 ]  Value: [1]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [sharpness                ]  Value: [1]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [width                    ]  Value: [0]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [height                   ]  Value: [0]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [type                     ]  Value: [99]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [quality                  ]  Value: [95]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [filename                 ]  Value: [image.jpg]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [flip                     ]  Value: [0]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [notificationimages       ]  Value: [1]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [consistentdelays         ]  Value: [1]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [timeformat               ]  Value: [%Y%m%d %H:%M:%S]
Jul 31 07:39:53 vstallsky allsky[6944]:      >>> Parameter [latitude                 ]  Value: [+43.71]
[... cut ...]
Jul 31 07:41:19 vstallsky allsky[6944]: Executing sunwait poll exit angle -8.0000 43,00000N 12,00000E > /dev/null
Jul 31 07:41:19 vstallsky allsky[6944]: -----
Jul 31 07:41:19 vstallsky allsky[6944]: STARTING EXPOSURE at: 2023-07-31 07:41:19   @ 1,52 ms
Jul 31 07:41:19 vstallsky allsky[6944]:  > Running: libcamera-still --output '/home/pi/allsky/tmp/image-20230731074119.jpg' --timeout 1 --nopreview --width 4056 --height 3040 --shutter 1520 --awb auto --saturation 1 --contrast 1 --sharpness 1 --brightness 0 --quality 100

EricClaeys, 2023-08-01:
There is a bug when the decimal separator is a comma and not a period.  I will try to fix this for the next point release.
Going forward,  please ATTACH the log file instead of copying/pasting it as I requested.  It is very difficult to read very long files on a phone.
Thanks for pointing out this problem.

EricClaeys, 2023-08-01:
Please try using setting latitude to `+43,71` and longitude to `+12,75`.

remidone, 2023-08-01:
this is the result

[image: Immagine 2023-08-01 131627.png]
You'll find attached allsky.log

EricClaeys, 2023-08-01:
@remidone  GitHub ignores attachments sent via email so you'll need to log in and attach the files.
Eric

remidone, 2023-08-01:
when I try using +43,71 and +12,75 I have this result in logs...
[image]

here it is 
[allsky.log](https://github.com/thomasjacquin/allsky/files/12232009/allsky.log)
allsky.log

EricClaeys, 2023-08-01:
Would you please run:
~~~
sudo systemctl stop allsky
truncate -s 0 /var/log/allsky.log
# Change "Debug Level" in the WebUI to 4 and save settings
~~~
After about a minute, ATTACH `/var/log/allsky.log`?
You can then set the Debug Level back to whatever it was, and use periods in the latitude and longitude.  The bug is truncating everything after the decimal point, which doesn't really matter.

If you attach a copy of your `~/allsky/config/ftp-settings.sh` file (after removing the password), I'll help you fix the `*** upload.sh: ERROR: '/allsky' directory not found` message.

remidone, 2023-08-01:
> After about a minute, ATTACH `/var/log/allsky.log`?

yes

> If you attach a copy of your `~/allsky/config/ftp-settings.sh` file (after removing the password), I'll help you fix the `*** upload.sh: ERROR: '/allsky' directory not found` message.

I haven't completed settings yet because I have some problem to install remote website. (extension modified)
[ftp-settings.txt](https://github.com/thomasjacquin/allsky/files/12232706/ftp-settings.txt)

EricClaeys, 2023-08-01:
Set `WEB_IMAGE_DIR=""` to get rid of that error message.  The WEB_*_DIR variables are only used if you have both a local and remote Website.

Can you attach `/var/log/allsky.log` ?

remidone, 2023-08-02:
Yesterday I attached allsky.log. Was it not good?

remidone, 2023-08-02:
You can find it two messages before

EricClaeys, 2023-08-02:
Fixed in v2023.05.01 point release 3.

closed 2023-08-02 by EricClaeys

remidone, 2023-08-04:
I''m sorry to tell you but I updated this file but nothing changed, same errors.

commit 64de869e (2023-08-02): Update allsky_common.cpp: better lat/long checking

Fixes #2895 and #2891