AllskyTeam/allsky #2904: [BUG]Overlay stops working totally

WVAstro, 2023-08-09:
### Environment

* Camera:  RPi HQ
* OS: Bullseye
* Allsky version: v2023.05.01_02 
     * The newest software includes a file called  ~/allsky/version
     * If it does not exist, run:       grep Software /var/log/allsky.log
* Pi model and amount of memory (4B 8G)

### Bug Description

* Module Overlay and Legacy overlay completely quit displaying on [images]([url](url))
* Can it be reproduced? YES  If so, how? Every reboot results in no overlay
* Did anything change? Yes  Any settings?  Multiple.

*  Files attached 

[configuration.json.txt](https://github.com/thomasjacquin/allsky/files/12297130/configuration.json.txt)
[settings.json.txt](https://github.com/thomasjacquin/allsky/files/12297131/settings.json.txt)
[overlaydebug.txt](https://github.com/thomasjacquin/allsky/files/12297133/overlaydebug.txt)
[remote_configuration.json.txt](https://github.com/thomasjacquin/allsky/files/12297134/remote_configuration.json.txt)
[ftp-settings.sh.txt](https://github.com/thomasjacquin/allsky/files/12297135/ftp-settings.sh.txt)
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/12297136/config.sh.txt)
[allsky.log](https://github.com/thomasjacquin/allsky/files/12297137/allsky.log)

[image]

[image]

### Log / configuration files

Follow the instructions for Reporting Issues in the Wiki, then ATTACH a copy of the file(s) above.
Do NOT copy/paste them into this Issue.

frankhirsch, 2023-08-09:
I think I may have just asked some similar question in the discord channel...
Probably try to disable sun and/or satellite options in the overlay module editor - that does the trick for me at the moment!

Alex-developer, 2023-08-09:
> ### Environment
> * Camera:  RPi HQ
> * OS: Bullseye
> * Allsky version: v2023.05.01_02
>   
>   * The newest software includes a file called  ~/allsky/version
>   * If it does not exist, run:       grep Software /var/log/allsky.log
> * Pi model and amount of memory (4B 8G)
> 
> ### Bug Description
> * Module Overlay and Legacy overlay completely quit displaying on [images](%5Burl%5D(url))
> * Can it be reproduced? YES  If so, how? Every reboot results in no overlay
> * Did anything change? Yes  Any settings?  Multiple.
> * Files attached
> 
> [configuration.json.txt](https://github.com/thomasjacquin/allsky/files/12297130/configuration.json.txt) [settings.json.txt](https://github.com/thomasjacquin/allsky/files/12297131/settings.json.txt) [overlaydebug.txt](https://github.com/thomasjacquin/allsky/files/12297133/overlaydebug.txt) [remote_configuration.json.txt](https://github.com/thomasjacquin/allsky/files/12297134/remote_configuration.json.txt) [ftp-settings.sh.txt](https://github.com/thomasjacquin/allsky/files/12297135/ftp-settings.sh.txt) [config.sh.txt](https://github.com/thomasjacquin/allsky/files/12297136/config.sh.txt) [allsky.log](https://github.com/thomasjacquin/allsky/files/12297137/allsky.log)
> 
> [image] [image]
> ### Log / configuration files
> Follow the instructions for Reporting Issues in the Wiki, then ATTACH a copy of the file(s) above. Do NOT copy/paste them into this Issue.

Thanks for those.

Can you please send me the overlay.json file as well

Thanks

Alex

WVAstro, 2023-08-10:
Unfortunately at this time I cannot. The sd card corrupted after a power outage so I’m currently attempting to recover.  I am also starting a new card from scratch so a lot of development still to go on the second sd card.  When is the next release scheduled?  I’m afraid to incorporate the fields I want to display on the second card until that bug is fixed.

[Non-Working-Screen Shot 2023-08-08 at 8 47 58 PM]<https://user-images.githubusercontent.com/141787442/259266652-3fe27c58-3e9c-4cf9-8e47-019e48163f27.png> [Working Screen Shot 2023-07-10 at 11 39 46 PM] <https://user-images.githubusercontent.com/141787442/259266059-32de0459-6f8e-481a-8d80-076c34696798.png>

Alex-developer, 2023-08-10:
> Unfortunately at this time I cannot. The sd card corrupted after a power outage so I’m currently attempting to recover. I am also starting a new card from scratch so a lot of development still to go [...]

No problem, we are hoping to get he next release out in a few weeks.

In the mean time if you setup the overlay you can just keep a backup copy of the overlay.json file if needed

Cheers

Alex

EricClaeys, 2023-08-12:
@Alex-developer Did you see the ERROR in allsky.log when trying to load satellite data?

Alex-developer, 2023-08-12:
@WVAstro 

@EricClaeys raised a good point that I missed, try clearing all entries from the satellite list. I'll add some more error handling around the downloading of TLE's

Cheers

Alex

WVAstro, 2023-08-15:
I have a rebuild in progress.  I had tried to add the ISS on the overlay previously.  Is there an error in that code?

EricClaeys, 2023-11-04:
@WVAstro, @Alex-developer, any update on this?  Is it fixed in Point Release 4 ?

EricClaeys, 2023-12-09:
No response from users that are normally responsive, so assuming it's fixed.

closed 2023-12-09 by EricClaeys