AllskyTeam/allsky #2909: Support RPi V2 camera

EricClaeys, 2023-08-13:
### Discussed in https://github.com/thomasjacquin/allsky/discussions/2908

Also, `install.sh` should list the RPi camera model, not the current generic "RPi HQ or compatible" message.

~~~
0 : ov5647 [2592x1944] (/base/soc/i2c0mux/i2c@1/ov5647@36)
Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
1296x972 [43.25 fps - (0, 0)/2592x1944 crop]
1920x1080 [30.62 fps - (348, 434)/1928x1080 crop]
2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]

Available controls for 2592x1944 SGBRG10_CSI2P mode:
----------------------------------------------------
AeConstraintMode : [0..3]
AeEnable : [false..true]
AeExposureMode : [0..3]
AeMeteringMode : [0..3]
AnalogueGain : [1.000000..63.937500]
AwbEnable : [false..true]
AwbMode : [0..7]
Brightness : [-1.000000..1.000000]
ColourGains : [0.000000..32.000000]
Contrast : [0.000000..32.000000]
ExposureTime : [130..969249]
ExposureValue : [-8.000000..8.000000]
FrameDurationLimits : [63965..1065021]
NoiseReductionMode : [0..4]
Saturation : [0.000000..32.000000]
ScalerCrop : [(0, 0)/64x64..(0, 0)/2592x1944]
Sharpness : [0.000000..16.000000]
~~~

EricClaeys, 2023-08-18:
Fixed in Point Release 3.

Mr-Groch, 2023-09-01:
Hi, I have some problem with this camera - it should be 2592x1944 but instead allsky keeps capturing 1944x2592 - it is annoying for my to view portrait (vertical) images - how can I rotate it?

EricClaeys, 2023-09-01:
@Mr-Groch.  Arrrrg!  I did it again - mixed up height and width.  Sorry.
You can fix by changing line 141 in ~/allsky/src/ASI_functions.cpp.  Just swap the numbers.
Then do:
~~~
# stop Allsky
cd ~/allsky/src
make capture_RPi
cp capture_RPi ../bin
# restart allsky
~~~

Mr-Groch, 2023-09-01:
Thanks, that helped for portrait orientation!

But still max exposure clipped to 969 ms is problem for me

EricClaeys, 2023-09-01:
@Mr-Groch please run `libcamera-still --list-cameras -v` and let me know what the ExposureTime values are.  They are the min and max times in us.  If you get 969249 for max I will ask the libcamera team why it's not 9 seconds.
Do you have a URL that shows 9 seconds max for the camera (not for the chip)?

Eric

Mr-Groch, 2023-09-01:
```
Available cameras
-----------------
0 : ov5647 [2592x1944] (/base/soc/i2c0mux/i2c@1/ov5647@36)
    Modes: 'SGBRG10_CSI2P' : 640x480 [58.92 fps - (16, 0)/2560x1920 crop]
                             1296x972 [43.25 fps - (0, 0)/2592x1944 crop]
                             1920x1080 [30.62 fps - (348, 434)/1928x1080 crop]
                             2592x1944 [15.63 fps - (0, 0)/2592x1944 crop]

    Available controls for 2592x1944 SGBRG10_CSI2P mode:
    ----------------------------------------------------
    AeConstraintMode : [0..3]
    AeEnable : [false..true]
    AeExposureMode : [0..3]
    AeMeteringMode : [0..3]
    AnalogueGain : [1.000000..63.937500]
    AwbEnable : [false..true]
    AwbMode : [0..7]
    Brightness : [-1.000000..1.000000]
    ColourGains : [0.000000..32.000000]
    Contrast : [0.000000..32.000000]
    ExposureTime : [130..969249]
    ExposureValue : [-8.000000..8.000000]
    FrameDurationLimits : [63965..1065021]
    NoiseReductionMode : [0..4]
    Saturation : [0.000000..32.000000]
    ScalerCrop : [(0, 0)/64x64..(0, 0)/2592x1944]
    Sharpness : [0.000000..16.000000]

```

here https://www.waveshare.com/wiki/RPi_Camera_(G)#libcamera-still you can see max exposure times table:

[image]

Mr-Groch, 2023-09-02:
I've checked manually in daytime taking pictures with 1s and 6s exposures with that camera - it looks that 6s exposure is working:

`libcamera-still -o html/allsky/1s.jpg --shutter 1000000 --gain 1 --awbgains 1,1 --immediate`
[image]

`libcamera-still -o html/allsky/6s.jpg --shutter 6000000 --gain 1 --awbgains 1,1 --immediate`
[image]

EricClaeys, 2023-09-02:
I posted an Issue on the libcamera-apps Github page asking why the camera reports max exposure of .9 sec instead of 6.
In the meantime you can change it by editing ~/allsky/src/ASI_functions.cpp around line 140.  It should be obvious. Then run
~~~
systemctl stop allsky
cd ~/allsky/src
make capture_RPi 
cp capture_RPi  ../bin
~~~
In the WebUI at the very bottom when displaying Advanced options, change `Camera Type` to Refresh and save the changes.

EricClaeys, 2023-09-09:
Can you try taking the 1 and 6 second shots at night, and add this to the command line: `--meta /tmp/m`.  After each picture, run `cat /tmp/m` and let me know what it says.

Mr-Groch, 2023-09-10:
I've repeated earlier cmds with `--metadata`, results:

- for `--shutter 1000000`:
```
{
    "ExposureTime": 999982,
    "FocusFoM": 272,
    "AnalogueGain": 1.000000,
    "ColourCorrectionMatrix": [ 1.974798, -0.621997, -0.352811, -0.353831, 1.554441, -0.200601, 0.006369, -0.720011, 1.713642 ],
    "FrameDuration": 1000112,
    "Lux": 0.611589,
    "AeLocked": false,
    "ColourGains": [ 1.000000, 1.000000 ],
    "DigitalGain": 1.000017,
    "ColourTemperature": 5600,
    "SensorBlackLevels": [ 1024, 1024, 1024, 1024 ],
    "ScalerCrop": "(0, 0)/2592x1944",
    "SensorTimestamp": 12983988312000
}
```
- for `--shutter 6000000`:
```
{
    "ExposureTime": 1064891,
    "FocusFoM": 274,
    "AnalogueGain": 1.000000,
    "ColourCorrectionMatrix": [ 1.974798, -0.621997, -0.352811, -0.353831, 1.554441, -0.200601, 0.006369, -0.720011, 1.713642 ],
    "FrameDuration": 1065021,
    "Lux": 0.574318,
    "AeLocked": false,
    "ColourGains": [ 1.000000, 1.000000 ],
    "DigitalGain": 4.000000,
    "ColourTemperature": 5600,
    "SensorBlackLevels": [ 1024, 1024, 1024, 1024 ],
    "ScalerCrop": "(0, 0)/2592x1944",
    "SensorTimestamp": 13086125263000
}
```

EricClaeys, 2023-09-11:
Can you post the complete command lines you used, and the camera model?  Also, did you time them with a wall clock or the `time` or `timex` command?
It is strange that the 60 second exposure reports ExposureTime as about 10 seconds.
Please also make sure to upgrade the `libcamera-apps` package.

Mr-Groch, 2023-09-17:
Sorry, I can't provide any info anymore - I've changed this camera to IMX378 with built-in 190 lens. RPI V1 camera was disappointing in night conditions...

BTW. I suggest to add IMX378 to list of supported cameras - it is working out of the box, AllSky detects it as RPI HQ camera.

EricClaeys, 2023-09-17:
Does the IMX378 have the same specs as the HQ?  Please run `libcamera-still --list-cameras`.

Mr-Groch, 2023-09-18:
```
pi@allsky:~/allsky $ libcamera-still --list-cameras
Available cameras
-----------------
0 : imx477 [4056x3040] (/base/soc/i2c0mux/i2c@1/imx477@1a)
    Modes: 'SRGGB10_CSI2P' : 1332x990 [120.05 fps - (696, 528)/2664x1980 crop]
           'SRGGB12_CSI2P' : 2028x1080 [50.03 fps - (0, 440)/4056x2160 crop]
                             2028x1520 [40.01 fps - (0, 0)/4056x3040 crop]
                             4056x3040 [10.00 fps - (0, 0)/4056x3040 crop]

```

Camera:
https://www.waveshare.com/imx378-190-12.3mp-camera.htm

EricClaeys, 2023-09-22:
Closing.  V1 camera is supported in current release with a bug fix in next release.  Unclear if anyone will use this for an allsky camera given its very short exposure limit.

closed 2023-09-22 by EricClaeys