AllskyTeam/allsky #2946: [BUG]

khobar95, 2023-08-31:
### Environment

RPi IMX378
Bullseye
v2023.05.01_02
Pi4B 4GB

### Bug Description
I am wanting to add a mask. On my PiZero running software v2023.05.01 I can click "Overlay Editor" and after a bit I get the Overlay Editor. However, on the Pi4B running _02 software, clicking Overlay Editor results in "xxx.xxx.xxx.xxx says A fatal error has occureed loading the application configuration." 

* Can it be reproduced?  If so, how?

Yes. Click on "Overlay Editor".

* Did anything change?  Any settings?  Most issues occur after a change was made.
This is the first time I've tried to add a mask on this system. It's been in operation only about 2-3 weeks. I've tried copying settings from the PiZero which does not have this issue.

* Please include the exact messages or a screenshot.
Clicking Overlay Editor results in "xxx.xxx.xxx.xxx says A fatal error has occureed loading the application configuration."

 ~/allsky/config/config.sh
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/12488331/config.sh.txt)

~/allsky/config/settings.json
[settings.json.txt](https://github.com/thomasjacquin/allsky/files/12488333/settings.json.txt)

### Log / configuration files

EricClaeys, 2023-08-31:
@Alex-developer 
Alex, can you help?

Alex-developer, 2023-08-31:
@khobar95 Thanks for logging it here, its far easier to sort out than on facebook!

This error is thrown when the overlay editor fails to load one of its configuration files and most of the time this is due to one of the files being corrupted.

Please can you zip up the config/overlay folder and upload it here, I can then check the files to see where the corruption is

Many Thanks

Alex

khobar95, 2023-09-01:
Well that's interesting. I went to do as you asked and found there was no overlay folder. There's no modules folder either. However, both of these are on my PiZero.

That might be the problem. :)

EricClaeys, 2023-09-01:
What's in `~/allsky/config` ?

khobar95, 2023-09-01:
cc.json  cc_RPi_HQ.json  config.sh  ftp-settings.sh  options.json  raspap.auth  settings.json  settings_RPi_HQ.json

khobar95, 2023-09-01:
There's a directory called config_repo that has modules and overlay directories. Did something go wrong during installation maybe?

khobar95, 2023-09-01:
No worries. The software itself is working fine as near as I can tell.

EricClaeys, 2023-09-01:
@khobar95 please attach your `~/allsky/config/installation_logs/install.sh.log`.  It will indicate if there was an installation issue.

EricClaeys, 2023-09-01:
Also, do those directories exist in ~/allsky-OLD?

khobar95, 2023-09-02:
HI Eric,

There are no directories un ~/allsky/config.

This was a brand new installation.

khobar95, 2023-09-02:
Since something obviously went wrong on my end and there's a new release, I
went ahead and did an update. Smooth as butter, and at least the overlay
editor comes up now so I suspect all is well now.

EricClaeys, 2023-09-02:
@khobar95 glad it's working now.
At a minimum the allsky/config/installation_logs should have been there since that's the first directory the install script creates.

closed 2023-09-02 by EricClaeys