AllskyTeam/allsky #2949: [BUG] Upload problem,directory not found , thanks.

northwolfwu, 2023-09-02:
### Environment

* Camera:  ZWO  224MC
* OS: Bullseye
* Allsky version:
     * v2023.05.01_02

* Pi model and amount of memory (4B, 8 GB Memory, 128G SD Card)

In the website, I use apache in ubuntu OS, the root website directory is /var/www/html, and I make a subdirectroy allsky to hold the websites.
http://stargazer.org.cn/allsky/

### Bug Description
I already install allsky and remote websites, but it can't upload the image to the websites.

Use the command: sudo cat /var/log/allsky.log
Sep  2 09:25:23 allskycam1 allsky[1154]: === Starting daytime capture ===
Sep  2 09:25:23 allskycam1 allsky[1154]: ==========
Sep  2 09:25:23 allskycam1 allsky[1154]: STARTING EXPOSURE at: 2023-09-02 09:25:23   @ 10.00 ms (0.01 sec)
Sep  2 09:25:24 allskycam1 allsky[1154]: STARTING EXPOSURE at: 2023-09-02 09:25:24   @ 36.10 ms (0.04 sec)
Sep  2 09:25:28 allskycam1 allsky[1278]: *** upload.sh: ERROR: '/var/www/html/allsky' directory not found!
Sep  2 09:26:09 allskycam1 allsky[1154]: STARTING EXPOSURE at: 2023-09-02 09:26:09   @ 31.05 ms (0.03 sec)
Sep  2 09:26:11 allskycam1 allsky[1711]: *** upload.sh: ERROR: '/var/www/html/allsky' directory not found!

but I use the followed command and it sounds successfully.
admin@allskycam1:~/allsky2/allsky/scripts $ ./upload.sh image.jpg /var/www/html/allsky image1.jpg
log:
Upload output from 'image.jpg:
   ---- Running connect program (ssh -a -x -s -l root -p 22 stargazer.org.cn sftp)
---> sending a packet, length=5, type=1(INIT), id=0
root@stargazer.org.cn's password: XXXX
<--- got a packet, length=179, type=2(VERSION), id=0
---- protocol version set to 3
---> sending a packet, length=10, type=16(REALPATH), id=1
<--- got a packet, length=31, type=104(NAME), id=1
---- home set to /root
---- checking directory `/root'
---> sending a packet, length=14, type=17(STAT), id=2
---> sending a packet, length=16, type=17(STAT), id=3
<--- got a packet, length=37, type=105(ATTRS), id=2
<--- got a packet, length=37, type=105(ATTRS), id=3
---- checking directory `/var/www/html/allsky'
---> sending a packet, length=29, type=17(STAT), id=4
---> sending a packet, length=31, type=17(STAT), id=5
<--- got a packet, length=37, type=105(ATTRS), id=4
<--- got a packet, length=37, type=105(ATTRS), id=5
---- path on wire is `/var/www/html/allsky'
---> sending a packet, length=29, type=11(OPENDIR), id=6
<--- got a packet, length=13, type=102(HANDLE), id=6
---- got file handle 00000000 (4)
---> sending a packet, length=13, type=12(READDIR), id=7
<--- got a packet, length=3579, type=104(NAME), id=7
---- file name count=30
---> sending a packet, length=13, type=12(READDIR), id=8
---> sending a packet, length=13, type=12(READDIR), id=9
<--- got a packet, length=28, type=101(STATUS), id=8
---- status code=1(EOF), message=End of file
---- eof
---> sending a packet, length=13, type=4(CLOSE), id=10
<--- got a packet, length=28, type=101(STATUS), id=9
---- status code=1(EOF), message=End of file
<--- got a packet, length=24, type=101(STATUS), id=10
---- status code=0(OK), message=Success
---- path on wire is `/var/www/html/allsky/x-669'
---> sending a packet, length=43, type=3(OPEN), id=11
<--- got a packet, length=13, type=102(HANDLE), id=11
---- got file handle 00000000 (4)
---> sending a packet, length=32793, type=6(WRITE), id=12
---> sending a packet, length=32793, type=6(WRITE), id=13
---> sending a packet, length=32793, type=6(WRITE), id=14
---> sending a packet, length=32793, type=6(WRITE), id=15
---> sending a packet, length=32793, type=6(WRITE), id=16
---> sending a packet, length=32793, type=6(WRITE), id=17
---> sending a packet, length=32793, type=6(WRITE), id=18
---> sending a packet, length=32793, type=6(WRITE), id=19
---> sending a packet, length=32793, type=6(WRITE), id=20
---> sending a packet, length=32793, type=6(WRITE), id=21
---> sending a packet, length=23656, type=6(WRITE), id=22
---> sending a packet, length=33, type=10(FSETSTAT), id=23
---> sending a packet, length=13, type=4(CLOSE), id=24
<--- got a packet, length=24, type=101(STATUS), id=12
---- status code=0(OK), message=Success
<--- got a packet, length=24, type=101(STATUS), id=13
---- status code=0(OK), message=Success
<--- got a packet, length=24, type=101(STATUS), id=14
---- status code=0(OK), message=Success
<--- got a packet, length=24, type=101(STATUS), id=15
---- status code=0(OK), message=Success
<--- got a packet, length=24, type=101(STATUS), id=16
---- status code=0(OK), message=Success
<--- got a packet, length=24, type=101(STATUS), id=17
---- status code=0(OK), message=Success
<--- got a packet, length=24, type=101(STATUS), id=18
---- status code=0(OK), message=Success
<--- got a packet, length=24, type=101(STATUS), id=19
---- status code=0(OK), message=Success
<--- got a packet, length=24, type=101(STATUS), id=20
---- status code=0(OK), message=Success
<--- got a packet, length=24, type=101(STATUS), id=21
---- status code=0(OK), message=Success
<--- got a packet, length=24, type=101(STATUS), id=22
---- status code=0(OK), message=Success
<--- got a packet, length=24, type=101(STATUS), id=23
---- status code=0(OK), message=Success
<--- got a packet, length=24, type=101(STATUS), id=24
---- status code=0(OK), message=Success
---- path on wire is `/var/www/html/allsky/image1.jpg'
---> sending a packet, length=40, type=13(REMOVE), id=25
<--- got a packet, length=24, type=101(STATUS), id=25
---- status code=0(OK), message=Success
---- path on wire is `/var/www/html/allsky'
---> sending a packet, length=29, type=11(OPENDIR), id=26
<--- got a packet, length=13, type=102(HANDLE), id=26
---- got file handle 00000000 (4)
---> sending a packet, length=13, type=12(READDIR), id=27
<--- got a packet, length=3569, type=104(NAME), id=27
---- file name count=30
---> sending a packet, length=13, type=12(READDIR), id=28
---> sending a packet, length=13, type=12(READDIR), id=29
<--- got a packet, length=28, type=101(STATUS), id=28
---- status code=1(EOF), message=End of file
---- eof
---> sending a packet, length=13, type=4(CLOSE), id=30
<--- got a packet, length=28, type=101(STATUS), id=29
---- status code=1(EOF), message=End of file
<--- got a packet, length=24, type=101(STATUS), id=30
---- status code=0(OK), message=Success
---- path on wire is `/var/www/html/allsky/x-669'
---- path on wire is `/var/www/html/allsky/image1.jpg'
---> sending a packet, length=70, type=18(RENAME), id=31
<--- got a packet, length=24, type=101(STATUS), id=31
---- status code=0(OK), message=Success
---- Disconnecting

[settings.json.txt](https://github.com/thomasjacquin/allsky/files/12502453/settings.json.txt)
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/12502456/config.sh.txt)
[ftp-settings.sh.txt](https://github.com/thomasjacquin/allsky/files/12502457/ftp-settings.sh.txt)

### Log / configuration files
[allsky.log.txt](https://github.com/thomasjacquin/allsky/files/12502459/allsky.log.txt)

EricClaeys, 2023-09-02:
@northwolfwu    Please take a look at the `Settings -> Allsky Website`,   in particular the bottom of the "ftp-settings.sh" section.  It will tell you what settings to use.
If you still have problems, the `Troubleshooting -> Allsky Website` page will tell you how to debug and fix the problem.

northwolfwu, 2023-09-02:
@EricClaeys Thank you very much. I change the ftp-settings.sh as followed, and it works greatly.

	# How will files be uploaded?
PROTOCOL="sftp"

	# The directory where the current image should be copied to.
IMAGE_DIR="/var/www/html/allsky"
WEB_IMAGE_DIR=""

	# The directory where the timelapse video should be uploaded to.
VIDEOS_DIR="/var/www/html/allsky/videos"
VIDEOS_DESTINATION_NAME=""
WEB_VIDEOS_DIR=""

	# The directory where the keogram image should be copied to.
KEOGRAM_DIR="/var/www/html/allsky/keograms"
KEOGRAM_DESTINATION_NAME=""
WEB_KEOGRAM_DIR=""

	# The directory where the startrails image should be copied to.
STARTRAILS_DIR="/var/www/html/allsky/startrails"
STARTRAILS_DESTINATION_NAME=""
WEB_STARTRAILS_DIR=""

EricClaeys, 2023-09-02:
Glad it's working!

closed 2023-09-02 by EricClaeys