AllskyTeam/allsky #2951: [BUG] New allsky camera doesn't appear on map...

Mr-Groch, 2023-09-03:
### Environment

* Camera:  RPi v1 (ov5647)
* OS: Bullseye
* Allsky version: v2023.05.01_03
* Pi 3b+ 1GB

### Bug Description

My Allsky is online from yesterday with Show on Map option enabled, and it doesn't appear on https://www.thomasjacquin.com/allsky-map/
  
```json
{
    "takeDaytimeImages": "1",
    "saveDaytimeImages": "1",
    "dayautoexposure": "1",
    "daymaxautoexposure": "500",
    "dayexposure": "1",
    "daymean": "0.5",
    "daymeanthreshold": "0.1",
    "daybrightness": "0",
    "daydelay": "30000",
    "dayautogain": "0",
    "daymaxautogain": "2",
    "daygain": "1",
    "daybin": "1",
    "dayawb": "1",
    "daywbr": "0",
    "daywbb": "0",
    "dayskipframes": "3",
    "dayTuningFile": "",
    "nightautoexposure": "1",
    "nightmaxautoexposure": "6000",
    "nightexposure": "500",
    "nightmean": "0.2",
    "nightmeanthreshold": "0.1",
    "nightbrightness": "0",
    "nightdelay": "30000",
    "nightautogain": "1",
    "nightmaxautogain": "20",
    "nightgain": "2",
    "nightbin": "1",
    "nightawb": "1",
    "nightwbr": "0",
    "nightwbb": "0",
    "nightskipframes": "2",
    "nightTuningFile": "",
    "config": "[none]",
    "extraArgs": "--post-process-file /home/pi/allsky/assets/libcamera-nightmode.json",
    "saturation": "1",
    "contrast": "1",
    "sharpness": "1",
    "width": "0",
    "height": "0",
    "type": "99",
    "quality": "90",
    "filename": "image.jpg",
    "flip": "0",
    "notificationimages": "1",
    "consistentDelays": "1",
    "timeformat": "%Y%m%d %H:%M:%S",
    "latitude": "50.3N",
    "longitude": "18.7E",
    "angle": "-6",
    "takeDarkFrames": "0",
    "useDarkFrames": "0",
    "locale": "pl_PL.UTF-8",
    "debuglevel": "0",
    "useLogin": "1",
    "alwaysshowadvanced": "1",
    "overlayMethod": "1",
    "showTime": "1",
    "temptype": "C",
    "showExposure": "1",
    "showGain": "1",
    "showBrightness": "0",
    "showMean": "0",
    "showFocus": "0",
    "text": "",
    "extratext": "",
    "extratextage": "0",
    "textlineheight": "30",
    "textx": "15",
    "texty": "35",
    "fontname": "0",
    "fontcolor": "255 0 0",
    "smallfontcolor": "0 0 255",
    "fonttype": "0",
    "fontsize": "7",
    "fontline": "1",
    "outlinefont": "0",
    "displaySettings": "0",
    "showonmap": "1",
    "websiteurl": "http://allsky.groch.cloud/allsky/",
    "imageurl": "http://allsky.groch.cloud/current/tmp/image.jpg",
    "location": "Gliwice, \u015aL, PL",
    "owner": "Damian Grocholski",
    "camera": "ov5647",
    "lens": "160\u00b0",
    "computer": "Raspberry Pi 3 B+, 1GB",
    "cameraType": "RPi",
    "cameraModel": "Version_1",
    "lastChanged": "2023-09-03 14:54:45"
}
```

I've also tried manually to run `scripts/postToMap.sh --debug`

```bash
postToMap.sh: Uploading map data.

Executing:
curl --silent -i -H 'Accept: application/json' -H 'Content-Type:application/json' --data '{
"location": "Gliwice, ŚL, PL",
"owner": "Damian Grocholski",
"latitude": "50.3N",
"longitude": "18.7E",
"website_url": "http://allsky.groch.cloud/allsky/",
"image_url": "http://allsky.groch.cloud/current/tmp/image.jpg",
"camera": "ov5647",
"lens": "160°",
"computer": "Raspberry Pi 3 B+, 1GB",
"allsky_version": "v2023.05.01_03",
"machine_id": "7f8906258fda49ba8854bad56a1107a7"
}' 'https://www.thomasjacquin.com/allsky-map/postToMap.php'

Returned:
HTTP/2 200
date: Sun, 03 Sep 2023 12:55:20 GMT
server: Apache
x-powered-by: PHP/8.0.12
content-type: text/html; charset=UTF-8
via: 1.1 varnish (Varnish/6.3), 1.1 varnish (Varnish/6.3)
vary: Accept-Encoding
x-cache-hits: 0
x-cache: MISS
accept-ranges: bytes
content-length: 9
age: 0

INSERTED.

postToMap.sh: Map data INSERTED.
```

### Log / configuration files

No problem in logs - postToMapsuccesfully

EricClaeys, 2023-09-03:
The only thing I can think of is that the two non-ASCII characters may be causing a problem. One is in the "location" and the other is the degree symbol in the "lens".
Try removing them in the WebUI.  Allsky should automatically upload the changes to the map.

Mr-Groch, 2023-09-03:
Thanks, just removing Ś characker from location was enough. Removed it and my allsky appeared on map. But I think that non-ascii characters, specially in location should be allowed - many location names, not only in Poland has specific local chars (unicode)

EricClaeys, 2023-09-03:
I agree unicorn should be allowed.  I will ask Thomas about it.

thomasjacquin, 2023-09-05:
I have replaced the character set from `latin1_swedish_ci` to `utf8mb4_unicode_ci `. You should now be able to post data with unicode characters. Let me know if it still doesn't work.

Mr-Groch, 2023-09-05:
I've bringed back Ś character to my location and force updated with `postToMap.sh --force` - I can see on AllSky map my full new location name with Ś character - so it works, thanks!

EricClaeys, 2023-09-05:
Thanks @thomasjacquin !!

closed 2023-09-05 by EricClaeys