AllskyTeam/allsky #2952: [BUG] /var/log/allsky.log fills up because of missing stroke color in overlays

astroimagineer, 2023-09-03:
### Environment

* Camera:  ZWO  178
* OS: Bullseye
* Allsky version: v2023.05.01_02
 * Pi 4, 4GB

### Bug Description

Taking a look at /var/log/allsky.log I find that it fills up quickly with the following (this is an excerpt);
~~~
Sep  3 18:12:05 dictallsky allsky[28498]: ERROR: The colour for field 'Exposure: ${sEXPOSURE}' is empty - Defaulting to white.
Sep  3 18:12:05 dictallsky allsky[28498]: ERROR: The colour for field 'Gain: ${GAIN}' is empty - Defaulting to white.
Sep  3 18:12:05 dictallsky allsky[28498]: ERROR: The colour for field 'Sensor temp: ${TEMPERATURE_C} C' is empty - Defaulting to white.
Sep  3 18:12:05 dictallsky allsky[28498]: ERROR: The colour for field 'CAMERA: ${CAMERA_TYPE} ${CAMERA_MODEL}' is empty - Defaulting to white.
~~~

Going into the overlay editor, editing any field and setting the **stroke colour** to anything but the empty default solves the reporting of the missing colour in the logfile and saves many MBs of data each day.

EricClaeys, 2023-09-03:
This issue and several others were fixed in point release 3 which we released last week.
You can upgrade like any other release and it will be fairly quick and save all your current settings.

astroimagineer, 2023-09-03:
I'm sorry I didn't notice this. Thanks! I'll update.

astroimagineer, 2023-09-03:
Thanks again

closed 2023-09-03 by astroimagineer