AllskyTeam/allsky #2955: Allsky Map problem:   ...current/tmp/image.jpg' does not appear to be a valid image.

Vejbymats, 2023-09-04:
### Discussed in https://github.com/thomasjacquin/allsky/discussions/2953

<div type='discussions-op-text'>

<sup>Originally posted by **Vejbymats** September  4, 2023</sup>
When adding (Show on Map) my site to the Allsky Map, I get the message:
ERROR: Image URL 'http://spacecam.worse-than.tv/current/tmp/image.jpg' does not appear to be a valid image.
It looks fine to me, and the image is reachable from "outside"
I do not know what is wrong.
Settings:
Version: v2023.05.01_03     [Local Website: v2023.05.01] (Same error in previous versions to)
Camera: ZWO ASI678MC 
Website URL: http://spacecam.worse-than.tv/allsky 
No binning, no resizing

What have I missed?
</div>

EricClaeys, 2023-09-05:
@Vejbymats   You are not using the standard remote Allsky Website so it's not working.  You are using a `<frame>` to point to the true location (http://nr10.homeip.net:8001/allsky) so the first call to retrieve your image says the result is "text" (which it is - it's a web page) and postToMap.sh fails since it's looking for a return type of "image".

There are a couple ways to fix this:
1. Use `http://nr10.homeip.net:8001/allsky` as your weburl and `http://nr10.homeip.net:8001/current/tmp/image.jpg` as the imageurl.
2. If you want to use the nicer "spacecam.worse-than.tv" URL, then I suggest installing a remote Allsky Website there - see the installation instructions on how.  That also has the advantage of not needing to open your local network to allow people to access your Pi, which is very dangerous unless you are VERY familiar with how to do it.

Let me know which route you take.

Vejbymats, 2023-09-05:
Thanks Eric!
That explains it!

Unfortunately another error showed trying the direct link:

ERROR: Website URL 'http://nr10.homeip.net/allsky' does not appear to be a valid web site.

Same error both with port 80 and the other with port 8001.
Both ports are open and redirected to local Pi in my firewall.

Vejbymats, 2023-09-05:
OK, when adding .php to the web adress it worked.

"http://nr10.homeip.net:8001/allsky" is not accepted by Allsky but
"http://nr10.homeip.net:8001/allsky.php" sort of works so that Allsky is accepting the URL and put my cam on the map,
But the link on the map does not work with the extra .php

Hmmm...

EricClaeys, 2023-09-05:
In the morning I'll give you a test version of postToMap.sh that outputs what the server is returning.
It should be failing on "http://nr10.homeip.net:8001/allsky.php" since that URL doesn't even exist.  Interesting.

Vejbymats, 2023-09-05:
Yes, it does the opposite now:
*** postToMap.sh: ERROR: Website URL 'http://nr10.homeip.net/allsky' does not appear to be a valid web site.

Ok, I'll wait for the file.

EricClaeys, 2023-09-08:
@Vejbymats,
copy this file to `/tmp` on your Pi, then run:
~~~
cd /tmp
mv postData.sh.txt  postData.sh
chmod 775 postData.sh
./postData.sh --force --debug
~~~
Then post the results here.

[postData.sh.txt](https://github.com/thomasjacquin/allsky/files/12555878/postData.sh.txt)

Vejbymats, 2023-09-08:
Yeah, that was funny?
./postToMap.sh --force --debug
*** postToMap.sh: ERROR: Image URL 'null' must begin with 'http:' or
'https:'.
ERROR: Website URL 'null' must begin with 'http:' or 'https:'.

Same answer both with the correct links and with the extra .php

EricClaeys, 2023-09-08:
Sorry, I sent you the wrong script to run.  Please run those commands with THIS file.

[postToMap.sh.txt](https://github.com/thomasjacquin/allsky/files/12558111/postToMap.sh.txt)

Vejbymats, 2023-09-08:
Here is the result:
./postToMap.sh --force --debug

check_URL() RET=0:
HTTP/1.1 301 Moved Permanently
Location: /allsky/
X-Content-Type-Options: nosniff
Cache-Control: max-age=31536000, immutable
Date: Fri, 08 Sep 2023 10:17:07 GMT
Server: Allsky
.

check_URL() RET=0:
HTTP/1.1 200 OK
Content-Type: image/jpeg
Accept-Ranges: bytes
ETag: "1949126931"
Last-Modified: Fri, 08 Sep 2023 10:17:06 GMT
Content-Length: 2048465
X-Content-Type-Options: nosniff
Cache-Control: no-cache, proxy-revalidate
Date: Fri, 08 Sep 2023 10:17:08 GMT
Server: Allsky
.

*** postToMap.sh: ERROR: Website URL 'http://nr10.homeip.net/allsky' does
not appear to be a valid web site.

EricClaeys, 2023-09-08:
I got it to work using `http://nr10.homeip.net/allsky/` and `http://nr10.homeip.net/current/tmp/image.jpg`.

I highly suggest either using "https" on your Pi or moving your Website to a hosted server.  Using "http" is insecure.

Vejbymats, 2023-09-08:
I think I'll do a complete new reinstall and see how that works out.

And I know about http, but I got stuck on this before adding certificate or
web host.
I like it to work locally before going to next step....

EricClaeys, 2023-09-09:
Not sure what a complete new reinstall will do since the problem appears to be with the url settings, not with Allsky itself.
Did you try using the URLs I used?

Vejbymats, 2023-09-09:
I did a fresh install anyway, just to see.
Anyway, now Allsky is accepting the URL*s, and now Allsky-Map it updated.

Thanks!

Next step is to fix the SSL...

Vejbymats, 2023-09-09:
And now also SSL is in place. 
All is well from my side!

Thanks Eric for your help!

closed 2023-09-09 by EricClaeys