AllskyTeam/allsky #2964: [BUG] Openweather module disables itself

Rapid1981, 2023-09-06:
I'm using the Openweather module for over 2 weeks now and in the last few days, it gets disabled on it's own. 
In the module manager, the `Enabled` checkbox unchecks itself every few days.

I enabled it again, save the settings, then after a few days (2-3) it's disabled again and it happens when the Allsky goes from DAY to NIGHT. 

Camera:  RPi HQ
OS: Bullseye
Allsky version:  v2023.05.01_02
Pi model: Model 4B Revision 1.2 (4 GB)

### configuration files
[config.sh.txt](https://github.com/thomasjacquin/allsky/files/12541032/config.sh.txt)
[settings.json.txt](https://github.com/thomasjacquin/allsky/files/12541033/settings.json.txt)

EricClaeys, 2023-09-06:
@Alex-developer, any ideas?

Alex-developer, 2023-09-06:
> I'm using the Openweather module for over 2 weeks now and in the last few days, it gets disabled on it's own. In the module manager, the `Enabled` checkbox unchecks itself every few days.
>
> I enabled [...]

This will be down to connection issues to the OW API. If the watchdog is enabled and a module takes longer than the 'Module Max Timer' to run it will be disabled.

There are two options

1) In the module manager increase the 'Module Max Time'
2) Disable the watchdog in the module manager

@EricClaeys I wonder if there is some mileage in adding a system to try and recover modules that are automatically disabled. Normally these are transient errors and I could add a system that periodically retries modules that have been automatically disabled.

EricClaeys, 2023-09-06:
@Alex-developer, Anything that helps avoid issues the user sees would be good.  In this case, wouldn't the ideal solution to be to determine why the modules are automatically disabled and fix that?  Or is it that the modules are disabled because they are taking too much time?  In that case, perhaps a message in the WebUI (where the other messages are) that states "module xyz took too much time and was disabled.  See xyz for more info" would help.

Alex-developer, 2023-09-07:
> @Alex-developer, Anything that helps avoid issues the user sees would be good. In this case, wouldn't the ideal solution to be to determine why the modules are automatically disabled and fix that? [...]

Eric

Its purely down to time, the flow manager has no idea why the module took too long and it would almost impossible to determine why. For example if a module tried to run whilst a timelapse was being generated it may run much slower or if a module uses an eternal API, like OW, and the API is running slow ....

It is already available in the UI but you have to go looking for it, its not obvious.

I'll have a think about a better way to handle this

closed 2023-09-09 by Alex-developer