AllskyTeam/allsky #2970: [BUG]

Jonk2, 2023-09-13:
### Environment

* Camera:  ZWO  or  RPi - include the model ZWO 178MC
* OS:  Bullseye 
* Allsky version:
     * The newest software includes a file called  ~/allsky/version
     * If it does not exist, run:       grep Software /var/log/allsky.log
* Pi model and amount of memory: Rpi 4B 8GB

### Bug Description

Fresh git clone today on a Pi that previously did not have allsky installed.

This is my backup PI I use for testing with the latest Bullseye installed.

The installer fails at saving the camera settings.

### Log / configuration files

[image]
[allsky.log](https://github.com/thomasjacquin/allsky/files/12602040/allsky.log)

EricClaeys, 2023-09-13:
@Jonk2   Jon,
Long time no talk.  Hope you're doing well!

Would you please attach `~/allsky/config/installation_logs/install.sh.log` ?  It will give me a better idea of why it couldn't save the capabilities.

Did you install the raspistill package, or did it come on Bullseye?

Also, would you please rename "allsky" to "allsky-SAVED", then do another `git`, then in the new `allsky/install.sh`, remove the `echo` on lines 375 and 376, and try the installation again?  The two white lines in the screenshot near the top should have been executed but I forgot to remove the "echo" so they were displayed instead.  It's possible that's what caused the problem.

EricClaeys, 2023-09-14:
@Jonk2, Jon,
Just remove the word `echo` from the lines so the commands are executed.  `echo` simply outputs whatever else is on the command line.  It doesn't execute anything.  I'm pretty sure I had it like that for testing and forgot to fix it.
Thanks.

Jonk2, 2023-09-14:
@EricClaeys 
Raspistill in the /usr/bin folder has a date of 22/03/23 which is when I installed a fresh OS, so it wasn't me manually.

The installation log from the attempt yesterday is attached which will tell you where the issue is.

You will also notice that lighttpd installation failed - I have a weird issue with this PI, upon booting it up, sometimes it grinds to a halt after a couple of minutes, then sorts itself out. I moved the PI back outside, booted it and started a fresh allsky clone - this is probably why it failed, just bad timing.

It worked on the 2nd attempt though. I think I tried to install allsky 2 or 3 more times last night before posting here, just to make sure.

Using analyze-blame, I can sometimes see up to 3 minutes have elapsed sometimes after a reboot, I have no idea why (possible DNS checking issue I've read) but I digress...

I've re-cloned into a fresh folder and removed echo from the 2 lines mentioned above:

[image]

Upon running the new install script...

[image]

The same error appears in the new installation log as the one aattached, i.e.

`Unable to create '/home/pi/allsky/config/options.json' and '/home/pi/allsky/config/settings.json' files.
[0m, RET=255:
Fatal error: Uncaught Error: Call to undefined function json_decode() in /home/pi/allsky/html/includes/functions.php:56
Stack trace:
#0 /home/pi/allsky/html/includes/createAllskyOptions.php(291): get_decoded_json_file('/home/pi/allsky...', true, 'ERROR: Unable t...')
#1 {main}
  thrown in /home/pi/allsky/html/includes/functions.php on line 56`

Let me know what you want me to try next and I'll get on it.

[install.sh.log](https://github.com/thomasjacquin/allsky/files/12612821/install.sh.log)

EricClaeys, 2023-09-15:
Would you please attach `/home/pi/allsky/config/installation_logs/lighttpd.install.log` ?  It should say why lighttpd installation failed.  I'm pretty sure that's why you are getting the `undefined function json_decode()` error - that function wasn't installed because lighttpd installation failed.

EricClaeys, 2023-09-15:
Did you exit the first installation while lighttpd was being installed?  Another installation started after the first "Installing the web server" message, and that's where the lighttpd installation failed.

~~~
* STARTING INSTALLATON AT Wed 13 Sep 20:59:51 BST 2023.

* No prior Allsky found.
* ZWO camera found.
* Using the 'master' branch.
* Displaying notification image 'InstallationInProgress.jpg'
* CURRENT_LOCALE=en_GB.UTF-8, TEMP_LOCALE=[[LANG=en_GB.UTF-8 ]]
* Using ZWO camera.
* Using current hostname of 'RORPI'.
* RAM_SIZE=8192, SUGGESTED_SWAP_SIZE=0.
* Size of current swap (2048 MB) is sufficient; no change needed.
* /home/pi/allsky/tmp is now in memory.
* Installing the web server.
* STARTING INSTALLATON AT Wed 13 Sep 21:17:50 BST 2023.

* No prior Allsky found.
* ZWO camera found.
...
* Installing the web server.
*** ERROR: 
**********
lighttpd installation failed
**********

* The full log file is in /home/pi/allsky/config/installation_logs/lighttpd.install.log
The end of the file is:
* Displaying notification image 'InstallationFailed.jpg'

* ENDING INSTALLATON AT Wed 13 Sep 21:18:00 BST 2023.
~~~

Jonk2, 2023-09-15:
> Did you exit the first installation while lighttpd was being installed? Another installation started after the first "Installing the web server" message, and that's where the lighttpd installation failed.

No, I restarted or rather retried install.sh straight after it failed, that was it. It did work the 2nd or 3rd time (so it says in the install log? )....I mention the PI booting delay above which may or may not be the issue. 
When I reboot it, sometimes even an ftp transfer fails after 20s or so, I have to wait a few minutes for it to work so that's another reason to suspect that. That isn't an allsky issue, I guess I just have to wait a few minutes after a reboot before I do anything manually.

The lighttpd install log is attached. It appears to confirm it is installed correctly.

[lighttpd.install.log](https://github.com/thomasjacquin/allsky/files/12623624/lighttpd.install.log)

EricClaeys, 2023-09-15:
Ok, canceling the installation during lighttpd installation explains why the second installation started.  Unfortunately because you tried again and it successfully installed lighttpd, the log file only shows the successfull attempt so I don't know why it failed the 2nd time.
I should change the installation so it APPENDS to the log files instead of truncating them.

I have never had to wait after a reboot, but I can understand why that could happen.  Luckily you shouldn't need to reboot often.

Is everything working now?

Jonk2, 2023-09-16:
I've deleted the previous allsky folder and re-cloned.

Ran the installation script, and the same error occurs:

[image]
[install.sh.log](https://github.com/thomasjacquin/allsky/files/12641303/install.sh.log)

EricClaeys, 2023-09-18:
I suspect that the initial install of lighttpd failed and was never fully fixed in subsequent installs, so let's try to uninstall it and start over.
Would you please run the following:
~~~
sudo systemctl stop hostapd
sudo systemctl stop lighttpd
sudo apt-get --assume-yes purge lighttpd php-cgi php-gd hostapd dnsmasq avahi-daemon
cd ~/allsky
mv  config  config-SAVED
./install.sh
~~~
This will remove the web-related packages and then run the installation again.  If you previously spent time updating your settings, you can simply copy the config-SAVED/setting.json to config after the installation completes.

If there are any problems, please attach ~/allsky/config/installation_logs/install.sh.log and any log files for packages that fail (the log file name will be in the error message).  Do not re-run the installation or do anything else.

Thanks - Eric

Jonk2, 2023-09-18:
@EricClaeys followed your instructions, unfortunately with the same result.

Here's a copy n paste of the console, the log is attached. It looks to be the same error.

**Linux RORPI 6.1.21-v8+ #1642 SMP PREEMPT Mon Apr  3 17:24:16 BST 2023 aarch64

The programs included with the Debian GNU/Linux system are free software;
the exact distribution terms for each program are described in the
individual files in /usr/share/doc/*/copyright.

Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
permitted by applicable law.
Last login: Sun Sep 17 21:39:15 2023 from 192.168.0.2

Wi-Fi is currently blocked by rfkill.
Use raspi-config to set the country before use.

pi@RORPI:~ $ sudo systemctl stop hostapd
pi@RORPI:~ $ sudo systemctl stop lighttpd
pi@RORPI:~ $ sudo apt-get --assume-yes purge lighttpd php-cgi php-gd hostapd dnsmasq avahi-daemon
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
The following packages were automatically installed and are no longer required:
  bind9-host bind9-libs libavahi-core7 libdaemon0 libfstrm0 libmaxminddb0 libprotobuf-c1
  libuv1 php-common php7.4-cgi php7.4-cli php7.4-common php7.4-gd php7.4-json
  php7.4-opcache php7.4-readline spawn-fcgi
Use 'sudo apt autoremove' to remove them.
The following packages will be REMOVED:
  avahi-daemon* dnsmasq* hostapd* ipp-usb* libnss-mdns* lighttpd* lighttpd-mod-deflate*
  lighttpd-mod-openssl* php-cgi* php-gd*
0 upgraded, 0 newly installed, 10 to remove and 0 not upgraded.
After this operation, 8,167 kB disk space will be freed.
(Reading database ... 111826 files and directories currently installed.)
Removing libnss-mdns:armhf (0.14.1-2) ...
Removing ipp-usb (0.9.17-3) ...
Removing avahi-daemon (0.8-5+deb11u2) ...
Created symlink /run/systemd/system/avahi-daemon.service → /dev/null.
Removed /run/systemd/system/avahi-daemon.service.
Removing dnsmasq (2.85-1) ...
Removing hostapd (2:2.9.0-21) ...
Removing lighttpd-mod-openssl (1.4.59-1+deb11u2) ...
Removing lighttpd-mod-deflate (1.4.59-1+deb11u2) ...
Removing lighttpd (1.4.59-1+deb11u2) ...
Removing php-cgi (2:7.4+76) ...
update-alternatives: using /usr/bin/php-cgi7.4 to provide /usr/bin/php-cgi (php-cgi) in auto mode
update-alternatives: using /usr/lib/cgi-bin/php7.4 to provide /usr/lib/cgi-bin/php (php-cgi-bin) in auto mode
Removing php-gd (2:7.4+76) ...
Processing triggers for man-db (2.9.4-2) ...
Processing triggers for dbus (1.12.24-0+deb11u1) ...
Processing triggers for libc-bin (2.31-13+rpt2+rpi1+deb11u5) ...
(Reading database ... 111646 files and directories currently installed.)
Purging configuration files for lighttpd (1.4.59-1+deb11u2) ...
removing dangling symlink /etc/lighttpd/conf-enabled/10-fastcgi.conf ...
removing dangling symlink /etc/lighttpd/conf-enabled/15-fastcgi-php.conf ...
dpkg: warning: while removing lighttpd, directory '/etc/lighttpd/conf-enabled' not empty so not removed
Purging configuration files for ipp-usb (0.9.17-3) ...
Purging configuration files for lighttpd-mod-openssl (1.4.59-1+deb11u2) ...
Purging configuration files for dnsmasq (2.85-1) ...
dpkg: warning: while removing dnsmasq, directory '/usr/lib/resolvconf' not empty so not removed
Purging configuration files for hostapd (2:2.9.0-21) ...
Removed /etc/systemd/system/hostapd.service.
Purging configuration files for avahi-daemon (0.8-5+deb11u2) ...
rmdir: failed to remove '/run/avahi-daemon': Directory not empty
Purging configuration files for libnss-mdns:armhf (0.14.1-2) ...
libnss-mdns.postrm: Checking NSS setup...
libnss-mdns.postrm: Removing mdns from NSS setup
Purging configuration files for lighttpd-mod-deflate (1.4.59-1+deb11u2) ...
Processing triggers for dbus (1.12.24-0+deb11u1) ...
pi@RORPI:~ $ cd ~/allsky
pi@RORPI:~/allsky $ mv  config  config-SAVED
pi@RORPI:~/allsky $ ./install.sh

****************************************
*** Welcome to the Allsky Installer! ***
****************************************

* ZWO camera found.
* Using ZWO camera.
* Size of current swap (2048 MB) is sufficient; no change needed.
* /home/pi/allsky/tmp is currently a memory filesystem; no change needed.

*** NOTICE:
**********
The following steps can take up to an hour depending on the speed of your Pi
and how many of the necessary dependencies are already installed.
You will see progress messages throughout the process.
At the end you will be prompted again for additional steps.
**********

* Installing the web server.
* Installing dependencies.
* Preparing Allsky commands.
* Modifying locations for WebUI.
* Setting up WebUI options and default settings for ZWO cameras.
* Making new settings file '/home/pi/allsky/config/settings.json'.

*** ERROR:
**********
Unable to save camera capabilities.
**********

pi@RORPI:~/allsky $**

[install.sh.log](https://github.com/thomasjacquin/allsky/files/12651160/install.sh.log)

Jonk2, 2023-09-25:
@EricClaeys 

Short of a full os reinstall (which doesn't help anyone figure this out) is there anything else you can suggest I try?

This PI is running in cli mode only and only running 1 python script reading and recording an i2c sensor. Nothing else.

EricClaeys, 2023-09-29:
@Alex-developer, do you have any ideas?  The error message is:
~~~
Fatal error: Uncaught Error: Call to undefined function json_decode() in /home/pi/allsky/html/includes/functions.php:56
~~~

@Jonk2 interrupted the initial installation as it was installing the web server (which includes php), so I'm assuming php wasn't completely installed, but says it is.  Hence php scripts can't find `json_decode()`.
Jon said he executed what I suggested:
~~~
sudo apt-get --assume-yes purge lighttpd php-cgi php-gd hostapd dnsmasq avahi-daemon
~~~
but it didn't help.

Jonk2, 2023-10-11:
@EricClaeys @Alex-developer 

Gents, I appreciate this is maybe a specific issue to me, rather than Allsky, but is there anything else to try before I wipe this PI and start again? Wiping isn't really an issue, but if there is a bug somewhere with the installer, perhaps I should wait to see if it can be found?

Alex-developer, 2023-10-11:
> atal error: Uncaught Error: Call to undefined function json_decode()

Sorry missed this. Sounds like the php-json package is missing. You could try installing that.

apt install -y php-json

EricClaeys, 2023-10-11:
@Jonk2, if what @Alex-developer suggested doesn't work and you decide to re-image, please save the config/installation_logs directory first in case we need to refer to it.  I still think it's likely that something related to php or json didn't get fully installed when you aborted the installation the first time, but it's reporting as installed.

Jonk2, 2023-10-12:
@EricClaeys The installer failed the first time with no previous allsky installed on a fairly fresh os (March this year I think). I tried again, same result which is why I raised it here.

@Alex-developer I followed your suggestion with the following result:

[image]

However the allsky installation script still fails with the same error message, even after a reboot.

So, I’ve saved the config/installation_logs folder in case anyone needs it and will write a fresh os using the pi installer as usual.

Then, hopefully it’ll work!

closed 2023-10-12 by Jonk2

sreegsnb, 2024-08-17:
Hi the above issue of installation error with "Unable to save camera capabilities" seems to be still occurring.  I have new Raspberry PI 5 (Operating System: Debian GNU/Linux 12 (bookworm); Kernel: Linux 6.6.31+rpt-rpi-2712; Architecture: arm64) and Arducam Imx519 and Arducam Owlsight pi cameras and encountered similar issue.  my log file below

* STARTING INSTALLATON AT Fri Aug 16 11:25:44 PM PDT 2024.

* No prior Allsky found.
* RPi camera found.
* Using the 'master' branch.
* Displaying notification image 'InstallationInProgress.jpg'
* CURRENT_LOCALE=en_US.UTF-8, TEMP_LOCALE=[[LANG=en_US.UTF-8 LANGUAGE=en_US.UTF-8 LC_ALL=en_US.UTF-8 ]]
* Using RPi camera.
* Using current hostname of 'gsnballskycam'.
* RAM_SIZE=8192, SUGGESTED_SWAP_SIZE=0.
* Size of current swap (200 MB) is sufficient; no change needed.
* /home/sreegsnb/allsky/tmp is currently a memory filesystem; no change needed.
* /home/sreegsnb/allsky/tmp already mounted.
* Installing the web server.
* Installing dependencies.
* Preparing Allsky commands.
* Modifying locations for WebUI.
* Setting up WebUI options and default settings for RPi cameras.
* Making new settings file '/home/sreegsnb/allsky/config/settings.json'.
* Executing makeChanges.sh --force --cameraTypeOnly   'cameraType' 'Camera Type' '' 'RPi'
* 
ERROR: Segmentation fault in capture_RPi -cmd libcamera-still
*** ERROR: 
**********
Unable to save camera capabilities.
**********

* Displaying notification image 'InstallationFailed.jpg'

* ENDING INSTALLATON AT Fri Aug 16 11:26:12 PM PDT 2024.

----------------------------------------------------------------------------------------------------

Please advise

sreegsnb, 2024-08-17:
Just FYI.. issue resolved.  Arducam Owlsight camera was the issue, removed the camera and reinstalled all worked fine on Raspberry Pi 5. 

Note for anyone trying to use RPi HQ camera or Arducam cameras with Raspberry Pi 5, please invest in new CSI cables (22pin to 15pin) or new 15pin cable, the stock cables that come with camera are not reliable and can cause "no camera available" errors.

Thanks