AllskyTeam/allsky #3008: ERROR: Missing configuration file 'configuration.json'. Cannot continue.

WirthmU, 2023-10-09:
I have tried to install a remote website using:

`website/install.sh --remote`

I can see all the files on the remote website including the updated config.js file. However when I tried to start the webiste I get this:

`ERROR: Missing configuration file 'configuration.json'. Cannot continue.`

The website URL is:
[80219.133./Bern/allsky_2/allsky](url)

Any help to this?

EricClaeys, 2023-10-09:
There is no config.js on the new Website; it was replaced by configuration.json.  The installation should have copied that file to the server.  Please attach the Website installation log file.  It's either in allsky/config or allsky/tmp.

The URL you provided doesn't work.  Would you please resend?

EricClaeys, 2023-10-09:
@WirthmU    `website/install.sh --remote` doesn't actually copy the web files to the remote site, it only copies configuration.json.  It sets up the Pi to know about the remote site, so whenever you change remote_configuration.json in the WebUI, or change settings like latitude, it will upload those to the remote site.
The log file indicates the installation copied configuration.json successfully.
It's possible it uploaded it to a different location than your manual upload.

Try this:  make a trivial change to remote_configuration.json via the "Editor" page of the WebUI.  Using Filezilla or something else, see if the configuration.json file on the server has that change.  If not, check ftp-settings.sh to make sure it's uploading to the correct place.

WirthmU, 2023-10-10:
What you mentioned worked and I can see the change on the website.
Thank you for your support.

closed 2023-10-10 by EricClaeys