AllskyTeam/allsky #3012: [BUG] Unable to build due to `-mfloat-abi=hard`

minichate, 2023-10-13:
### Environment

* Camera:  ASI178MC
* OS: Raspbian 12
* Allsky version: v2023.05.01_03
* Pi 4B with 8GB RAM

### Bug Description

Unable to build `src/` directory due to FPU error:

~~~
Building capture_ZWO.o ...
cc1plus: error: '-mfloat-abi=hard': selected architecture lacks an FPU
make: *** [Makefile:131: capture_ZWO.o] Error
~~~

### Log / configuration files
[make_all.log](https://github.com/thomasjacquin/allsky/files/12896246/make_all.log)

minichate, 2023-10-13:
I can get a little farther by changing: 

```
diff --git a/src/Makefile b/src/Makefile
index af794304..d8b3d096 100644
--- a/src/Makefile
+++ b/src/Makefile
@@ -69,8 +69,7 @@ else ifeq ($(arch), arm64)
 else           # $(platform) = armv7l or $(arch) = armhf
   CC = arm-linux-gnueabihf-g++
   AR= arm-linux-gnueabihf-ar
-  CFLAGS += -march=armv7 -mthumb
-  ZWOSDK = -Llib/armv7 -I./include
+  ZWOSDK = -Llib/armv8 -I./include
 endif
```

Where it then fails with:

```
$ make
Building capture_ZWO.o ...
Building allsky_common.o ...
2023-10-13 13:16:59 Building capture_ZWO program...
/usr/bin/ld: lib/armv8/libASICamera2.a: error adding symbols: file format not recognized
collect2: error: ld returned 1 exit status
make: *** [Makefile:134: capture_ZWO] Error 1
```

Which makes me believe this is an armv7 vs. armv8 problem, perhaps?

EricClaeys, 2023-10-13:
Are you running Buster or Bullseye?  I am not sure which is version 12.
And 32 or 64 bits?

minichate, 2023-10-13:
`bookworm` -- the latest version! https://www.raspberrypi.com/news/bookworm-the-new-version-of-raspberry-pi-os/

minichate, 2023-10-13:
I'll file an issue for the other `bookworm` related bugs I've found around python dependencies not working with the system Python (3.11) bundled on that OS!

EricClaeys, 2023-10-13:
@minichate, Christopher, if you post any more Bookworm-related things, please use this Issue, which I renamed.

s-valve, 2023-10-14:
Hi guys, i've got the same failure and have no idea how to solve. I have to say, I'm no familiar with Linux... Maybe you can give me a hint what to change in the makefile or any other location. thanks, Martin
[make_all.log](https://github.com/thomasjacquin/allsky/files/12905437/make_all.log)

Alex-developer, 2023-10-14:
> Hi guys, i've got the same failure and have no idea how to solve. I have to say, I'm no familiar with Linux... Maybe you can give me a hint what to change in the makefile or any other location. thanks, Martin
> 
> Used Raspi OS: Release date: October 10th 2023 System: 32-bit Kernel version: 6.1 Debian version: 12 (bookworm) Size: 2,606MB
> 
> [make_all.log](https://github.com/thomasjacquin/allsky/files/12905437/make_all.log)

There are a few issues with Bookworm that I am working on. As soon as I have a working installation I will update this issue

Cheers

Alex

s-valve, 2023-10-14:
Merci

EricClaeys, 2023-10-14:
@s-valve you can always install Bullseye.

Alex-developer, 2023-10-14:
Just a quick update.

I have a version that now runs on Bookworm. Next stage is to get it tested fully then we can think about releasing it

Alex

Jonk2, 2023-10-15:
@Alex-developer I have a fresh 32bit bookworm lite I've just installed, only to find out I cannot pip any 3rd party pythhon libraries any more.

Whilst I try and get my head round that, I'll be happy to test your new bookworm-proof allsky! I have nothing to lose if there are any issues.

Will it be put into the dev branch?

One thing I would suggest in general, is can the installer run without having to plug in a camera? If so, I personally would prefer that, as I often bring the test pi indoors, leaving the camera outside.

Alex-developer, 2023-10-15:
> @Alex-developer I have a fresh 32bit bookworm lite I've just installed, only to find out I cannot pip any 3rd party pythhon libraries any more.
>
> Whilst I try and get my head round that, I'll be [...]

The Point-Release-4 has changes for Bookworm - Please bear in mind that this is NOT fully tested yet. I have installed it on a few Pi's and it installs ok, there is a dev branch for the extra modules that also supports Bookworm

Once you have the new version to add any new python modules you will need to put them in the venv so something like

`source /home/pi/allsky/venv/bin/activate`

`pip3 whatever module`

`deactivate`

I'll leave @EricClaeys to answer the question about the camera as thats his area really.

Cheers

Alex

Jonk2, 2023-10-15:
Ok I see the dev branch is newer - I'll try that one, see how it goes.

EricClaeys, 2023-10-16:
@Jonk2 please do NOT try the dev branch for Allsky. It is not ready.

We changed the installer in the last major release to require that the camera be connected so it could determine what brand and model was being used and then set the default, minimum, and maximum values in the WebUI. It also solved other problems people were having.
Can you explain more why it would be beneficial to not require it to be plugged in during installation?  I can't think of any use case for that.

Eric

varunmehta, 2023-10-20:
@Alex-developer, @EricClaeys  & @Jonk2 

Successful able to start building (off `master`) on bookworm (other issues still there, working on it), here are the steps I followed; 

1. Install venv

```
sudo apt-get install python3.11-venv
``` 

2. Then create env with all site-packages 

```
cd allsky 
python -m venv --system-site-packages env
source env/bin/activate
```

3. The terminal prompt changes to env

``` 
(env) pi@allsky:~/allsky $ 
``` 

4. Execute `./install.sh` 

> new log with venv
> ```
> * Installing PHP modules.
> * Installing other PHP dependencies.
> * Installing Python_dependencies:
> *    === Package #  1 of 15: [numpy>=1.25.0]
> *    === Package #  2 of 15: [opencv-python<=4.6.0.66]
> *    === Package #  3 of 15: [Pillow]
> *    === Package #  4 of 15: [pyephem]
> ... 
> ``` 
> 
> old log without venv on bookworm 
> ``` 
> * Installing Python_dependencies:
> *    === Package #  1 of 15: [numpy>=1.25.0]
> 
> *** ERROR:
> **********
> Python_dependencies [numpy>=1.25.0] failed
> **********
> 
> * The full log file is in /home/pi/allsky/config/installation_logs/Python_dependencies.1.log
> The end of the file is:
> 
>     For more information visit http://rptl.io/venv
> ```
> Also refer https://www.raspberrypi.com/documentation/computers/os.html#python-on-raspberry-pi 

### Now `scipy` is failing
So got bookworm to start building with that, but `scipy` is acting up, working to debug that 

```
 × Building wheel for scipy (pyproject.toml) did not run successfully.
  │ exit code: 1
  ╰─> [920 lines of output]
      setup.py:563: DeprecationWarning:

        `numpy.distutils` is deprecated since NumPy 1.23.0, as a result
        of the deprecation of `distutils` itself. It will be removed for
        Python >= 3.12. For older Python versions it will remain present.
        It is recommended to use `setuptools < 60.0` for those Python versions.
        For more details, see:
          https://numpy.org/devdocs/reference/distutils_status_migration.html
```

Adding info if anyone else wants to continue from here, will update as I get further along...

Alex-developer, 2023-10-20:
Thanks for that - there is a branch point-release-4 that contains a 'working' install for bookworm but it's not been fully tested. Please feel free to give it a try and see if I have missed anything 

Cheers

Alex

Sundancer78, 2023-10-20:
Scipy doesn't work for me either...

How is the Point-Release-4 branch installed?

EricClaeys, 2023-10-20:
On the `git clone ...` command line, add `--branch Point-Release-4`.
Keep in mind we have not tested it as a release, although the updates have been tested by themselves.

Jonk2, 2023-10-21:
> Can you explain more why it would be beneficial to not require it to be plugged in during installation? I can't think of any use case for that.

@EricClaeys I have 2 systems, 1 outside on the roof with the Pi being relatively inaccessible - this is running an older Allsky that I only update when I'm happy to.

The other PI is in the ROR shed, over PoE, with a fixed camera. Quite often I need to power down the PI and bring it in as I don't have remote power on it yet. As it's a test PI, I often bring it in to write a new OS and bringing in the camera too is not sensible for me.

@Alex-developer 

I've tried your suggestions, ran the installation script (with env at the prompt) and I have this as a result:

> (env) pi@rorpi:~ $ cd allsky
> (env) pi@rorpi:~/allsky $ ./install.sh
> * STARTING INSTALLATON AT Sat 21 Oct 14:34:34 BST 2023.
> 
> 
> ****************************************
> *** Welcome to the Allsky Installer! ***
> ****************************************
> 
> * ZWO camera found.
> * Using ZWO camera.
> * Size of current swap (100 MB) is sufficient; no change needed.
> mount: (hint) your fstab has been modified, but systemd still uses
>        the old version; use 'systemctl daemon-reload' to reload.
> * /home/pi/allsky/tmp is now in memory.
> 
> *** NOTICE:
> **********
> The following steps can take up to an hour depending on the speed of your Pi
> and how many of the necessary dependencies are already installed.
> You will see progress messages throughout the process.
> At the end you will be prompted again for additional steps.
> **********
> 
> * Installing the web server.
> * Installing dependencies.
> * Preparing Allsky commands.
> 
> *** ERROR:
> **********
> Compile failed
> **********
> 
> * The full log file is in /home/pi/allsky/config/installation_logs/make_all.log
> The end of the file is:
> Building capture_ZWO.o ...
> cc1plus: error: ‘-mfloat-abi=hard’: selected architecture lacks an FPU
> make[1]: *** [Makefile:131: capture_ZWO.o] Error 1
> make[1]: Leaving directory '/home/pi/allsky/src'
> make: *** [Makefile:24: all] Error 2
> 
> * ENDING INSTALLATON AT Sat 21 Oct 14:38:40 BST 2023.

It's worth noting that I installed the lite version of bookworm as I have no need for the desktop environment. I had to install python-full (1st 'huh?' moment), git wasn't recognised as a command and am still nonethewiser how to install '3rd party' python packages in a venv, but I'll do some reading.

Is the lite version causing these issues? (Missing packages and "cc1plus: error: ‘-mfloat-abi=hard’: selected architecture lacks an FPU")

I've looked up that message and there are a lot of results, some pre-bookworm by the looks of it.

The pi is a 4B 8GB and I'm running the os from an m.2/usb3 combo.

Allsky logs attached.

[lighttpd.install.log](https://github.com/thomasjacquin/allsky/files/13060832/lighttpd.install.log)
[make_all.log](https://github.com/thomasjacquin/allsky/files/13060833/make_all.log)
[make_deps.log](https://github.com/thomasjacquin/allsky/files/13060834/make_deps.log)
[install.sh.log](https://github.com/thomasjacquin/allsky/files/13060835/install.sh.log)
[status.txt](https://github.com/thomasjacquin/allsky/files/13060836/status.txt)

Alex-developer, 2023-10-21:
> @EricClaeys I have 2 systems, 1 outside on the roof with the Pi being relatively inaccessible - this is running an older Allsky that I only update when I'm happy to.
>
> The other PI is in the ROR [...]

I think you are still using the master branch try the following on a clean install of Bookworm (Doesn't matter which one, the git install is ONLY required for the lite versions but won't hurt to run it on a desktop version)

sudo apt install git
git clone https://github.com/thomasjacquin/allsky.git
cd allsky
git checkout Point-Release-4
./install.sh

There is now a check in the master branch that will warn you ig you are trying to install on Bookworm

Jonk2, 2023-10-21:
I ran git clone --branch Point-Release-4 https://github.com/thomasjacquin/allsky.git then python -m venv --system-site-packages env etc.

Alex-developer, 2023-10-21:
No need for any of that. The Point release 4 does everything for you. Just checkout the branch and run the install as per normal

Jonk2, 2023-10-21:
Ok, am writing a fresh PiOS Bookworm Lite 32Bit now, then will try to install allsky before I touch anything else!

Jonk2, 2023-10-21:
Ok, new PI OS Bookworm Lite 32Bit installed.

Carried out the instructions as above, same result:

> 
> pi@RORPi:~ $ git clone https://github.com/thomasjacquin/allsky.git
> Cloning into 'allsky'...
> remote: Enumerating objects: 25228, done.
> remote: Counting objects: 100% (470/470), done.
> remote: Compressing objects: 100% (271/271), done.
> remote: Total 25228 (delta 237), reused 387 (delta 189), pack-reused 24758
> Receiving objects: 100% (25228/25228), 113.01 MiB | 13.50 MiB/s, done.
> Resolving deltas: 100% (15812/15812), done.
> pi@RORPi:~ $ cd allsky
> pi@RORPi:~/allsky $ git checkout Point-Release-4
> branch 'Point-Release-4' set up to track 'origin/Point-Release-4'.
> Switched to a new branch 'Point-Release-4'
> pi@RORPi:~/allsky $ ./install.sh
> 
> * You need to TYPE 'yes' to continue the installation.
> This is to make sure you read it.
> 
> pi@RORPi:~/allsky $ ./install.sh
> * STARTING INSTALLATON AT Sat 21 Oct 16:11:24 BST 2023.
> 
> 
> ****************************************
> *** Welcome to the Allsky Installer! ***
> ****************************************
> 
> * ZWO camera found.
> * Using ZWO camera.
> * Size of current swap (100 MB) is sufficient; no change needed.
> mount: (hint) your fstab has been modified, but systemd still uses
>        the old version; use 'systemctl daemon-reload' to reload.
> * /home/pi/allsky/tmp is now in memory.
> 
> *** NOTICE:
> **********
> The following steps can take up to an hour depending on the speed of your Pi
> and how many of the necessary dependencies are already installed.
> You will see progress messages throughout the process.
> At the end you will be prompted again for additional steps.
> **********
> 
> * Installing the web server.
> * Installing dependencies.
> * Preparing Allsky commands.
> 
> *** ERROR:
> **********
> Compile failed
> **********
> 
> * The full log file is in /home/pi/allsky/config/installation_logs/make_all.log
> The end of the file is:
> Building capture_ZWO.o ...
> cc1plus: error: ‘-mfloat-abi=hard’: selected architecture lacks an FPU
> make[1]: *** [Makefile:131: capture_ZWO.o] Error 1
> make[1]: Leaving directory '/home/pi/allsky/src'
> make: *** [Makefile:24: all] Error 2
> 
> * ENDING INSTALLATON AT Sat 21 Oct 16:15:35 BST 2023.
> 
>

Alex-developer, 2023-10-21:
> Ok, new PI OS Bookworm Lite 32Bit installed.
>
> Carried out the instructions as above, same result:

Oops I am an idiot !!! The updates were in my fork of the release not in the allsky one !!!

I have just merged them and testing it now, I'll let you know shortly if its ok to try

Alex-developer, 2023-10-21:
@Jonk2 

I have sorted it now - Please try the install again

Sorry for the confusion - all my fault

Cheers

Alex

Jonk2, 2023-10-21:
No worries, we're all idiots really!

I'll try again in a bit and let you know.

Jonk2, 2023-10-21:
Slightly off topic, but is it easy / possible to install 3rd party python libraries with the installer? I've moved from BME280 to SHT31-D and the bookworm version does not allow simple PIP installs any more.
This may be easy for linux fluent people to get their head around, but for me at the moment, it's not.
Basically I'm struggling to install the adafruit sht31-d library for my python heating and cooling script (replace BME with SHT).
This bookworm release as you know has changed a lot of things. Yes I can use buster but there will become a time to upgrade so I'd rather do it now.

Alex-developer, 2023-10-21:
> Slightly off topic, but is it easy / possible to install 3rd party python libraries with the installer? I've moved from BME280 to SHT31-D and the bookworm version does not allow simple PIP installs [...]

Yes very easy. After the installation is finished enter the following

`source ~/allsky/venv/bin/activate`

The prompt will change to have (venv) a the start

`(venv) pi@book:~ $`

Then do your pip installs as required and once you are finished type deactivate at the prompt

Sundancer78, 2023-10-22:
> Ich denke, Sie verwenden immer noch den Master-Zweig. Versuchen Sie Folgendes bei einer Neuinstallation von Bookworm (egal welche, die Git-Installation ist NUR für die Lite-Versionen erforderlich, kann aber nicht schaden, wenn Sie sie auf einer Desktop-Version ausführen).
> 
> sudo apt install git git clone https://github.com/thomasjacquin/allsky.git cd allsky git checkout Point-Release-4 ./install.sh
> 
> Es gibt jetzt eine Prüfung im Hauptzweig, die Sie warnt, wenn Sie versuchen, auf Bookworm zu installieren

Thanks! Installation goes smoothly. AllSky then works on an RPi 4, 4GB with Rpi HQ camera module.

EricClaeys, 2023-10-22:
We recommend 
~~~
git clone --branch Point-Release-4 https://github.com/thomasjacquin/allsky.git
cd ~/allsky
./install.sh 
~~~

But the other method works as well.
If you install this pre-release please let us know how it goes, good or bad.

Sundancer78, 2023-10-22:
does the AllSky website installer also need to be adapted to Bookworm? Or can the installation be carried out easily?

EricClaeys, 2023-10-22:
@Sundancer78, not sure.  We haven't tested the Website installer yet, but I suspect it'll work fine.  It doesn't compile anything or use Python which caused problems for Allsky.

If you try it, let us know how it goes.

@Alex-developer, have you tested website/install.sh?

Alex-developer, 2023-10-22:
> @Sundancer78, not sure. We haven't tested the Website installer yet, but I suspect it'll work fine. It doesn't compile anything or use Python which caused problems for Allsky.
> 
> If you try it, let us know how it goes.
> 
> @Alex-developer, have you tested website/install.sh?

No I haven't tested the website installer

Jonk2, 2023-10-23:
Good news @Alex-developer the new point 4 release has installed on Bookworm LITE 32Bit without any errors.

Now the testing begins!

> 
> Linux RORPi 6.1.0-rpi4-rpi-v8 #1 SMP PREEMPT Debian 1:6.1.54-1+rpt2 (2023-10-05)                                                                                                                                                              aarch64
> 
> The programs included with the Debian GNU/Linux system are free software;
> the exact distribution terms for each program are described in the
> individual files in /usr/share/doc/*/copyright.
> 
> Debian GNU/Linux comes with ABSOLUTELY NO WARRANTY, to the extent
> permitted by applicable law.
> Last login: Mon Oct 23 20:45:16 2023 from 192.168.0.2
> pi@RORPi:~ $ git clone https://github.com/thomasjacquin/allsky.git
> Cloning into 'allsky'...
> remote: Enumerating objects: 25289, done.
> remote: Counting objects: 100% (531/531), done.
> remote: Compressing objects: 100% (314/314), done.
> remote: Total 25289 (delta 272), reused 426 (delta 207), pack-reused 24758
> Receiving objects: 100% (25289/25289), 113.04 MiB | 12.44 MiB/s, done.
> Resolving deltas: 100% (15847/15847), done.
> pi@RORPi:~ $ cd allsky
> pi@RORPi:~/allsky $ git checkout Point-Release-4
> branch 'Point-Release-4' set up to track 'origin/Point-Release-4'.
> Switched to a new branch 'Point-Release-4'
> pi@RORPi:~/allsky $ ./install.sh
> * STARTING INSTALLATON AT Mon 23 Oct 20:58:17 BST 2023.
> 
> 
> ****************************************
> *** Welcome to the Allsky Installer! ***
> ****************************************
> 
> * ZWO camera found.
> * Using ZWO camera.
> * Size of current swap (100 MB) is sufficient; no change needed.
> mount: (hint) your fstab has been modified, but systemd still uses
>        the old version; use 'systemctl daemon-reload' to reload.
> * /home/pi/allsky/tmp is now in memory.
> 
> *** NOTICE:
> **********
> The following steps can take up to an hour depending on the speed of your Pi
> and how many of the necessary dependencies are already installed.
> You will see progress messages throughout the process.
> At the end you will be prompted again for additional steps.
> **********
> 
> * Installing the web server.
> * Installing dependencies.
> * Preparing Allsky commands.
> * Modifying locations for WebUI.
> * Setting up WebUI options and default settings for ZWO cameras.
> * Making new settings file '/home/pi/allsky/config/settings.json'.
> * Executing makeChanges.sh --force --cameraTypeOnly  --debug 'cameraType' 'Camera Type' '' 'ZWO'
> * makeChanges.sh: cameraType: Old=[], New=[ZWO]
> Calling capture_ZWO -cc_file '/home/pi/allsky/config/cc.json'
> '/home/pi/allsky/config/overlay/config/overlay-ZWO.json' doesn't exist yet - ignoring.
> Calling: /home/pi/allsky/html/includes/createAllskyOptions.php --force --debug
>         --cc_file /home/pi/allsky/config/cc.json
>         --options_file /home/pi/allsky/config/options.json
>         --settings_file /home/pi/allsky/config/settings.json
>    Argument debug
> cameraType=ZWO, cameraModel=ASI178MC
> Camera-specific settings file exists (no): /home/pi/allsky/config/settings_ZWO_ASI178MC.json.
> Creating camera-specific settings file: /home/pi/allsky/config/settings_ZWO_ASI178MC.json.
> Linking /home/pi/allsky/config/settings_ZWO_ASI178MC.json to /home/pi/allsky/config/settings.json.
> * Settings files:
>     -rw-r--r-- 2 pi pi 2135 Oct 23 21:03 /home/pi/allsky/config/settings.json
>     -rw-r--r-- 2 pi pi 2135 Oct 23 21:03 /home/pi/allsky/config/settings_ZWO_ASI178MC.json
> * Keeping 'en_GB.UTF-8' locale.
> * Setting permissions on /var/log/allsky.log and /var/log/allskyperiodic.log.
> * Installing PHP modules.
> * Installing other PHP dependencies.
> * Installing Python_dependencies for Bookworm:
> *    === Package #  1 of 15: [numpy>=1.25.0]
> *    === Package #  2 of 15: [opencv-python<=4.7.0.72]
> *    === Package #  3 of 15: [Pillow]
> *    === Package #  4 of 15: [pyephem]
> *    === Package #  5 of 15: [skyfield]
> *    === Package #  6 of 15: [astral]
> *    === Package #  7 of 15: [pytz]
> *    === Package #  8 of 15: [scipy<=1.11.3]
> *    === Package #  9 of 15: [paho-mqtt]
> *    === Package # 10 of 15: [photutils]
> *    === Package # 11 of 15: [astropy]
> *    === Package # 12 of 15: [suncalc]
> *    === Package # 13 of 15: [Adafruit-Blinka]
> *    === Package # 14 of 15: [vcgencmd]
> *    === Package # 15 of 15: [requests]
> * Installing Trutype fonts.
> * Setting up modules and overlays.
> * Copying '/home/pi/allsky/config/overlay/config/overlay-ZWO.json' to 'overlay.json'.
> * Prompting for Latitude and Longitude.
> * Updating some '/home/pi/allsky/config/config.sh' variables.
> * Setting permissions on web-related files.
> * Adding pi to www-data group.
> addgroup: addgroup with two arguments is an unspecified operation.
> * Creating/updating sudoers file.
> 
> *** NOTICE:
> **********
> You need to reboot the Pi before Allsky will work.
> **********
> 
> 
> *** NOTICE:
> **********
> Please provide feedback on the new auto-exposure algorithm by entering a Discussion item in GitHub.
> You can disable it by changing 'New Exposure Algorithm' in the WebUI.
> **********
> 
> 
> * Installation is done but the Pi needs a reboot.
> 
> 
> * ENDING INSTALLATON AT Mon 23 Oct 21:08:40 BST 2023.

Jonk2, 2023-10-23:
Ok, feedback time.

I went through and entered some exposure setting, copying the allsky camera outside (same ZWO178MC, but cool version)... I am not getting anythign back from the camera.

These are the messages at the top of the admin page:

> ***  /home/pi/allsky/bin/capture_ZWO: WARNING: ASISetControlValue() for  control 5, value=0, camNum=0, makeAuto=0 failed: ASI_ERROR_GENERAL_ERROR  (16)   (October 23,  9:23:51 pm BST) |  
> -- | --
>  
> > /home/pi/allsky/bin/capture_ZWO: ERROR: Failed getting image: ASI_ERROR_TIMEOUT #1 (with 0.8 exposure = YES) (11)   (October 23,  9:25:31 pm BST) |  
>  
> ***  /home/pi/allsky/bin/capture_ZWO: WARNING: ASISetControlValue() for  control 1, value=32, camNum=0, makeAuto=0 failed:  ASI_ERROR_GENERAL_ERROR (16)   (2 occurrences, last on October 23,  9:25:33 pm BST) |  
>  
> *** /home/pi/allsky/bin/capture_ZWO: ERROR: flushBufferedImages() setControl() returned ASI_ERROR_GENERAL_ERROR (16)   (2 occurrences, last on October 23,  9:25:33 pm BST) |  
>  
> ***  /home/pi/allsky/bin/capture_ZWO: WARNING: ASISetControlValue() for  control 1, value=10000000, camNum=0, makeAuto=0 failed:  ASI_ERROR_GENERAL_ERROR (16)   (2 occurrences, last on October 23,  9:25:34 pm BST) |  
>  
> > /home/pi/allsky/bin/capture_ZWO: ERROR: Failed getting image: ASI_ERROR_TIMEOUT #2 (with 0.8 exposure = YES) (11)   (October 23,  9:25:59 pm BST)
> 

I will reboot and see what happens but I'm posting this as initial feedback only.

Jonk2, 2023-10-23:
> Yes very easy. After the installation is finished enter the following
>
> `source ~/allsky/venv/bin/activate`
>
> The prompt will change to have (venv) a the start
>
> `(venv) pi@book:~ $`
>
> Then do your pip [...]

Well, that comes up with the same error message....

> 
> pi@RORPi:~ $ source ~/allsky/venv/bin/activate
> (venv) pi@RORPi:~ $ sudo pip3 install adafruit-circuitpython-sht31d
> error: externally-managed-environment
> 
> × This environment is externally managed
> ╰─> To install Python packages system-wide, try apt install
>     python3-xyz, where xyz is the package you are trying to
>     install.
> 
>     If you wish to install a non-Debian-packaged Python package,
>     create a virtual environment using python3 -m venv path/to/venv.
>     Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
>     sure you have python3-full installed.
> 
>     For more information visit http://rptl.io/venv
> 
> note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
> hint: See PEP 668 for the detailed specification.
> (venv) pi@RORPi:~ $

EricClaeys, 2023-10-23:
@Jonk2  Jon, would you please clear the log file, then change Debug Level to 4 and after a few minutes attach the log file?
Also, try increasing and decreasing the USB value.

It looks as if the Pi can't communicate with the camera.

Jonk2, 2023-10-25:
> @Jonk2 Jon, would you please clear the log file, then change Debug Level to 4 and after a few minutes attach the log file? Also, try increasing and decreasing the USB value.
> 
> It looks as if the Pi can't communicate with the camera.

A friend of mine popped over last night and brought his PI and ZWO178MC. I tried his camera on my PI / latest point-4 allsky and it was fine.

So, as far as I can tell, I have an issue with my camera.

EricClaeys, 2023-10-25:
Can you try changing the USB with your camera?  Also if possible, try everything on the Troubleshooting->ZWO Cameras page related to the TIMEOUT errors.  I suspect your camera is very sensitive to USB speed and/or power levels

Jonk2, 2023-10-25:
Would that suggest something has changed in the code / bookworm? The camera has been fine up until now so it's a bit odd.

Jonk2, 2023-10-25:
I’ve plugged the camera into a pc and it works perfectly fine. I will try auto and different bandwidth figures later on. I’ve already tried a usb2 socket on the pi rather than the original usb3 socket, same result.

EricClaeys, 2023-10-25:
Some cameras start having problems with a new release or stop having problems, even if the changes "shouldn't" have any impact.  I don't know if that's because even minor changes impact the CPU usage and/or timing, or there is a bug in the code that only appears under certain conditions and those conditions change.  Or there is a bug in the ZWO library.  I have looked and looked everywhere but found nothing.

EricClaeys, 2023-10-25:
Have you tried standing on your left foot and chanting????

Seriously though, other people have reported exactly the same things.  All the suggestions on the Troubleshooting->ZWO Cameras page helped one or more people.

Plugging into a PC changes two variables: 1) the PC's USB port may provide more power. 2) Windows uses a different ZWO library.  Either of those could be the cause.

Jonk2, 2023-10-25:
If dancing around could help...

I sometimes shout at things...it does nothing but can be amusing.

Seriously though, I've cleared the allsky log and restarted.

Before I did this, I tried auto bandwidth, anything between 10-99 on manual, turned off v0.8 exposure, turned off new experimental exposure and swapped between USB2 and USB3 outputs. Nothing has worked. The PI is indoors, dry etc and is supplied via a POE hat, plenty of power.

The previous version of allsky worked with the camera which is why I initially suspected it had developed a fault, but as I said it works fine on a PC.

My friend's 178MC worked on this PI / point-4 combo first time, with no issues last night using the same cable, so the only thing I can susggest is either bookworm or allsky (or ZWO library) has an issue.

I could revert back to either buster or an older allsky to see if it then works.

I'll let you look at the log and see if you can see something that might be causing this.

This is the test PI, so I'm happy to break it if needs be (by that I mean try whatever you need me to!).
[allsky.log](https://github.com/thomasjacquin/allsky/files/13170578/allsky.log)

EricClaeys, 2023-10-25:
I will look at the log when I am on my PC.
Your friend's camera worked on your Pi with the same cable?  The ONLY difference was the camera?
If so, that doesn't necessarily mean anything is wrong with your camera.  It may be fine but more sensitive to something that changes between releases.

The hard part in troubleshooting this is that us developers don't have that problem.

Jonk2, 2023-10-25:
Yes that's right, same model of camera, I literally unplugged mine and plugged his in - worked fine.

I have no idea if there is an age difference or the FW of the camera might be different, but it wa another 178MC, same as mine.

Jonk2, 2023-10-30:
@EricClaeys Ok, little update.

I swapped the long USB cable for a short USB3 cable and the camera connects and shows an image. In the past 3 or so days, I have seen the odd timeout (my usb is set to automatic).

There must have been a change in bookworm on the usb control, or something else, as the camera worked perfectly fine with that cable previously (prior to wiping the drive and installing the latest pios). It also works fine on a Windows PC (I know there are differences).

So, in my case, the camera and bookworm allsky appear to be working ok at the moment, but it's worth noting that there is something different.

I will do some more testing (indoors) and will report anything I think may help.

EricClaeys, 2023-10-30:
It really seems like a lot of ZWO cameras on the Pi are very sensitive to something (power, USB speed, ???), and if that something changes the camera starts having problems.
A year ago some reported his camera gave TIMEOUT errors when it reached a certain temperature.  That could be happening with other people as well.

Jonk2, 2023-10-30:
Well, my initial thought that my camera was faulty is put to bed.

The other issue is not all PIs are the same either, so something I see others may not see and vice versa.

On another note (possibly bookworm related) I still cannot get the AS_CPUTEMP to show a number:

[image]

[image]

[image]

EricClaeys, 2023-10-30:
Is AS_CPUTEMP in the "Allsky Variables" list, or just the "All Variables" list?  The ??? usually means it's undefined which means it's not in "Allsky Variables"

Jonk2, 2023-10-30:
It was in the all variables list, so I pressed '+', filled in some boxes and the '+' box then became greyed out.

In the Allsky variables, I have just noticed this:

[image]

Is this the one to use?

Jonk2, 2023-10-30:
I think yes...

[image]

EricClaeys, 2023-10-30:
Looks right.  I've asked @Alex-developer about the possibility of having just on list in the future.  Several people have had the same problem you did.

Alex-developer, 2023-10-30:
@EricClaeys the new webui only has one list :-)

Jonk2, 2023-11-05:
Gents, I've just installed the latest point-4 release (renaming allsky to allsk-OLD first, the previous point-4 release) and the installation script has reported an error copying the modules directory:

```
pi@RORPi:~/allsky $ git checkout Point-Release-4
branch 'Point-Release-4' set up to track 'origin/Point-Release-4'.
Switched to a new branch 'Point-Release-4'
pi@RORPi:~/allsky $ ./install.sh
* STARTING INSTALLATON AT Sun  5 Nov 21:19:49 GMT 2023.

********************************************************
*** Welcome to the Allsky Installer - v2023.05.01_04 ***
********************************************************

* ZWO camera found.
* Size of current swap (2048 MB) is sufficient; no change needed.
* /home/pi/allsky/tmp is currently a memory filesystem; no change needed.
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.

*** NOTICE:
**********
The following steps can take up to an hour depending on the speed of your Pi
and how many of the necessary dependencies are already installed.
You will see progress messages throughout the process.
At the end you will be prompted again for additional steps.
**********

* Installing the web server.
* Installing dependencies.
* Preparing Allsky commands.
* Modifying locations for WebUI.
* Setting up WebUI options for ZWO cameras.
* Restoring camera-specific settings files:
*       settings_ZWO_ASI178MC.json
* Making new settings file '/home/pi/allsky/config/settings.json'.
* Executing makeChanges.sh --cameraTypeOnly  --debug 'cameraType' 'Camera Type' 'ZWO' 'ZWO'
* makeChanges.sh: cameraType: Old=[ZWO], New=[ZWO]
Calling capture_ZWO -cc_file '/home/pi/allsky/config/cc.json'
Calling: /home/pi/allsky/html/includes/createAllskyOptions.php --debug
        --cc_file /home/pi/allsky/config/cc.json
        --options_file /home/pi/allsky/config/options.json
        --settings_file /home/pi/allsky/config/settings.json
   Argument debug
cameraType=ZWO, cameraModel=ASI178MC
Camera-specific settings file exists (yes): /home/pi/allsky/config/settings_ZWO_ASI178MC.json.
Using existing /home/pi/allsky/config/settings_ZWO_ASI178MC.json.
Linking /home/pi/allsky/config/settings_ZWO_ASI178MC.json to /home/pi/allsky/config/settings.json.
* Settings files:
    -rw-rw-r-- 2 pi pi 2376 Oct 29 19:23 /home/pi/allsky/config/settings.json
    -rw-rw-r-- 2 pi pi 2376 Oct 29 19:23 /home/pi/allsky/config/settings_ZWO_ASI178MC.json
* Keeping 'en_GB.UTF-8' locale.
* Setting permissions on /var/log/allsky.log and /var/log/allskyperiodic.log.
* Installing PHP modules and dependencies.
* Installing python3-full libgfortran5 libopenblas0-pthread.
* Installing Python_dependencies for :
*    === Package #  1 of 15: [numpy>=1.25.0]
*    === Package #  2 of 15: [opencv-python<=4.7.0.72]
*    === Package #  3 of 15: [Pillow]
*    === Package #  4 of 15: [pyephem]
*    === Package #  5 of 15: [skyfield]
*    === Package #  6 of 15: [astral]
*    === Package #  7 of 15: [pytz]
*    === Package #  8 of 15: [scipy<=1.11.3]
*    === Package #  9 of 15: [paho-mqtt]
*    === Package # 10 of 15: [photutils]
*    === Package # 11 of 15: [astropy]
*    === Package # 12 of 15: [suncalc]
*    === Package # 13 of 15: [Adafruit-Blinka]
*    === Package # 14 of 15: [vcgencmd]
*    === Package # 15 of 15: [requests]
[... 40 lines cut ...]
**********

*     'overlay' directory
*     '/home/pi/allsky-OLD/config/overlay/extra' directory
*     WebUI security settings
*     'remote_configuration.json': NO PRIOR VERSION
*     'config.sh' file, as is
*         CONFIG_SH_VERSION=1, PRIOR=1
*     ftp-settings.sh, as is
*         FTP_SH_VERSION=1, PRIOR=1
* Updating some '/home/pi/allsky/config/config.sh' variables.
* Setting permissions on web-related files.
* Adding pi to www-data group.
addgroup: addgroup with two arguments is an unspecified operation.
* Creating/updating sudoers file.

* Installation is done and Allsky is ready to start.

* Enjoy Allsky!

* ENDING INSTALLATON AT Sun  5 Nov 21:26:19 GMT 2023.

pi@RORPi:~/allsky $

```

Alex-developer, 2023-11-06:
I think I know what this is - I'll do a quick test

Alex-developer, 2023-11-06:
Can you try the upgrade again please I have added a fix for this

Jonk2, 2023-11-08:
> Can you try the upgrade again please I have added a fix for this

I've had to wipe my drive and start afresh - I may have a hardware issue, so this means that I cannot try the fix until the next time there's a new test release.

If there is an issue when that time comes, I'll report back.

Jonk2, 2023-11-10:
Right a little update.

I have confirmed (to myself at least) that my issues were caused by hardware, not software.

I have 2 RPI4s both with POE Hats and an M.2 SSD in an enclosure. 

For months, the fixed PI has been running version 2023-05 and would lock me out (error 500, no SSH, no web access) but still appeared to be running. A hard power down to reboot was all I could do.

This would be random, day or night and sometimes once every 3 days, or sometimes up to 30 days.

I thought it may be the OS, so I changed it to boot to CLI only (no desktop). The PI ran for 3 months without issue. Then it crashed again, same issue.

The 2nd PI that I move between indoors and the ROR shed for testing new allskys etc, had the issue I mentioned before about the camera timeouts after installing bookworm lite and the latest point-4 allsky.

A shorter cable appeared to solve that. Since then, every reboot would take 5-10 minutes before I could do anything with it, and this past week, it wouldn't even obtain the server-set DHCP IP address, so I couldn't even access anything on the PI.

So....I decided to start afresh with a standard USB3 memory stick.

Since last night, I have had zero issues booting, multiple reboots, no camera timeouts (with the previously used longer cable I thought was the culprit), I can ssh in within a minute or so of booting, allsky is working perfectly fine, my scripts are running fine and am getting good results.

The issue (I'm fairly sure) was down to the M.2 SSD enclosure - I can only think that it is unstable, and probably drawing the limit of power, especially with a camera connected to USB3 too.

I am now very confident that allsky on bookworm lite 32Bit is running as intended, but I will monitor for the next week and let you know if I find anything else to discuss.

Here's a nice view from within my ROR whilst I concentrate the TEC140 on the Dumbell Nebula!

[image]

[19 more replies]

closed 2023-12-08 by EricClaeys