AllskyTeam/allsky #3020: AllSky software installation failed

42-Stefan, 2023-10-19:
Raspberry Pi 4 Model B Rev 1.5  4GB
RPi-HQ-camera

on the command -> tail -n 4 /proc/cpuinfo :
Hardware: BCM2835
Revision: c03115
Serial: 10000000ff82110c

Raspberry Pi OS with desktop
Release date: October 10th 2023
System: 32-bit
Kernel version: 6.1
Debian version: 12 (bookworm)

### Bug Description

Hi, I'm new to Github and also making my first experience with the Raspberry Pi.
I'm on loan for everything this is about.  
I would like to get an AllSkyCam up and running, but am failing with the AllSky software installation.

The OS installation works. After OS installation I always make an update with:
sudo apt-get update
sudo apt-get dist-upgrade
That works also well.
Then I download the software from Github and start the installation. 
But the installation aborts every time with errors like this:

*** Welcome to the Allsky Installer! ***
* RPi camera found.
* Using RPi camera.
* Size of current swap (100 MB) is sufficient; no change needed.
mount: (hint) your fstab has been modified, but systemd still uses
       the old version; use 'systemctl daemon-reload' to reload.
* /home/sttie/allsky/tmp is now in memory.
*** NOTICE:
* Installing the web server.
* Installing dependencies.
* Preparing Allsky commands.

*** ERROR:
Compile failed

* The full log file is in /home/sttie/allsky/config/installation_logs/make_all.log
Log file content:
make[1]: Entering directory '/home/sttie/allsky/src'
Building capture_ZWO.o ...
cc1plus: error: '-mfloat-abi=hard': selected architecture lacks an FPU
make[1]: *** [Makefile:131: capture_ZWO.o] Error 1
make[1]: Leaving directory '/home/sttie/allsky/src'
make: *** [Makefile:24: all] Error 2

I also tried Bookworm 64bit lite, the result was: 
…
* Installing the web server.
* Installing dependencies.
* Preparing Allsky commands.
* Modifying locations for WebUI.
* Setting up WebUI options and default settings for RPi cameras.
* Making new settings file '/home/sttie/allsky/config/settings.json'.
* Keeping 'en_GB.UTF-8' locale.
* Setting permissions on /var/log/allsky.log and /var/log/allskyperiodic.log.
Failed to restart rsyslog.service: Unit rsyslog.service not found.
* Installing PHP modules.
* Installing other PHP dependencies.
* Installing Python_dependencies:
*    === Package #  1 of 15: [numpy>=1.25.0]

*** ERROR
Python_dependencies [numpy>=1.25.0] failed

* The full log file is in /home/sttie/allsky/config/installation_logs/Python_dependencies.1.log
content of the file is:

sttie@pi:~/allsky/config/installation_logs $ nano Python_dependencies.1.log
  GNU nano 7.2                Python_dependencies.1.log
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.

    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    For more information visit http://rptl.io/venv
note: If you believe this is a mistake, please contact your Python installation>
hint: See PEP 668 for the detailed specification. 

Do I need to try any other OS version?
Would be great if somebody could help me.
Please consider I’am a very beginner. 
Thank you! Stefan

Markh129, 2023-10-19:
I have the same problem. New Pi4b and new install of Pi OS. HElp !

Markh129, 2023-10-19:
I tried another install and got further, but it still crashed. Now I get as far as Python stuff and it stops at #11. Here is the log file. Thanks for any help.
Mark
[Python_dependencies.11.log](https://github.com/thomasjacquin/allsky/files/13047038/Python_dependencies.11.log)

Markh129, 2023-10-19:
Ok, tried to reinstall and all went well. Sorry for the trouble.

Alex-developer, 2023-10-19:
Bookworm isnt supported yet, We have a release in testing that supports Bookworm so you can wait for that, hopefully wont be too long or use Bullseye in the emantime

Cheers

Alex

EricClaeys, 2023-10-19:
@Markh129, would you please run
~~~
cat /etc/os.release
uname -a
~~~
on your Bookworm system and let me know the results?  It'll help us support Bookworm.
Thanks - Eric

EricClaeys, 2023-10-19:
@Alex-developer, I think we should update the "master" install.sh to check for Bookworm, and print a message and exit.  Once Point Release 4 is out that check will go away.
Agree?

Alex-developer, 2023-10-19:
> @Alex-developer, I think we should update the "master" install.sh to check for Bookworm, and print a message and exit. Once Point Release 4 is out that check will go away. Agree?

Excellent idea :-)

42-Stefan, 2023-10-20:
hello, I tried a new OS version:

Raspberry Pi OS (Legacy) with desktop
•	Release date: May 3rd 2023
•	System: 32-bit
•	Kernel version: 6.1
•	Debian version: 11 (bullseye)
•	Size: 872MB

after installation I made the updates as before.
I changed some points in the config.
-> Interface options -> I1 Legacy Camera -> Enable
-> Performance Options -> P4 Fan -> GPIO 14 -> 60°C

I started the installation, camera was found. 
I confirmed all the points at the beginning of the installation. 

Success!  The issue is solved. Thank you!

EricClaeys, 2023-10-20:
Good to hear!