AllskyTeam/allsky #3031: Missing Directories

LB376, 2023-10-24:
Hello,
I have had a hardware failure and am having to rebuild my Allsky with a new board.
The install seemed to be fine and the WebUI is up and running without any problems. My live view is uploading to my website, but I am not getting any timelapses, keograms or star trails uploading (they are in the Web UI). 
I ran the test on the ftp settings and the test file uploaded to my website - so that isn't the issue.
I did some more digging around and I seem to be missing the directories.
This is the information I get when I run:

`scripts/check_allsky.sh`

PROTOCOL set to insecure 'ftp'.  Try to use 'ftps' or 'sftp' instead.
-----
WEB_VIDEOS_DIR is set to '/home/lisab/allsky/html/allsky/videos' but it does not exist.
-----
WEB_KEOGRAM_DIR is set to '/home/lisab/allsky/html/allsky/keograms' but it does not exist.
-----
WEB_STARTRAILS_DIR is set to '/home/lisab/allsky/html/allsky/startrails' but it does not exist.

Sorry about this - I know it's user error, but I can't work out where I've gone wrong.
Thanks,
Lisa

LB376, 2023-10-24:
Sorry, I've realised that I skipped the Website install!

closed 2023-10-24 by LB376

EricClaeys, 2023-10-24:
Lisa, FYI, the next major version of Allsky will include the Website on the Pi so no installation of it is needed, which will eliminate the problem you had.  The WEB_*_DIR settings won't be needed since the "startrails", "keograms", and "timelapse" directories will already be there.  All you'll need to do to use the Website on the Pi is enable it in the WebUI.  Very simple, and I hope it'll eliminate a LOT of problems and confusion.

LB376, 2023-10-24:
Thanks Eric 😃
I actually tried to delete my issue because I realised my error the moment I posted it! I’ve enabled the Website and it all looks good now.
Best wishes,
Lisa

EricClaeys, 2023-10-24:
the Issue was deleted on GitHub but I still had the email.  It's good for me to see user errors since that gives me ideas on how to make Allsky more foolproof, like redoing the local Website.

LB376, 2023-10-24:
Thanks again Eric. Really appreciate it! 😊

LB376, 2023-10-28:
Hi Eric,
I am so sorry to bother you (again), but I am getting an error message.
Everything seems to be working fine. The images are uploading to my website no problem. But I have the "Warning: data.json is 9 days old" message appearing.
I ran ./post?Data.sh -debug and got the following:

WEBSITES=both, WEBSITES_TO_TO=
cp '/home/lisab/allsky/config/settings.json' '/home/lisab/allsky/html/allsky/viewSettings'
Uploading '/home/lisab/allsky/config/settings.json' to /allsky.lisabromfield.space/viewSettings

Commands used are in: /home/lisab/allsky/tmp/lftp_cmds/PostData.txt
Upload output from '/home/lisab/allsky/config/settings.json:
   *** upload.sh: ERROR, RET=1:
FILE_TO_UPLOAD='/home/lisab/allsky/config/settings.json'
REMOTE_HOST='allsky.lisabromfield.space'
REMOTE_DIR='/allsky.lisabromfield.space/viewSettings/'
TEMP_NAME='PostData-1125'
DESTINATION_NAME='settings.json'

And further down...

cd: Access failed: 550 Can't change directory to /allsky.lisabromfield.space/viewSettings: No such file or directory
---> QUIT
<--- 221-Goodbye. You uploaded 0 and downloaded 0 kbytes.
<--- 221 Logout.

postData.sh: Unable to upload '/home/lisab/allsky/config/settings.json'.

Are you able to help with this? Would you like me to raise this as an issue on GitHub?
Thanks,
Lisa

LB376, 2023-10-28:
Lisa Bromfield would like to recall the message, "[thomasjacquin/allsky] Missing Directories (Issue #3031)".

LB376, 2023-10-28:
Hi Eric,
Not sure if you received my latest email about the upload error but I have been able to fix it (I've tried recalling the email).
I realised that I didn't update the Allsky Website Master file when I rebuilt the Allsky Pi. I've done that now and everything seems to be working.
Sorry about this...hopefully you wont here from me for a while!
Thanks,
Lisa

EricClaeys, 2023-10-28:
Lisa,
What is the "Allsky Website Master" file?
You have a local AND remote Allsky Website,  right?  What is your `${IMAGE_DIR}` set to in ftp-settings.sh?  When you ftp into the remote server, where are the Allsky files like `index.php` and `configuration.json`?

Your posts don't bother me at all.  In fact, "user error" posts give me great feedback on how to improve Allsky to require less experience and be more bullet proof.  If you had a problem or made a mistake in an area, it's very likely others have or will at some point.  My desire is to lessen the chance of that happening.

As an example, many users have difficulties setting the Website variables in ftp correctly so their uploads fail and it'sa frustratingexperience. It's too complicated and error prone - it took me well over a year to fully understand it!  In the next release of Allsky, using a local Website will only require enabling it in the WebUI - no variables need to be set.  And remote Websites will only need the IMAGE_DIR variable set plus of course the remote host and password.  Those confusing WEB_*_DIR variables are gone - they were my dumbest idea a couple years ago.

Keep the feedback coming.  It helps make Allsky better.

LB376, 2023-10-28:
Thanks Eric! I really appreciate your help and support 😊
The ftp was working, everything was uploading to my website but I was getting an error in the WebUI (screenshot attached). After I emailed you, it occurred to me that the allsky-website-master directory<https://github.com/thomasjacquin/allsky-website> on the remote server was no longer compatible. The ones I had on the remote server were from my previous build of last year (sorry I don’t know what version) and that they were likely out of date for the newest version of Allsky. I downloaded the Zip file, followed the instructions for uploading to the server and am happy to report that everything is now working perfectly, see here<https://allsky.lisabromfield.space/> (it’s another grey and dreary day here…sigh).
I’ve run <./postData.sh –debug> just to be sure and there were no errors.
I had been testing inside but since everything seems OK, I’ve now moved the Allsky back outside into the enclosure.
Thanks again,
Lisa

EricClaeys, 2023-10-28:
Lisa,
When replying via email GitHub deleters attachments so your screenshot got lost.  You have to log into GitHub for attachments. 
Having the new Allsky but older Website explains the `Can't change directory to /allsky.lisabromfield.space/viewSettings: No such file or directory` error since the old release didn’t have a `viewSettings` directory.
This is a perfect example where Allsky should have realized your remote Website was running an old version and told you that, rather than giving a useless message about a missing directory.
I will look into improving that.  Thanks