AllskyTeam/allsky #3046: Fields on overlay editor are messed up

heruan, 2023-10-27:
After configuring our layout, when I go back to the editor all fields are shifter left or right.

This is how it appears live (correct positions):
[image]

This is how the overlay editor appears:
[image]

Many text fields are not in the right position and I always need to move them back when I change something before saving.

Is there something I can do? I looked for other similar issues but I couldn't find any, so I guess I'm missing something here.

AllSky version: v2023.05.01
Chrome version: 118.0.5993.117

Alex-developer, 2023-10-27:
This was fixed in the latest release. Please consider upgrading

Thanks

Alex

heruan, 2023-10-29:
Thanks! I updated to v2023.05.01_03 and it seemed to be fixed, then it happened again for a single field, now it looks like to be more stable.

Not sure if related to the update, but this morning some fields disappeared:
[image]
They are all fields from periodic jobs (Pi Status, OpenWeather, Dew Heater) but the modules are enabled and there are no errors in `allskyperiodic.log` or `allsky.log` either.

How can I debug when a field is not rendered? Are there other log files I'm missing?

EricClaeys, 2023-10-29:
@Alex-developer.  Alex ?

Alex-developer, 2023-10-29:
> Thanks! I updated to v2023.05.01_03 and it seemed to be fixed, then it happened again for a single field, now it looks like to be more stable.
>
> Not sure if related to the update, but this morning [...]

Set the debug level to 4 and have a look in /var/log/allsky.log You will see a whole load of output when the overlay module runs and it will give you more information as to why a field has not been included. Feel free to post an excerpt from the log if you would like me to take a look

Cheers

Alex

heruan, 2023-11-13:
Looks like some periodic modules were disabling, RaspPi Status and OpenWeather. I've set the module timeout to 60 seconds, now it looks stable. Closing this, hope for good!

closed 2023-11-13 by heruan