AllskyTeam/allsky #3059: [BUG]

Croniac, 2023-10-28:
### Environment

* Camera:  ZWO ASI178-MC
* OS: Bullseye
* Allsky version: 2023.05.01_03
* Pi model and amount of memory: 4B 1GB

### Bug Description

I am attempting to upload via GCS.  If I use the upload.sh manually with a test file I have success.  This is my first attempt using the upload functions.

I have "IMAGE_UPLOAD_ORIGINAL_NAME = "false", thus when I look in /allsky/tmp/ if I do find the current image stored there are IMAGE.JPG.    

However when the script runs automatically, it attempts to pick up the original file name from that location and fails as seen in the log copied below.  I'm double confused here because 1) I know the connection is good if manually done.  2) I've told it to upload image.jpg, so why is it messing about with the image-yyymmddhhmmss.jpg version? 

I have tried setting the "IMAGE_UPLOAD_ORIGINAL_NAME = "true" however still do not have success, no file arrives at GCS

Thanks for any help

Oct 28 15:55:39 allsky allsky[2395]: INFO: ----------------------- Running Module allsky_saveimage.py -----------------------
Oct 28 15:55:39 allsky allsky[2395]: INFO: Attempting to load allsky_saveimage
Oct 28 15:55:39 allsky allsky[2395]: INFO: Image /home/croni/allsky/tmp/image-20231028155529.jpg Saved, quality 80
Oct 28 15:55:39 allsky allsky[2395]: INFO: Ignoring watchdog for module saveimage
Oct 28 15:55:39 allsky allsky[2395]: INFO: day flow Complete...
Oct 28 15:55:41 allsky allsky[2407]: upload.sh: Uploading /home/croni/allsky/tmp/image-20231028155529.jpg to gcs allsky/images
Oct 28 15:55:41 allsky allsky[2407]: Upload output from '/home/croni/allsky/tmp/image-20231028155529.jpg:
Oct 28 15:55:41 allsky allsky[2407]:    /home/croni/allsky/scripts/upload.sh: line 178: gsutil: command not found

### Log / configuration files
[Settings-configs.zip](https://github.com/thomasjacquin/allsky/files/13196914/Settings-configs.zip)

Croniac, 2023-10-29:
Using the test example from the documentation, I get the success result shown below... but the script fails when running under allsky as shown above.

~~~
croni@allsky:~ $ echo hi > /tmp/test.txt
~/allsky/scripts/upload.sh --debug /tmp/test.txt /images test.txt
Upload output from '/tmp/test.txt:
   Copying file:///tmp/test.txt [Content-Type=text/plain]...
/ [1 files][    3.0 B/    3.0 B]
Operation completed over 1 objects/3.0 B.

~~~

EricClaeys, 2023-10-29:
It's failing because it can't find the `gsutil` command.
The documentation states:
~~~
NOTE: The gsutil command must be installed somewhere in the standard $PATH, usually in /usr/bin. Make sure you authenticate the cli tool with the correct user as well.
~~~
Did you do that?

Croniac, 2023-10-29:
Yes, and what is interesting about that is that it is in the path... I can enter the GSUTIL command from anywhere I can think of and it finds it just fine, but the script does not.   Also interestingly, it still doesn't respect the IMAGE_UPLOAD_ORIGINAL_NAME = "false" regardless -- So that seems like a bug somewhere that needs addressing also

Here is the path (which was true earlier as well):
~~~
croni@allsky:~/allsky/scripts $ cat $PATH
cat: '/home/croni/.local/bin:/home/croni/google-cloud-sdk/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games': No such file or directory
~~~  

I added the full path to the line in the upload script and the upload works.

EricClaeys, 2023-10-29:
@Croniac, the $PATH variable when you are logged in includes the directory where the `gsutil` command is, but Allsky is using a different $PATH that doesn't include the directory where `gsutil`.
Please run `type gsutil` and let me know what it says.  Then add this line to scripts/upload.sh on line 2:
~~~
echo "PATH=${PATH}"
~~~
Do a `tail -for/var/log/allsky.log` and when you see a line that contains PATH=xxxx let me know what it says.

I will look at the IMAGE_UPLOAD_ORIGINAL_NAME bug.

EricClaeys, 2023-10-29:
For the IMAGE_UPLOAD_ORIGINAL_NAME bug, in `scripts/upload.sh` please add `/${DESTINATION_NAME}` to lines 176 and 178 after the `${REMOTE_DIR}`.
See the similar "s3" code a few lines above.  Then remove the change you made for "image.jpg".
I am 99% sure that will fix that bug because I did the same thing for s3 back in May.

Croniac, 2023-10-30:
Thanks for your help Eric,  i'm all good now.   Also ended up not using gcs in the end for other reasons.  But hopefully your $Destination_name research will help out the next guy who does.

EricClaeys, 2023-10-30:
Issue resolved

closed 2023-10-30 by EricClaeys