AllskyTeam/allsky #3105: [BUG] Overlays disappear if .json file added to /home/pi/allsky/config/overlay/extra with no data or incorrect data

Jonk2, 2023-11-10:
Latest point-4 for testing.

I have some python scripts running that I constantly edit and improve (I'm very much a begineer still). These scripts save a .json file for sensor readings, sun position, darkness levels etc.

I'm working on one at the moment for uptime, which has issues as it's WIP. However, when it writes a .json file, either blank or with some information (the python script stops with an error), the overlay completely disappears until I delete the corresponding .json file. On the next frame, the overlay returns.

Does this mean that there is a bug, or is there a reason that I need to understand?

They way around it for me is to write the files somwhere else until I have a working script, then simply change the file location to /home/pi/allsky/config/overlay/extra.

Alex-developer, 2023-11-10:
Couple of things

1) Don't create blank json files - I suspect thats throwing an error if you look in the logs
2) If the file is not blank are you sure the json is valid?

Jonk2, 2023-11-10:
In the allsky.log, I see multiple lines like this:

2023-11-10T21:39:04.783764+00:00 RORPi allsky[4293]: INFO: Loading Data File /home/pi/allsky/config/overlay/extra/rpi_uptime.json
2023-11-10T21:39:04.783888+00:00 RORPi allsky[4293]: ERROR: Expecting value: line 1 column 1 (char 0)

What other logs should I check? 

I didn't purposely put a blank file there I just noticed the filesize = 0 on one of the tests. I'm trying to learn how to read a file from the pi (uptime and cpu temperature at the moment) to write the value periodically to a json file to not only use on the allsky overlay, but eventually on my own webpage.

When the write_to_json function is called, I'm getting all sorts of errors I'm trying to get my head round (serialization is the main one), ints vs floats vs strings etc. Easy for some, not for me!

EricClaeys, 2023-11-11:
@Jonk2 if you want to test if a .json file is valid you can use the `json_pp` command (or maybe `pp_json`).

Jonk2, 2023-11-11:
Update - I found the issue with my script (typo) and my json file for that particular script is working fine now.

I still don't know why the overlay disappears when a suspect json file is there, but I suppose that's something you clever guys could look at! Also if the json file ages, it certainly makes sense to me to have a default string, -NaN- perhaps rather than that text field disappearing?

Alex-developer, 2023-11-11:
> I still don't know why the overlay disappears when a suspect json file is there

Its a bug :-( I'll fix it in the point 4 release tomorrow

EricClaeys, 2023-11-11:
@Jonk2 when the overlay disappears it usually means an error occurred in one of the .py scripts that we didn't check for and the script died.  If you let us know the message in the log file we can fix it.  Typically the error message is produced every image.

EricClaeys, 2023-11-11:
@Jonk2 can you explain more what your "if the json file ages" comment is referring to?
Is this when a variable is "too old" instead of displaying the last (old) value it doesn't display anything for the value?

Jonk2, 2023-11-11:
In my case yesterday, as above the only error I could see in the allsky.log was:

2023-11-10T21:39:04.783764+00:00 RORPi allsky[4293]: INFO: Loading Data File /home/pi/allsky/config/overlay/extra/rpi_uptime.json
2023-11-10T21:39:04.783888+00:00 RORPi allsky[4293]: ERROR: Expecting value: line 1 column 1 (char 0)

And this script (rpi_uptime.py) was the one I had a typo in.

Basically, it was writing the output of a function, rather than calling the function first and writing the result.

Of course when I changed the json filesave location the problem went away from allsky.

There was a point I wrote 4 integers to the json, ending up with [1, 3 , 40, 33] for example which wouldn't work.

As I say, I'm not a coder, just a dabbler so if I'm making a fuss out of nothing, that's fine, I just like to test and help where possible.

However, if Alex says it's a bug, then at least I found it haha.

Jonk2, 2023-11-11:
> @Jonk2 can you explain more what your "if the json file ages" comment is referring to? Is this when a variable is "too old" instead of displaying the last (old) value it doesn't display anything for the value?

Yes, if a file is older than the default 550s (?) then from what I can tell the entire text field that calls that data disappears.

At least, that's what I think I can see.

Alex-developer, 2023-11-12:
> Yes, if a file is older than the default 550s (?) then from what I can tell the entire text field that calls that data disappears.
>
> At least, that's what I think I can see.

Not quite.

In the json file you can specify the expiry time for each field, if none are defined then the expiry time of the file will be used - Have a look in the dcos there are some examples of this

Alex

Jonk2, 2023-11-12:
Ok, I'll keep an eye on things to see if anything disappears. perhaps it was related to a lot of messing about I've been doing and not actually an issue.
So if the data has expired, could something else rather than the latest older data be shown, otherwise, I wouldn't know it was out of date data? NaN seems to be popular for a lot of other things out there.

EricClaeys, 2023-11-12:
@Jonk2  
> I just like to test and help where possible

And we VERY MUCH appreciate you helping make Allsky better!

EricClaeys, 2023-11-12:
@Alex-developer    I like @Jonk2's idea of having something other than the last value appear for out-of-date data.  Perhaps "OLD" ?  "NaN" really only applies to numbers and many people won't know what it means.

Alex-developer, 2023-11-12:
> @Alex-developer I like @Jonk2's idea of having something other than the last value appear for out-of-date data. Perhaps "OLD" ? "NaN" really only applies to numbers and many people won't know what it means.

I agree. Maybe even grey them out a little as well. I am going to fix the bug that breaks the overlay if a corrupt json file is found so will see how easy it is to add this at the same time

Jonk2, 2023-11-12:
True, 'Not a Number' is not widely known, but my guess is, if anyone saw a NaN on their display, they'd look it up!

Anyway, yes please, something rather than nothing would be great and help people debug their own work (like me pretty much every time I touch it haha)

Alex-developer, 2023-11-15:
@Jonk2 I added a few things to POint-release-4

1) Corrupted json files will now be ignored
2) In the overlay settings there is a new field where yo9u can enter the text to display instead of a variable value if its expired. If this field is blank then the fields will be removed s they are now

I am still testing this but feel free it give it a go

EricClaeys, 2023-11-15:
This is a new one.  I suspect install.sh got corrupted.  Would you please run
~~~
head -30 ~/allsky/install.sh
ls -l ~/allsky/install.sh ~/allsky-OLD/install.sh 
~~~

Jonk2, 2023-11-15:
```
pi@RORPi:~/allsky $ head -30 ~/allsky/install.sh
#!/bin/bash
# shellcheck disable=SC2154             # referenced but not assigned

[[ -z ${ALLSKY_HOME} ]] && export ALLSKY_HOME="$(realpath "$(dirname "${BASH_ARGV0}")")"
ME="$(basename "${BASH_ARGV0}")"

#shellcheck source-path=.
source "${ALLSKY_HOME}/variables.sh"                                    || exit "${ALLSKY_ERROR_STOP}"
#shellcheck source-path=scripts
source "${ALLSKY_SCRIPTS}/functions.sh"                                 || exit "${ALLSKY_ERROR_STOP}"

# This file defines functions plus sets many variables.
#shellcheck source-path=scripts
source "${ALLSKY_SCRIPTS}/installUpgradeFunctions.sh"   || exit "${ALLSKY_ERROR_STOP}"

if [[ ${EUID} -eq 0 ]]; then
        display_msg error "This script must NOT be run as root, do NOT use 'sudo'."
        exit 1
fi

# This script assumes the user already did the "git clone" into ${ALLSKY_HOME}.

# Some versions of Linux default to 750 so web server can't read it
chmod 755 "${ALLSKY_HOME}"                                                      || exit "${ALLSKY_ERROR_STOP}"
cd "${ALLSKY_HOME}"                                                                     || exit "${ALLSKY_ERROR_STOP}"

# PRIOR_ALL_DIR is passed to us and is the location of an optional prior copy of Allsky.
PRIOR_CONFIG_DIR="${PRIOR_ALLSKY_DIR}/config"
PRIOR_CONFIG_FILE="${PRIOR_CONFIG_DIR}/config.sh"
PRIOR_FTP_FILE="${PRIOR_CONFIG_DIR}/ftp-settings.sh"    # may change depending on old version
pi@RORPi:~/allsky $ ls -l ~/allsky/install.sh ~/allsky-OLD/install.sh
-rwxr-xr-x 1 root root 108866 Nov 15 20:06 /home/pi/allsky/install.sh
-rwxr-xr-x 1 pi   pi   108731 Nov 10 17:32 /home/pi/allsky-OLD/install.sh
pi@RORPi:~/allsky $

```

EricClaeys, 2023-11-15:
The "root" is definitely a problem.  The installation runs as "pi", and only "root" (or `sudo`) can change a file owned by root.
Please run `ls -l ~/allsky`.

Do you remember if you used `sudo` to run the `git clone ...` command?
If so, run `sudo rm -fr ~/allsky` then rerun the git clone without sudo.

Jonk2, 2023-11-15:
```
pi@RORPi:~/allsky $ ls -l ~/allsky
total 200
-rwxr-xr-x 1 root root  12682 Nov 15 20:06 allsky.sh
drwxr-xr-x 2 root root   4096 Nov 15 20:04 assets
drwxr-xr-x 4 root root   4096 Nov 15 20:06 config_repo
drwxr-xr-x 9 root root   4096 Nov 15 20:06 html
-rwxr-xr-x 1 root root 108866 Nov 15 20:06 install.sh
-rw-r--r-- 1 root root   1070 Nov 15 20:04 LICENSE
-rw-r--r-- 1 root root   1246 Nov 15 20:04 Makefile
drwxr-xr-x 2 root root   4096 Nov 15 20:04 notification_images
-rw-r--r-- 1 root root   9866 Nov 15 20:06 README.md
drwxr-xr-x 3 root root   4096 Nov 15 20:06 scripts
drwxr-xr-x 5 root root   4096 Nov 15 20:06 src
-rwxr-xr-x 1 root root   2032 Nov 15 20:06 test.sh
-rwxr-xr-x 1 root root   1256 Nov 15 20:04 uninstall.sh
-rwxr-xr-x 1 root root   6965 Nov 15 20:04 upgrade.sh
-rw-r--r-- 1 root root   6622 Nov 15 20:06 variables.sh
-rw-r--r-- 1 root root     15 Nov 15 20:06 version
drwxr-xr-x 2 root root   4096 Nov 15 20:04 website
pi@RORPi:~/allsky $
```

I may have run git clone as sudo, it's a habit because learning Linux and having to be told off every 5 minutes for not using sudo is annoying, so therefore usually easier to sudo everything!

I'll try again without sudo, back in a minute...

Jonk2, 2023-11-15:
Ok, it's now installing without sudo being used.

I know it's easy for Linux users, but for people like me who have used Windows since dos became windows it's tricky to 'migrate' and understand everything.

Not a criticism, but maybe a bit of text in the installation instructions to make it clear not to sudo anything is needed?

Jonk2, 2023-11-15:
Ok, latest version installed but with an error (see below)

I noticed the 'allsky is ready to start' image....will it start automatically if I leave it? I restarted it by saving the settnigs (albeit unchanged)

All settings, config etc look to have copied. The ${....}s are all there on the overlay, but I don't see any new additions for how it handles blank / incompatible jsons, or image / text z-index order.....maybe I  need to llok harde, this is just my initial response.
```

pi@RORPi:~ $ cd allsky
pi@RORPi:~/allsky $ git checkout Point-Release-4
branch 'Point-Release-4' set up to track 'origin/Point-Release-4'.
Switched to a new branch 'Point-Release-4'
pi@RORPi:~/allsky $ ./install.sh
* STARTING INSTALLATON AT Wed 15 Nov 20:36:21 GMT 2023.

********************************************************
*** Welcome to the Allsky Installer - v2023.05.01_04 ***
********************************************************

* ZWO camera found.
* Size of current swap (2048 MB) is sufficient; no change needed.
* /home/pi/allsky/tmp is currently a memory filesystem; no change needed.

*** NOTICE:
**********
The following steps can take up to an hour depending on the speed of your Pi
and how many of the necessary dependencies are already installed.
You will see progress messages throughout the process.
At the end you will be prompted again for additional steps.
**********

* Installing the web server.
* Installing dependencies.
* Preparing Allsky commands.
* Modifying locations for WebUI.
* Setting up WebUI options for ZWO cameras.
* Restoring camera-specific settings files:
*       settings_ZWO_ASI178MC.json
* Making new settings file '/home/pi/allsky/config/settings.json'.
* Executing makeChanges.sh --cameraTypeOnly  --debug 'cameraType' 'Camera Type' 'ZWO' 'ZWO'
* makeChanges.sh: cameraType: Old=[ZWO], New=[ZWO]
Calling capture_ZWO -cc_file '/home/pi/allsky/config/cc.json'
Calling: /home/pi/allsky/html/includes/createAllskyOptions.php --debug
        --cc_file /home/pi/allsky/config/cc.json
        --options_file /home/pi/allsky/config/options.json
        --settings_file /home/pi/allsky/config/settings.json
   Argument debug
cameraType=ZWO, cameraModel=ASI178MC
Camera-specific settings file exists (yes): /home/pi/allsky/config/settings_ZWO_ASI178MC.json.
Using existing /home/pi/allsky/config/settings_ZWO_ASI178MC.json.
Linking /home/pi/allsky/config/settings_ZWO_ASI178MC.json to /home/pi/allsky/config/settings.json.
* Settings files:
    -rw-rw-r-- 2 pi pi 2379 Nov 10 17:52 /home/pi/allsky/config/settings.json
    -rw-rw-r-- 2 pi pi 2379 Nov 10 17:52 /home/pi/allsky/config/settings_ZWO_ASI178MC.json
* Keeping 'en_GB.UTF-8' locale.
* Setting permissions on /var/log/allsky.log and /var/log/allskyperiodic.log.
* Installing PHP modules and dependencies.
* Installing python3-full libgfortran5 libopenblas0-pthread.
* Installing Python_dependencies for :
*    === Package #  1 of 15: [numpy>=1.25.0]
*    === Package #  2 of 15: [opencv-python<=4.7.0.72]
*    === Package #  3 of 15: [Pillow]
*    === Package #  4 of 15: [pyephem]
*    === Package #  5 of 15: [skyfield]
*    === Package #  6 of 15: [astral]
*    === Package #  7 of 15: [pytz]
*    === Package #  8 of 15: [scipy<=1.11.3]
*    === Package #  9 of 15: [paho-mqtt]
*    === Package # 10 of 15: [photutils]
*    === Package # 11 of 15: [astropy]
*    === Package # 12 of 15: [suncalc]
[... 48 lines cut ...]
Copying 'modules' directory had problems.
**********

*     'overlay' directory
*     '/home/pi/allsky-OLD/config/overlay/extra' directory
*     WebUI security settings
*     'remote_configuration.json': NO PRIOR VERSION
*     'config.sh' file, as is
*         CONFIG_SH_VERSION=1, PRIOR=1
*     ftp-settings.sh, as is
*         FTP_SH_VERSION=1, PRIOR=1
* Updating some '/home/pi/allsky/config/config.sh' variables.
* Setting permissions on web-related files.
* Adding pi to www-data group.
addgroup: addgroup with two arguments is an unspecified operation.
* Creating/updating sudoers file.

* Installation is done and Allsky is ready to start.

* Enjoy Allsky!

* ENDING INSTALLATON AT Wed 15 Nov 20:49:42 GMT 2023.

pi@RORPi:~/allsky $
```

Jonk2, 2023-11-15:
Maybe it's me (usually is!) but now the overlay is completely blank...

Trying the sample data in the overlay manager I get this:

[image]

Hoever it is enabled:

[image]

Jonk2, 2023-11-15:
Comparing the allsky-OLD and allsky folders side by side, I can see the following:

1. config/raspap.auth is owned by pi / www-data in the new installation, but by root in the OLD.
2. config/overlay/tmp (the folder) is owned by pi / www-data in the new installation, but by pi / pi in the OLD.
3. config/overlay/extras/ all files are owned by pi / www-data in the new installation, but by pi / pi in the OLD.

If you need anything else, please let me know.

EricClaeys, 2023-11-15:
Jon, I'll let @Alex-developer comment on the modules issue.

For the "addgroup: addgroup with two arguments is an unspecified operation." message, would you please run:
~~~
id pi
~~~
Did you do this install on a new system, or were you upgrading?  And which OS?
I may put a check in install.sh to see if ~/allsky is owned by root, and if so, produce an appropriate message.  That would have saved you a headache.
`sudo` should only be used when it's needed.  All our instructions say when it should be needed.  I'll think about a good message as you mention.  Using sudo when it's not needed may or may not bite you.

EricClaeys, 2023-11-15:
Jon, BTW, I never take suggestions for improvements as criticism.  I realize Allsky is "lacking" in some areas - that's why I got involved in the first place.  Keep on suggesting improvements!!!

Jonk2, 2023-11-15:
```
pi@RORPi:~/allsky/scripts $ id pi
uid=1000(pi) gid=1000(pi) groups=1000(pi),4(adm),20(dialout),24(cdrom),27(sudo),29(audio),44(video),46(plugdev),60(games),100(users),102(input),105(render),106(netdev),995(spi),994(i2c),993(gpio)
pi@RORPi:~/allsky/scripts $
```
This is a newish (couple of weeks tops) bookworm lite and this is the first upgrade from a previous point-4 to today's latest point-4.

No problem, it's hard to convey context when typing online! I'd do more to help if I could but I'm not good enough with linux or scripting, I just bumble through what I can pick up.

EricClaeys, 2023-11-15:
Interesting, install.sh is supposed to add "pi" to the "www-data" group, which is what the web server runs as.  Clearly there's a bug since you got the `addgroup: addgroup with two arguments is an unspecified operation` error message.

I actually like that you're not experienced with Linux and scripting.  You provide valuable feedback that most people in your situation don't.  Alex and I are pretty experienced so it's easy for us to assume everyone is, which is a bad habit.

Alex-developer, 2023-11-15:
> Maybe it's me (usually is!) but now the overlay is completely blank...
> 
> Trying the sample data in the overlay manager I get this:
> 
> [image]
> 
> However it is enabled:
> 
> [image]
> 
> Putting 2 and 2 togetter here, but there was that error during installation about the modules directory...
> 
> ```
> 
> *** ERROR:
> **********
> Copying 'modules' directory had problems.
> **********
> ```

Jon

If you look in the allsky error log I bet there is an error thats stopping the flow runner from running?

I have this issue on a Pi5 and its next on my list to fix

Alex

Jonk2, 2023-11-15:
Well at least I'm uesful for something. I help because I don't want this to go stale again like it did when you got involved.

I know it won't now, but by helping you of course it helps me, win win!

It's the best allsky software I know, and I spent ages trying others before I found this, what 3-4 years ago.

Jonk2, 2023-11-15:
> Jon
>
> If you look in the allsky error log I bet there is an error thats stopping the flow runner from running?
>
> I have this issue on a Pi5 and its next on my list to fix
>
> Alex

Here's the latest entry, it's the same for every image saved since I updated a couple fo hours ago...
```

2023-11-15T22:51:50.188318+00:00 RORPi allsky[19941]: -----
2023-11-15T22:51:50.188557+00:00 RORPi allsky[19941]: STARTING EXPOSURE at: 2023-11-15 22:51:50   @ 30.0 sec
2023-11-15T22:51:50.188660+00:00 RORPi allsky[19941]:     > Histogram set exposure to 30.0 sec
2023-11-15T22:51:50.493664+00:00 RORPi allsky[19941]:   > Image took 307.0 ms to save (average 310.1 ms).
2023-11-15T22:51:50.500703+00:00 RORPi allsky[6573]: saveImage.sh NIGHT /home/pi/allsky/tmp/image-20231115225119.jpg DATE=20231115 TIME=225119 AUTOEXPOSURE=0 sAUTOEXPOSURE= EXPOSURE_US=30000000 sEXPOSURE=30.0 sec AUTOGAIN=0 sAUTOGAIN= GAIN=215 AUTOWB=0 sAUTOAWB= WBR=88 WBB=90 BRIGHTNESS=50 MEAN=0 FULLMEAN=0 TEMPERATURE_C=21 TEMPERATURE_F=70 BIN=1 FLIP=none BIT_DEPTH=8 DARKFRAME=0 eOVERLAY=1 AUTOUSB=1 USB=50
2023-11-15T22:51:51.499839+00:00 RORPi allsky[6592]: Traceback (most recent call last):
2023-11-15T22:51:51.500121+00:00 RORPi allsky[6592]:   File "/home/pi/allsky/scripts/flow-runner.py", line 53, in <module>
2023-11-15T22:51:51.500321+00:00 RORPi allsky[6592]:     import allsky_shared as shared
2023-11-15T22:51:51.500441+00:00 RORPi allsky[6592]:   File "/home/pi/allsky/scripts/modules/allsky_shared.py", line 21, in <module>
2023-11-15T22:51:51.500556+00:00 RORPi allsky[6592]:     import board
2023-11-15T22:51:51.500649+00:00 RORPi allsky[6592]:   File "/home/pi/allsky/venv/lib/python3.11/site-packages/board.py", line 45, in <module>
2023-11-15T22:51:51.500847+00:00 RORPi allsky[6592]:     from adafruit_blinka.board.raspberrypi.raspi_4b import *
2023-11-15T22:51:51.501094+00:00 RORPi allsky[6592]:   File "/home/pi/allsky/venv/lib/python3.11/site-packages/adafruit_blinka/board/raspberrypi/raspi_4b.py", line 6, in <module>
2023-11-15T22:51:51.501239+00:00 RORPi allsky[6592]:     from adafruit_blinka.microcontroller.bcm2711 import pin
2023-11-15T22:51:51.501336+00:00 RORPi allsky[6592]:   File "/home/pi/allsky/venv/lib/python3.11/site-packages/adafruit_blinka/microcontroller/bcm2711/pin.py", line 5, in <module>
2023-11-15T22:51:51.501461+00:00 RORPi allsky[6592]:     from RPi import GPIO
2023-11-15T22:51:51.501648+00:00 RORPi allsky[6592]: ModuleNotFoundError: No module named 'RPi'
2023-11-15T22:51:52.514487+00:00 RORPi allsky[6573]: NUM_IMAGES=7
2023-11-15T22:51:52.514762+00:00 RORPi allsky[6573]: saveImage.sh: Not creating mini timelapse: 1 images(s) left.
```

Alex-developer, 2023-11-15:
You could try

`source ~/allsky/venv/bin/activate`
`pip3 install rpi.gpio`

There is a small chance that will fix it

Jonk2, 2023-11-15:
Here goes...

```
(venv) pi@RORPi:~ $  pip3 install rpi.gpio
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting rpi.gpio
  Using cached https://www.piwheels.org/simple/rpi-gpio/RPi.GPIO-0.7.1-cp311-cp311-linux_armv7l.whl (67 kB)
Installing collected packages: rpi.gpio
Successfully installed rpi.gpio-0.7.1
(venv) pi@RORPi:~ $

```

Jonk2, 2023-11-15:
The flow-runner messages are no longer in the log but the result is the same, no overlay and the same 'make sure it's enabled' message.

Alex-developer, 2023-11-16:
> The flow-runner messages are no longer in the log but the result is the same, no overlay and the same 'make sure it's enabled' message.
> 
> 

Jon

Is it enabled in the main settings, check it's set to overlay and not legacy

Jonk2, 2023-11-16:
I'm at work now but can log in remotely (allsky only, not SSH).

It's shown as module, which has carried over from the update.

[image]

EricClaeys, 2023-12-09:
@Alex-developer, @Jonk2,
is this fixed in Point Release 4?

Alex-developer, 2023-12-09:
Yes invalid json files are now ignored

EricClaeys, 2023-12-09:
Closing.  Issue resolved in Point Release 4.

closed 2023-12-09 by EricClaeys