AllskyTeam/allsky #3150: WebUI formatting problem

MightySpaceman, 2023-11-28:
Platform: Rpi 4B 4gb
Camera: ZWO ASI 120MM
OS: Raspbian/Bullseye

[image]

As the image shows, something is definitely wrong with formatting. Not sure because I don't have much experience with webservers myself, but it doesn't allow any use of the webUI at all.

EricClaeys, 2023-11-28:
Is this the first time Allsky was installed on this Pi?  Did you reboot after the installation?

Alex-developer, 2023-11-29:
Doesn't look like php is being interpreted, There is something wrong with the lighttpd configs. If you can I would reflash the card and reinstall allsky

MightySpaceman, 2023-12-13:
> Is this the first time Allsky was installed on this Pi? Did you reboot after the installation?

Yes. And first time installing lighttpd as well, nothing had run previously.

EricClaeys, 2023-12-13:
@MightySpaceman, please attach `~/allsky/config/installation_logs/install.sh.log` and `~/allsky/config/installation_logs/lighttpd.install.log`.
After that, try reinstalling: rename the current "allsky" to "allsky-OLD" and do another "git clone" then "./install.sh".  It should save all your settings and go relatively fast.

EricClaeys, 2023-12-29:
@MightySpaceman, any update?

EricClaeys, 2024-01-30:
Closing issue - no response from user in over 4 weeks.

closed 2024-01-30 by EricClaeys