AllskyTeam/allsky #3153: Darkened Corner on Images

RLWagner, 2023-11-28:
### Environment

* Camera:  ZWO  ASI678MC
* OS: Bullseye
* Allsky version: v2023.05.01
* Pi 3 Model B Revision 1.2 1GB

### Bug Description
All images have the upper left 1/4 frame (exactly 1/4) significantly darkened (about 25%) relative to the remainder of the frame.  I tried turning off all the annotations (also in the top left corner) and it had no effect.  Shooting images with other software does not show the effect.  This problem has existed since I installed this camera.  It did not occur with a previous ASI120MC.

Richard
[image]

EricClaeys, 2023-11-29:
Can you post an image that more clearly shows the problem?  The image above looks fine to me.
Also, please attach your ~/allsky/config/config.sh file.

RLWagner, 2023-11-29:
I have enhanced the contrast across the difference in sky background in the attached.  The dark area shows much more visibly in the timelapse as it remains stationary while the stars (and clouds, sigh) move.
[image]
And here is my config.sh.
[config.txt](https://github.com/thomasjacquin/allsky/files/13504140/config.txt)

EricClaeys, 2023-11-30:
I see it now.  My first guess would have been a hardware problem, but it works on other software so that's not likely to be the reason.

What other software does it work on?

Is that on the Pi or a PC?

EricClaeys, 2023-12-09:
@RLWagner, any update?

RLWagner, 2023-12-19:
Sorry it's been so long to reply - the weather here has been awful and unsuitable to open the observatory to get a usable daytime image.  The darkened corner does not appear in daytime allsky images, and does not appear in images taken with other software on either RPi or Windows PCs.  It seems to be a problem only with nighttime images from allsky.

EricClaeys, 2023-12-19:
That's real strange.  Try setting `Overlay Method` to "Legacy" to rule out the overlay module.  It's possible you have a rectangle or image in your Overlay that is partially transparent and only shows up on dark images.

H-who, 2023-12-28:
Had a similar issue. I had moved to the ASI178 after utilizing the ASI224. It appeared to be an issue with the software knowing the sensor size. A re-install of allsky with the new camera attached fixed the issue. ~ Clear Skies

EricClaeys, 2023-12-29:
@H-who, when you first started using the ASI178 did you select `Refresh` from the Camera Type field in the WebUI?  That causes Allsky to poll the attached camera to get its sensor size and other capabilities.  That's what installing Allsky does.

H-who, 2023-12-29:
I did, I was on version 3.  Tried rebooting as well. When moving between the overly editors on my 2nd allsky with another 224 on board you could also see an additional variable field on the image that was at the bottom of the smaller square in the image. That suggested to me the software was not reading the image sensor size correctly. I messed around for a bit but besides refreshing the camera I decided to re-flash. In doing so I moved to allsky v4 (did not want to leave that out).

EricClaeys, 2023-12-29:
@H-who, what is the "smaller square in the image"?
When you switch camera models of the same camera brand, Allsky currently uses the same configuration  file which will likely have incorrect screen resolutions.
We need to change this to have different overlay configurations for each camera model, like we do for the allsky settings.

H-who, 2023-12-29:
In the provided images above by OP, you can see a quarter of the image (top left) is essentially a smaller square within the larger image.  Unlike theirs, I also had an overlay variable of $CameraModel (similar) at the bottom of that inlayed smaller square. That is what tipped me off to something in config being screwy. Yes, on the different camera model/resolutions recommendation. I figured it pulled the sensor size information on original install and just was not updating even with the refresh and reboot.

EricClaeys, 2023-12-29:
@Alex-developer,
If a user had a low-resolution camera and replaced it with a higher resolution one but didn't `Refresh` the Camera Type in the WebUI, what impact would that have on the Overlay Editor?

Would they see something like the second screenshot above where there is a darker rectangle whose size is that of the old camera?

EricClaeys, 2023-12-29:
@H-who,
By default the Camera Type and Model are on the overlay at the bottom left, down 1070 pixels for ZWO cameras.  We need to change that since the ASI120, and possibly other cameras, have a resolution smaller than that.
For discussion sake, assume that appeared at the bottom of your images.  When you installed a higher resolution camera those fields would still be 1070 pixels from the top, but would now be above the bottom - the higher resolution the higher they would be.
This is what you saw.  It's an artifact of using pixels for locations rather than, for example, percents or some other unit that scales with the resolution.

EricClaeys, 2025-05-10:
Closing issue.
The Camera Type on the overlay now appears at the top, regardless of the sensor resolution.

closed 2025-05-10 by EricClaeys