AllskyTeam/allsky #3165: Need to identify RPi 5 models

EricClaeys, 2023-12-08:
In the file /home/pi/allsky/html/includes/system.php you could include the Pi5 in the RPiVersion() function. e.g.
~~~
'c04170' => 'Raspberry Pi 5 Model B Rev 1.0 (4 GB)',
'd04170' => 'Raspberry Pi 5 Model B Rev 1.0 (8 GB)'
~~~

@Adler6907, can you think of a better way to get info on the Pi rather than using those codes? They are an administrative headache sine the list keeps growing.

The info we want is the model (3B, 4B, etc.) and ideally the memory and Rev. We can determine the memory at run time but I have no idea how to determine the Rev without using those codes.

@Adler6907, @Alex-developer, can you think of any reason why we should NOT set the Computer field at installation per the above comments, then make the field read-only so we have consistency? That field is only used for display purposes in the WebUI and Allsky Website; we don't make any decisions based on it.

Alex-developer, 2023-12-08:
I would leave it as it is. New Pi's are not released that often and its not hard to add the new model numbers

EricClaeys, 2023-12-16:
Fixed in Point Release 4.

PR #3193 (2023-12-16): Update system.php: add Pi 5 models
fixes #3165
closed 2023-12-16 by EricClaeys