AllskyTeam/allsky #3172: [ENHANCEMENT] Support "Focus" mode

EricClaeys, 2023-12-10:
Add setting in WebUI to Enable Focus Mode:

1. Restart Allsky with the settings listed below.
2. Turn on display of Focus Mertic.
3. Disable all other overlays except for time. Maybe install a temporary json overlay file that only contains those fields.
4. Disable all other post-processing: capture program writes directly to image.jpg (assuming overlay system can read/write that file).  Do not save in allsky/images.  Do not upload.
5. Use 1/2 second (?) fxed  delay. That should be enough time to overlay a couple text fields.
6. Have WebUI Liveview refresh every second (?).
7. WebUI should display "Focus Mode" on the liveview page.
8. Restore everything to prior state when disabling Focus Mode.

EricClaeys, 2023-12-10:
@Alex-developer,
How hard would it be for the module system to support this?
Any other comments?

Alex-developer, 2023-12-28:
@EricClaeys Not really something the module system could do. Feels like it should be a script outside of Allsky

magore, 2024-01-25:
It would be good if this option would disable saves while focusing also

EricClaeys, 2024-01-25:
@magore, Mike, my # 4 above suggests that:  `Do not save in allsky/images`.
It has to save in allsky/tmp in order for the focus metric to be added and the image displayed.  allsky/tmp should be a memory-based filesystem so saving there doesn't impact the SD card.

magore, 2024-01-26:
I meant to say not save in the images folder option - tmp is fine as it is ram based

EricClaeys, 2025-10-22:
In v2025.xx.xx.

closed 2025-10-22 by EricClaeys