AllskyTeam/allsky #3175: [ENHANCMENT] Support IMX462

Dhovin, 2023-12-11:
### Environment

* Camera:  Arducam Pivariety IMX462
* OS: Bullseye 64bit lite
* Allsky version: attempting to install latest version
* Pi model: Pi 4B 8GB

### Bug Description

Attempting to use Starvis camera for night sky captures. 

### Log / configuration files

[IMX462.txt](https://github.com/thomasjacquin/allsky/files/13630398/IMX462.txt)

Dhovin, 2023-12-11:
I have been testing with manual still images. I don't know what i'm doing really but i think i took with with a 5 second exposure. used the below line. Wondering if i made a mistake getting this camera.

`libcamera-still -q 95 -o test.jpg --shutter 5000000 --gain 1 --awbgains 1,1`

[image]

EricClaeys, 2023-12-11:
For the green, the color gains go from 0 to 32 so you likely need to increase them.
Run
~~~
libcamera-still  -v -o test.jpg --shutter 5000000 --metadata meta.txt
~~~
and then upload `meta.txt`.  It should list many of default values.

Dhovin, 2023-12-12:
[meta.txt](https://github.com/thomasjacquin/allsky/files/13642877/meta.txt)
[image]

I also tried different gain settings from 1 to 32 and the image is still very green. any ideas?

Dhovin, 2023-12-12:
[meta10.txt](https://github.com/thomasjacquin/allsky/files/13643009/meta10.txt)
[image]

I found a good balance with adding `--awbgains 1.7,1.7`. It could probably be refined more but it takes me 2 minutes just to get one shot.

EricClaeys, 2023-12-12:
~~~
Available cameras
-----------------
0 : arducam-pivariety [1920x1080 10-bit RGGB] (/base/soc/i2c0mux/i2c@1/arducam_pivariety@c)
    Modes: 'SRGGB10_CSI2P' : 1920x1080 [60.00 fps - (0, 0)/1920x1080 crop]

    Available controls for 1920x1080 SRGGB10_CSI2P mode:
    ----------------------------------------------------
    AeConstraintMode : [0..3]
    AeEnable : [false..true]
    AeExposureMode : [0..3]
    AeMeteringMode : [0..3]
    AnalogueGain : [1.000000..200.000000]
    AwbEnable : [false..true]
    AwbMode : [0..7]
    Brightness : [-1.000000..1.000000]
    ColourGains : [0.000000..32.000000]
    Contrast : [0.000000..32.000000]
    ExposureTime : [14..0]
    ExposureValue : [-8.000000..8.000000]
    FrameDurationLimits : [16666..15534444]
    NoiseReductionMode : [0..4]
    Saturation : [0.000000..32.000000]
    ScalerCrop : [(0, 0)/64x64..(0, 0)/1920x1080]
    Sharpness : [0.000000..16.000000]

{
    "ExposureTime": 5000000,
    "FocusFoM": 1351,
    "AnalogueGain": 1.330000,
    "ColourCorrectionMatrix": [ 1.831948, -0.820011, -0.011936, -0.430259, 1.753510, -0.323250, -0.071906, -0.713727, 1.785643 ],
    "FrameDuration": 5000059,
    "Lux": 0.090005,
    "AeLocked": false,
    "ColourGains": [ 1.000000, 1.000000 ],
    "DigitalGain": 1.891665,
    "ColourTemperature": 4500,
    "SensorBlackLevels": [ 4096, 4096, 4096, 4096 ],
    "ScalerCrop": "(0, 0)/1920x1080",
    "SensorTimestamp": 805369547000
}
~~~

EricClaeys, 2023-12-12:
Run `grep libcamera-still /var/log/allsky.log | tail -1` and post the results.  Usually if it takes a long time to get one shot the camera is trying to do auto exposure or auto gain or auto white balance, where it has to take several pictures.

EricClaeys, 2023-12-12:
Arducam IMX462 support will be in Point Release 4.
Closing this Issue.

closed 2023-12-12 by EricClaeys