AllskyTeam/allsky #3185: [BUG] ERROR: _initialiseSun failed ... and the proposed solution

Adler6907, 2023-12-15:
RPi5, Bookwoorm, branch point release 4

I added a variable `${SUN_DAWN}` in the Overlay Editor and enabled "Include Sun" under "Layout Defaults" in the Overlay Editor Options.
No values were output in the overlay.
The following error was written in allksy.log.
`ERROR: _initialiseSun failed on line 1079 - Sun never reaches 6 degrees below the horizon, at this location.`

In /home/pi/allsky/scripts/modules/allsky_overlay.py the value for `lat` and `lon` was determined too high by a factor of 100 in the `_initialiseSun` function.
```
lat = self._convertLatLon(self._observerLat)
lon = self._convertLatLon(self._observerLon)
```

I divided the value by 100 and the overlay worked correctly and there were no more errors.
```
lat = self._convertLatLon(self._observerLat) / 100
lon = self._convertLatLon(self._observerLon) / 100
```

In both functions, `lat` and `lon` are determined in the same way. I also made the correction there without testing it.
`_initialiseSunOld` and `_initSatellites`

The `_initialiseMoon` function determines `lat` and `lon` differently. This works correctly.

EricClaeys, 2023-12-15:
@Alex-developer, FYI

EricClaeys, 2023-12-15:
@Adler6907, what are your `Latitude` and `Longitude` in the WebUI ?

Adler6907, 2023-12-15:
51.16N and 14.5E

Alex-developer, 2023-12-17:
There must be something else going on here. I have setup my dev server (pi4) to the same lat/lon and timezone and get the following results which are accurate

[image]

Could you please post your ~/allsky/config/settings.json file here so that I can take a look, I will test this on a pi5 tomorrow as well

Thanks

Adler6907, 2023-12-17:
{
    "takeDaytimeImages": "1",
    "saveDaytimeImages": "0",
    "dayautoexposure": "1",
    "daymaxautoexposure": "1000",
    "dayexposure": "500",
    "daymean": "0.5",
    "daymeanthreshold": "0.05",
    "daybrightness": "0",
    "daydelay": "180000",
    "dayautogain": "0",
    "daymaxautogain": "16",
    "daygain": "1",
    "daybin": "1",
    "dayawb": "0",
    "daywbr": "2.5",
    "daywbb": "2",
    "dayskipframes": "5",
    "dayTuningFile": "",
    "nightautoexposure": "1",
    "nightmaxautoexposure": "60000",
    "nightexposure": "3000",
    "nightmean": "0.2",
    "nightmeanthreshold": "0.1",
    "nightbrightness": "0",
    "nightdelay": "1",
    "nightautogain": "1",
    "nightmaxautogain": "16",
    "nightgain": "14",
    "nightbin": "1",
    "nightawb": "1",
    "nightwbr": "3.47",
    "nightwbb": "2",
    "nightskipframes": "1",
    "nightTuningFile": "",
    "config": "[none]",
    "extraArgs": "",
    "saturation": "1",
    "contrast": "1",
    "sharpness": "1",
    "width": "0",
    "height": "0",
    "type": "99",
    "quality": "95",
    "filename": "image.jpg",
    "flip": "0",
    "notificationimages": "1",
    "consistentDelays": "1",
    "timeformat": "%Y%m%d %H:%M:%S",
    "latitude": "51.16N",
    "longitude": "14.5E",
    "angle": "-7",
    "takeDarkFrames": "0",
    "useDarkFrames": "0",
    "locale": "de_DE.UTF-8",
    "debuglevel": "0",
    "useLogin": "1",
    "alwaysshowadvanced": "1",
    "overlayMethod": "1",
    "showTime": "1",
    "showTemp": "0",
    "temptype": "C",
    "showExposure": "1",
    "showGain": "1",
    "showBrightness": "0",
    "showMean": "0",
    "showFocus": "0",
    "text": "Bautzen -Ost-",
    "extratext": "",
    "extratextage": "0",
    "textlineheight": "65",
    "textx": "592",
    "texty": "232",
    "fontname": "0",
    "fontcolor": "255 255 255",
    "smallfontcolor": "0 0 255",
    "fonttype": "0",
    "fontsize": "6",
    "fontline": "1",
    "outlinefont": "0",
    "displaySettings": "0",
    "showonmap": "0",
    "websiteurl": "",
    "imageurl": "",
    "location": "Bautzen -Ost-",
    "owner": "My Name",
    "camera": "RPi HQ",
    "lens": "Fisheye f2.2 - 190\u00b0 1.45mm",
    "computer": "Raspberry Pi 5 8GB",
    "cameraType": "RPi",
    "cameraModel": "HQ",
    "lastChanged": "2023-12-16 10:02:51"
}

Alex-developer, 2023-12-18:
> very much
> 
> { "takeDaytimeImages": "1", "saveDaytimeImages": "0", "dayautoexposure": "1", "daymaxautoexposure": "1000", "dayexposure": "500", "daymean": "0.5", "daymeanthreshold": "0.05", "daybrightness": "0", "daydelay": "180000", "dayautogain": "0", "daymaxautogain": "16", "daygain": "1", "daybin": "1", "dayawb": "0", "daywbr": "2.5", "daywbb": "2", "dayskipframes": "5", "dayTuningFile": "", "nightautoexposure": "1", "nightmaxautoexposure": "60000", "nightexposure": "3000", "nightmean": "0.2", "nightmeanthreshold": "0.1", "nightbrightness": "0", "nightdelay": "1", "nightautogain": "1", "nightmaxautogain": "16", "nightgain": "14", "nightbin": "1", "nightawb": "1", "nightwbr": "3.47", "nightwbb": "2", "nightskipframes": "1", "nightTuningFile": "", "config": "[none]", "extraArgs": "", "saturation": "1", "contrast": "1", "sharpness": "1", "width": "0", "height": "0", "type": "99", "quality": "95", "filename": "image.jpg", "flip": "0", "notificationimages": "1", "consistentDelays": "1", "timeformat": "%Y%m%d %H:%M:%S", "latitude": "51.16N", "longitude": "14.5E", "angle": "-7", "takeDarkFrames": "0", "useDarkFrames": "0", "locale": "de_DE.UTF-8", "debuglevel": "0", "useLogin": "1", "alwaysshowadvanced": "1", "overlayMethod": "1", "showTime": "1", "showTemp": "0", "temptype": "C", "showExposure": "1", "showGain": "1", "showBrightness": "0", "showMean": "0", "showFocus": "0", "text": "Bautzen -Ost-", "extratext": "", "extratextage": "0", "textlineheight": "65", "textx": "592", "texty": "232", "fontname": "0", "fontcolor": "255 255 255", "smallfontcolor": "0 0 255", "fonttype": "0", "fontsize": "6", "fontline": "1", "outlinefont": "0", "displaySettings": "0", "showonmap": "0", "websiteurl": "", "imageurl": "", "location": "Bautzen -Ost-", "owner": "My Name", "camera": "RPi HQ", "lens": "Fisheye f2.2 - 190\u00b0 1.45mm", "computer": "Raspberry Pi 5 8GB", "cameraType": "RPi", "cameraModel": "HQ", "lastChanged": "2023-12-16 10:02:51" }

Thanks, still looking at this as at the moment I cannot make it go wrong !

Alex-developer, 2023-12-19:
Tried it on the pi5 (64bit lite OS) using Berlin Timezone and your settings file. Checking the sun times for Bautzen they match those displayed on the overlay.

Given the message about the sun never reaching 6 degrees below the Horizon there must be something wrong with the lat/lon or sunangle in the settings file, even though it works fine on my pi !

Bit stumped on this one

Adler6907, 2023-12-20:
I had an idea when I read “Berlin Timezone” from you and tried it out straight away.
I always set my RPi to location settings "de_DE.UTF-8". If I set it back to en_GB.UTF-8 everything is fine.

Debug with "de_DE.UTF-8":
```
2023-12-20T10:17:44.854553+01:00 allsky1 allsky[9514]: ERROR: _initialiseSun failed on line 1082 - Sun never reaches 6 degrees below the horizon, at this location.
2023-12-20T10:17:44.854573+01:00 allsky1 allsky[9514]: Latitude: 5116.0 - Longitude: 145.0
2023-12-20T10:17:48.181355+01:00 allsky1 allsky[9625]: self._observerLat= 51.16N
2023-12-20T10:17:48.181408+01:00 allsky1 allsky[9625]: self._observerLon= 14.5E
2023-12-20T10:17:48.181427+01:00 allsky1 allsky[9625]: lat= 5116.0
2023-12-20T10:17:48.181443+01:00 allsky1 allsky[9625]: lon= 145.0
```

Debug with "en_GB.UTF-8":
```
2023-12-20T10:12:56.486786+01:00 allsky1 allsky[18579]: self._observerLat= 51.16N
2023-12-20T10:12:56.486851+01:00 allsky1 allsky[18579]: self._observerLon= 14.5E
2023-12-20T10:12:56.486869+01:00 allsky1 allsky[18579]: lat= 51.16
2023-12-20T10:12:56.486886+01:00 allsky1 allsky[18579]: lon= 14.5
```

So the problem lies in the "float" function in "allsky_shared.py"
def float(val):
```
     if not isinstance(val, str):
         val = locale.str(val)
     val = locale.atof(val)
     return val
```

But I would expect "locale.atof" to work properly.
Maybe something needs to change here in "allsky_shared.py"
```
try:
     locale.setlocale(locale.LC_ALL, '')
except:
     passport
```

Adler6907, 2023-12-20:
I was able to solve the problem in the `_initSatellites` function with my "crutch" in `float`, although I don't understand why.

The code later calls the `float` function from "allsky_shared.py" in `_convertLatLon`.
```
lat = self._convertLatLon(self._observerLat)
lon = self._convertLatLon(self._observerLon)
```

The functions `_initialiseMoon` and `_initialiseSun` work great.

But not in the `_initSatellites` function, even though it is the same code. But if you do the conversion before calling this line
```
satellite = EarthSatellite(tles[1], tles[2], tles[0], ts)
```
my "crutch" works.
Something is being changed in `EarthSatellite` so that 
```
val = locale.atof(val)
``` 
with a comma instead of a period no longer works properly.

Alex-developer, 2023-12-20:
Great spot :-)

I had a feeling it was locale based.

I will work on a fix for all locales

Alex-developer, 2023-12-20:
Added a quick fix.

First test is your coordinates, Locale as de_DE.UTF-8 and entering the lat/lon with a '.' rather than a ','

Initial results look good but I need to test it further

[image]

Alex-developer, 2023-12-21:
Test this using de_DE.UTF-8  and en_GB.UTF-8

Results from de_DE.UTF-8

[image]

Results from en_GB.UTF-8

[image]

Adler6907, 2023-12-21:
Perfect 👍🏻

Alex-developer, 2023-12-21:
> 51.16N and 14.5E

> Perfect 👍🏻

Spotted another bug !!!!

Alex-developer, 2023-12-21:
Fix now added

Alex-developer, 2023-12-22:
Closing this as it will be in the PR4 release which will be out in the next couple of days

closed 2023-12-22 by Alex-developer