AllskyTeam/allsky #3226: [BUG] Image not acquiring with version: 20v2023.05.01_04

ShianMull, 2023-12-23:
### Environment

* Camera:  ZWO  224MC
* OS: RPi OS latest version
* Allsky version: 20v2023.05.01_04
* Pi model and amount of memory RPi 4 B+ V1.5 (4GB)

### Bug Description

Although the ZWO camera was detected OK in initial configuration, no images can be acquired, even though all settings are as required and double checked.

This is a new install of the latest Allsky version with a completely fresh install of latest RPi OS version.  Previous (very old) versions of Allsky have worked fine.

Log file reports an issue with line 276 of allsky.sh:

2023-12-23T13:43:15.466453+00:00 aurora allsky[2480]: ==========
2023-12-23T13:43:15.466687+00:00 aurora allsky[2480]: === Starting daytime capture ===
2023-12-23T13:43:15.466779+00:00 aurora allsky[2480]: ==========
2023-12-23T13:43:15.501429+00:00 aurora allsky[2480]: STARTING EXPOSURE at: 2023-12-23 13:43:15   @ 100.00 ms (0.10 sec)
2023-12-23T13:43:16.004391+00:00 aurora allsky[2358]: /home/aurora/allsky/allsky.sh: line 276:  2480 Bus error               "${ALLSKY_BIN}/${CAPTURE}" -cmd "${RPi_COMMAND_TO_USE}" -debuglevel "${ALLSKY_DEBUG_LEVEL}" -config "${ARGS_FILE}"
2023-12-23T13:43:16.004759+00:00 aurora allsky[2358]: ***
2023-12-23T13:43:16.004857+00:00 aurora allsky[2358]: *** After fixing, restart the allsky service. ***
2023-12-23T13:43:16.004946+00:00 aurora allsky[2358]: ***
2023-12-23T13:43:16.548931+00:00 aurora allsky[2358]:      ***** AllSky Stopped *****

Restarting the RPi has no effect and the Pi is connected to a 5A power supply to ensure there is ample power.

Any thoughts on this?

Many thanks

Mark

bribon2, 2023-12-23:
The same error. White images appear to me during the day (I have not yet been able to test at night), it is as if self-exposure will not work.

bribon2, 2023-12-23:
[image]
This is how I see the images during the day. It is in autoexposure at 100ms. If I set it manually, if I reduce the exposure to less than 7ms I get an error and the system does not work. With 7ms you can see something but I still need less brightness

ShianMull, 2023-12-23:
I’m not seeing day acquisition at 100ms , nor night at all.  It seems to fail upon initial image acquisition….

cheers

Mark

EricClaeys, 2023-12-23:
All,
Sorry about this.  I think this related to the newest ZWO release.

Please set the Debug Level to 4 and check for the "Bus error".  If you see it, please attach the /var/log/allsky.log file.

Then please copy the allsky-OLD/bin/ZWO_capture file to allsky/bin and restart Allsky.

EricClaeys, 2023-12-23:
If you are NOT seeing g the `Bus error`, please open a separate Issue.  The Bus error appears to be happening when the first picture is taken, so if you are getting pictures,  even if they are too bright, you likely have a different problem.

I will investigate the Bus error.  It's strange, because right before releasing Point Release 4 yesterday both Alex and I successfully installed it and are using it.

Alex-developer, 2023-12-23:
Its running fine on my ASI178MM not seen any bus errors, could be camera specific?

EricClaeys, 2023-12-23:
@Alex-developer, could be.  It's running fine on my 290 and 178.

ShianMull, 2023-12-24:
I wondered about camera specific, but it worked with previous versions of Allsky….

The software also had no problems detecting the camera was a 224MC on first install, so it must have interfaced through the camera firmware/hardware to determine this?

I’ll run the debug in the morning UK time and report what I find.

cheers

Mark

duckend, 2023-12-24:
@EricClaeys - you requested a copy of my cc.json file...

[cc.json](https://github.com/thomasjacquin/allsky/files/13761079/cc.json)

EricClaeys, 2023-12-24:
The newest ZWO library, 1.32 was added to Point Release 4 a couple weeks ago.  It added some new items like GPS status.  It initially caused capture_ZWO to crash because it was trying to access, for example, item 20 in an array that only had 15 entries.
I'm assuming the `Bus error` is doing something similar which is why the problem didn't occur in the last release and didn't occur if you had a beta version of Point Release 4.

@duckend, thanks for the cc files.  They are the same.  Please set the Debug Level to 4 and check for the "Bus error". If you see it, attach the `/var/log/allsky.log` file.

Then please copy `~/allsky-OLD/bin/capture_ZWO` to `~/allsky/bin` and restart Allsky.  It should work.

duckend, 2023-12-24:
@EricClaeys - requested allsky.log attached - Bus error happens immediately

[allsky.log](https://github.com/thomasjacquin/allsky/files/13761226/allsky.log)

EricClaeys, 2023-12-24:
Folks,
Please copy the attached files (after removing the ".txt" from the end of their names) to `~/allsky/src`, then run:
~~~
cd ~/allsky/src
sudo systemctl stop allsky
make capture_ZWO
cp capture_ZWO ../bin
sudo truncate -s 0 /var/log/allsky.log
# Set Debug Level to 4 in the WebUI and press the "Save changes" button.
~~~
It will likely stop again with the `Bus error`.  Please attach `/var/log/allsky.log`.  I added some debugging statements to the program to get a better idea where it's stopping.

Thanks - Eric

[ASI_functions.cpp.txt](https://github.com/thomasjacquin/allsky/files/13761305/ASI_functions.cpp.txt)
[capture_ZWO.cpp.txt](https://github.com/thomasjacquin/allsky/files/13761308/capture_ZWO.cpp.txt)

duckend, 2023-12-24:
Rebuilt and new log attached

[newallsky.log](https://github.com/thomasjacquin/allsky/files/13761335/newallsky.log)

EricClaeys, 2023-12-24:
@duckend, thanks much.  I will have updated file with more debugging statements tomorrow.

ShianMull, 2023-12-24:
Hi Eric,

Well, I just booted up the RPi to make the changes and I have not had a bus error this morning!  It seems to be capturing 'bad images', which others had reported, but I'm now stumped as to what has happened to suddenly make the system apparently work - I've touched nothing since trying all day yesterday.

Anyway, I've installed the extra txt files and attach the log file as requested in case it's going to help this issue.

Festive greetings!

Cheers

Mark
[allsky.log](https://github.com/thomasjacquin/allsky/files/13761361/allsky.log)

ShianMull, 2023-12-24:
Ah ha!!

OK, so the only difference between yesterday and today is that I left the lens cap on the camera this morning.  I just took it off to try and adjust exposure settings and it yet again gave bus error.

I've attached the latest log file with the lens cap off to give the bus error.

[allsky.log](https://github.com/thomasjacquin/allsky/files/13761387/allsky.log)

ShianMull, 2023-12-24:
<just to confirm - lens cap on, no error, lens cap off, bus error.  repeatable!>

ShianMull, 2023-12-24:
Can you attached a copy of the previous capture_ZWO as this is a new build and I don't have the immediately preceding capture_ZWO.

Cheers

duckend, 2023-12-24:
@ShianMull - I suspect you will need to build it on your pi, but git has all the previous variants available to you.

I'm sure there are more efficient ways to achieve it. The below will clone the repository into a new temp folder and the checkout command rolls it back in time to just before Eric introduced the new ZWO sdk.

```
cd ~
mkdir tempcopy
cd ~/tempcopy
git clone https://github.com/thomasjacquin/allsky.git
cd ~/tempcopy/allsky
git checkout ff0c09008c0fffb3009efd9c68100f8c490554a8
cd ~/tempcopy/allsky/src
sudo systemctl stop allsky
make capture_ZWO
cp capture_ZWO ~/allsky/bin
cd ~

```

Restart allsky and if it is now capturing images the temporary directory "tempcopy" can be removed.

ShianMull, 2023-12-24:
Thanks Duckend,  Apologies, I was just after the previous ZWO library to recompile on the Pi, but requested the compile capture module instead.  Perhaps you're right that I'll just download the previous version and run from there...

Festive greetings,

Mark

Pswoboda1, 2023-12-24:
[allsky.log](https://github.com/thomasjacquin/allsky/files/13762184/allsky.log)

bus error for me also. i think I uploaded the allsky.log correctly.

This is all new to me, so be gentle.

thx,
Pat

EricClaeys, 2023-12-24:
@Pswoboda1, did you unplug the camera around 6:53 pm and plug it back in around 8:21 pm?  The ASI_ERROR_CAMERA_REMOVED messages indicate that.  Or the USB cable was loose.

Pswoboda1, 2023-12-24:
I confirmed also. Lens cap on, no bus error. go to try to build with the previous capture_ZWO

EricClaeys, 2023-12-24:
@All, please use this file - same instructions as 6 hours ago.  A couple of you had the `Bus error` after **xxx Getting ASI_BANDWIDTHOVERLOAD** so I added more debugging statements after that.
Thanks for helping debug this!!

[capture_ZWO.cpp.txt](https://github.com/thomasjacquin/allsky/files/13762246/capture_ZWO.cpp.txt)

Pswoboda1, 2023-12-24:
@EricClaeys I could have. I disconnected and plugged it into my computer to make sure the camera was working.

I just did what @ShianMull did and put the lens cap on and rebooted (and set bad images to false). It will take images with the lens cap on, but when I take the cap off I get the bus error.

EricClaeys, 2023-12-24:
@Pswoboda1,
Please use the most recent file I uploaded and stop Allsky, then run `sudo truncate -s 0 /var/log/allsky.log`, then put the lens cap on and start Allsky.  After you get a picture remove the lens cap.  When it gives `Bus error`, upload the log file.   I'm very curious to see what the difference is in the code between lens cap on and off.

duckend, 2023-12-24:
Hi @EricClaeys new log file...

[allsky_202312241600.log](https://github.com/thomasjacquin/allsky/files/13762285/allsky_202312241600.log)

EricClaeys, 2023-12-24:
@Pswoboda1, by the way, Welcome to Allsky!

EricClaeys, 2023-12-24:
@duckend, thanks.  Please try this file.  Appreciate the help.

[capture_ZWO.cpp.txt](https://github.com/thomasjacquin/allsky/files/13762318/capture_ZWO.cpp.txt)

duckend, 2023-12-24:
No problem - you are doing all the work, new bits from log will attach log in a minute

```
2023-12-24T16:09:45.961376+00:00 allsky allsky[67382]: xxx Getting ASI_WB_R
2023-12-24T16:09:45.961494+00:00 allsky allsky[67382]: xxx Getting ASI_WB_B
2023-12-24T16:09:45.961596+00:00 allsky allsky[67382]: xxx Getting ASI_BANDWIDTHOVERLOAD
2023-12-24T16:09:45.961689+00:00 allsky allsky[67382]: xxx ASI_BANDWIDTHOVERLOAD returned ASI_SUCCESS (0), status=ASI_SUCCESS (0)
2023-12-24T16:09:45.961809+00:00 allsky allsky[67382]: xxx takeOneExposure returning 0x0
2023-12-24T16:09:45.984943+00:00 allsky allsky[67254]: /home/iain/allsky/allsky.sh: line 276: 67382 Bus error               "${ALLSKY_BIN}/${CAPTURE}" -cmd "${RPi_COMMAND_TO_USE}" -debuglevel "${ALLSKY_DEBUG_LEVEL}" -config "${ARGS_FILE}"
2023-12-24T16:09:45.985221+00:00 allsky allsky[67254]: ***
2023-12-24T16:09:45.985343+00:00 allsky allsky[67254]: *** After fixing, restart the allsky service. ***
2023-12-24T16:09:45.985447+00:00 allsky allsky[67254]: ***
2023-12-24T16:09:46.575901+00:00 allsky allsky[67254]:      ***** AllSky Stopped *****

```

Pswoboda1, 2023-12-24:
@EricClaeys, see attached. thanks for the help!

[allsky.log](https://github.com/thomasjacquin/allsky/files/13762360/allsky.log)

duckend, 2023-12-24:
We have just transitioned to night mode which has changed the behaviour seen in the logs a bit although it still results in a bus error. Attached in case it helps.

(Note this wasn't truncated from previous log so ignore first run which is still daytime)
[allsky_202312241639.log](https://github.com/thomasjacquin/allsky/files/13762374/allsky_202312241639.log)

ShianMull, 2023-12-24:
Latest log file attached with Eric's latest debug version.  I seemed to get an error with lens cap on this time, but running with images when lens cap off - I think this might be a switch from daytime to nighttime behaviour, but it's an odd one if so.

[allsky.log](https://github.com/thomasjacquin/allsky/files/13762386/allsky.log)

This shows some symptoms of perhaps a pixel bit depth definition error in the ZWO driver, or perhaps in the way the bit depth is read into Allsky routines?  If there is a mismatch in reading the per pixel values, it might also explain the pedestal giving white-outs that some have experienced - just a guess?

EricClaeys, 2023-12-24:
Another one to try.  There are two places Allsky tries to capture an image.  It appears that one of them is causing the problem so I changed the debugging to make it more obvious which one is which.
When posting the log file, please mention `VERSION 5`.

[capture_ZWO.cpp.txt](https://github.com/thomasjacquin/allsky/files/13762419/capture_ZWO.cpp.txt)

EricClaeys, 2023-12-24:
@ShianMull,
What specifically in the log file makes you think there's possibly a pixel bit depth definition error?  I don't see anything that would indicate that.  FYI, the camera is reporting a MAX bit depth of 12 bits per pixel but since you are generating a color image, each red/green/blue pixel is only 8 bits.

In your second run of Allsky in your log file at nightime the 10 second exposure produced an image with a mean of 0 (pure black).  I assume that's with the lens cap on.  However, the image was rejected because its **MEAN=99.82** was greater than the threshold of **90**.  There's likely a bug there, but that's a different problem from the Bus error.  It should have been rejected because the mean of 0 was less than the threshold of 0.1.

duckend, 2023-12-24:
Version 5 log attached

[allsky_202312241727.log](https://github.com/thomasjacquin/allsky/files/13762448/allsky_202312241727.log)

Pswoboda1, 2023-12-24:
[allskyVERSION5.log](https://github.com/thomasjacquin/allsky/files/13762457/allskyVERSION5.log)

VERSION 5 attached.

EricClaeys, 2023-12-24:
Version 6.
Please copy the attached file (after removing the ".txt" from the end of their names) to ~/allsky/src, then run:
~~~
cd ~/allsky/src
sudo systemctl stop allsky
make capture_ZWO
cp capture_ZWO ../bin
sudo truncate -s 0 /var/log/allsky.log
sudo systemctl start allsky
~~~

[capture_ZWO.cpp.txt](https://github.com/thomasjacquin/allsky/files/13762483/capture_ZWO.cpp.txt)

Pswoboda1, 2023-12-24:
[allskyVERSION6.log](https://github.com/thomasjacquin/allsky/files/13762497/allskyVERSION6.log)

duckend, 2023-12-24:
Version 6

[allsky_202312241751.log](https://github.com/thomasjacquin/allsky/files/13762499/allsky_202312241751.log)

ShianMull, 2023-12-24:
Nothing specific in the log file - I have seen problems in other imaging equipment when a device driver has a bug where value output from a CCD/CMOS array has bits shifted in register.  Has led to interesting effects such as pedestals and out of range values.You’re correct that the 2nd run is with lens cap on.I will try V6 asap, but must now join the family for Christmas Eve.Merry Christmas, everyone,Mark

EricClaeys, 2023-12-24:
@Pswoboda1, interesting that you had 2 successful images and then got `Segmentation fault` (not `Bus error`, but very similar) on the 3rd one and all 3 pictures were at 0.1 second but the first two had a mean of 0 and the 3rd one had a mean of 185.
Please try again with a daytime "Max Auto-Exposure" of 10 seconds?

@duckend, you got a `Bus error` on the first image of 10 seconds with a mean of 0.  Can you try again but set your nighttime "Manual Exposure" to 60 and nighttime "Max Auto-Exposure" to 60?

In both your cases you are trying to take a picture with a duration longer than your Max.  That could be related to the problem.  I should add a check for that.

Pswoboda1, 2023-12-24:
[allskyVERSION6v2.log](https://github.com/thomasjacquin/allsky/files/13762673/allskyVERSION6v2.log)

i dropped the exposure down, but still getting the error.

duckend, 2023-12-24:
Revised run with 60 on manual & max exposure. Now getting a Seg Fault but at the same spot the bus error was previously occurring.

[allsky_202312241951.log](https://github.com/thomasjacquin/allsky/files/13762701/allsky_202312241951.log)

duckend, 2023-12-24:
@EricClaeys I think I may have discovered the cause of the memory corruption, it was in the computeHistogram function...

```
	// For RGB24, data for each pixel is stored in 3 consecutive bytes: blue, green, red.
	// For all image types, each row in the image contains one row of pixels.
	// currentBpp doesn't apply to rows, just columns.
//x int on = 0;
//x static int did = 0; did++;
	switch (cg.imageType) {
	case IMG_RGB24:
	case IMG_RAW8:
	case IMG_Y8:
		for (int y = roiY1; y < roiY2; y++) {
			for (int x = roiX1; x < roiX2; x += currentBpp) {
				int i = (cg.width * y) + x;
				int avg = buf[i];
				if (cg.imageType == IMG_RGB24) {
					// For RGB24 this averages the blue, green, and red pixels.
*******				avg += buf[i+1] + buf[i+2];
				}
//x if (useHistogramBox && did <=5) { printf("avg[%d]=%d\n", ++on, avg); }
******			histogram[avg]++;
			}
		}
		break;

```

avg should be divided by 3 before being used as in index into the histogram array.

I've tried a local change and this stops the Seg Fault/Bus Error. Im not sure the exposure calc is running correctly but that is a problem for another day.

Merry Christmas all.

Alex-developer, 2023-12-24:
@EricClaeys @duckend 

Just tried the same 'fix' and it does stop the bus error.

I agree with @duckend that we need to look at the histogram code.

EricClaeys, 2023-12-25:
@duckend, thanks.  FYI, your nighttime max is 60 ms (0.06 seconds).  In the WebUI you need to enter `60000` to get 60 seconds since the input is in ms (1000 ms = 1 second).  This shouldn't impact the bug though.

EricClaeys, 2023-12-25:
@duckend,  **You 'da man!!**
I changed the histogram function a week or so ago, and after looking at the prior version I see that I left out the divide by 3.

**@All: please use this (hopefully fixed and final) version 7.**

I think the reason that changing the exposure time changed the error from `Bus error` to `Segmentation fault` is because adding the brightness of 3 low numbers (red, green, blue) gave a low number that didn't overrun the array, but increasing the exposure gave three higher numbers which when added was greater than the size of the array, so it overwrote whatever happened to be after the array in memory.

[capture_ZWO.cpp.txt](https://github.com/thomasjacquin/allsky/files/13763874/capture_ZWO.cpp.txt)

EricClaeys, 2023-12-25:
@duckend,
I still think there is a lingering bug somewhere that causes some people with ZWO cameras (especially the ASI120) to get a lot of ASI_ERROR_TIMEOUT's which eventually stop Allsky.  It seems like every new release impacts different people, even if we only make a very small change to the capture program.  If memory is being overwritten somewhere else that could easily explain the changes.

I'd REALLY appreciate it if you could look at capture_ZWO.cpp, allsky_common.cpp, and ASI_functions.cpp for any other possible bugs.  I've looked several times but never found anything.

ShianMull, 2023-12-25:
Hi Eric, Sorry to report that the latest capture_ZWO.cpp has not cured the problem with the original issue.

[allsky.log](https://github.com/thomasjacquin/allsky/files/13765444/allsky.log)

I ran the new .cpp with lens cap on, then removing it, and then starting Allsky to replicate the error.

ShianMull, 2023-12-25:
Duckend, Alex, just a quick question - are you still using  ~/allsky-OLD/bin/capture_ZWO from ver3 as Eric previously suggested changing, or reverted to the capture_ZWO included in Ver4?

duckend, 2023-12-25:
@ShianMull - Can you try building the capture_ZWO application again - your log file doesn't have any "xxxxxx takeOneExposure() returning.." entries that would be output by the latest code that Eric posted. It is also still posting log entries that have been removed from the latest source.

duckend, 2023-12-25:
Version 7 which is 5 posts up @ShianMull - build it in the new ~/allsky/src directory

ShianMull, 2023-12-25:
Sorry - having a senior moment - forgot to rebuild!

ShianMull, 2023-12-25:
Update - capture in daytime now looks to be working!  Many thanks All, and I’ll give it a go capturing through a whole day-night cycle.

Merry Christmas,

Mark

Pswoboda1, 2023-12-25:
It's working for me also. Thanks everyone for all the help!

Merry Christmas,
Pat

EricClaeys, 2023-12-26:
Thanks for testing everyone.
I will remove the Debug lines and post the update to the master branch.  Alex will need to approve when he wakes up.

closed 2023-12-26 by Alex-developer

EricClaeys, 2023-12-26:
@all, to get the official fix, simply re-install Allsky.
Stop Allsky
cd
mv allsky allsky-OLD
Run the git clone...
cd allsky
./install.sh 

It should only take a few minutes.

PR #3238 (2023-12-26): Capture fixes: bug fix
Fixes #3226 

Fix bug where histogram value wasn't divided by 3 for color images, so value could be greater than 255, thereby modifying memory outside the array.