AllskyTeam/allsky #3249: [ENHANCEMENT] Use combined startrails and keogram program

EricClaeys, 2023-12-29:
@ckuethe created a draft of a combined startrails and keogram program that would read the images once and produce both a startrails image and a keogram image at the same time, thereby decreasing the overall processing time.

EricClaeys, 2024-02-12:
Also allow a user-defined time range to be used, e.g., 4 pm to 6 pm, or more likely, "daytime" or "nighttime".

This could be implemented using `--from <start_time>` and `--to <end_time>` options.
If no `--to` is given, all images after `<start_time>` until the last image in the directory would be used.
If no `--from` is given, all images starting at the first one in the directory up to and including `<end_time>` would be used.

A new setting called "Include Images" could be used to specify what to include:
* All images
* Day only
* Night only
* Custom times (which would show "from" and "to" fields)

`sunwait` could be used to get the day/night transition times for "Day only" and "Night only"

ssagerian, 2024-02-12:
I guess always  related the start_hour to be the start of dusk and the end_hour to be the start of dawn in keeping with the nomenclature of the documentation.

Also, it might make for a cleaner keogram if we had the option to cut off the upper and lower 20pixs of each strip since this is typically ground lighting or ground images. so a normal image that might be 760pix height would end up being 720 after trimming.
maybe --trim or --trim_upper --trim_lower or --trim 20 --tu 20 --tl 10 etc

ssagerian, 2024-02-12:
ah ok, thanks for the clarification on Angle, works for me. 

I suggested that the trim feature just for the keograms, are you saying trim the actual images in the image folder prior to running the keogram program?

EricClaeys, 2024-02-12:
@ssagerian, Steve, yes, that's what I was asking about.  The "CROP" settings in config.sh could be used to get rid of the unwanted parts before the image is saved in the allsky/images folder, but it would impact startrails, timelapse, and of course the current image.

ssagerian, 2024-02-12:
right, I was only proposing this feature for the keogram, not wanting to impact other images.  I archive my daily images after a couple of days and hope to use them for future projects so cropping some of the raw data (as in image-XXXXXXX.jpg files) would make those projects complicated. Adding the feature gives someone the flexibility to do so if they actually notice the issue and feel its a problem. For others, they may never even notice. If I am alone in this opinion, then its ok to drop it from the issue list. I rewrote the keogram.cpp program, convert it to python and can make it an experimental module for my own use.

EricClaeys, 2025-05-10:
Closing.
This is so low of a priority we'll never get to it.

closed 2025-05-10 by EricClaeys