AllskyTeam/allsky #3262: [BUG] "Fatal: index-pack failed' during Repository Clone

rdwhynot, 2023-12-30:
# Is this a QUESTION or FEATURE REQUEST?

&nbsp; 
## Look in the Wiki before submitting this Issue
The Wiki has lots of information including troubleshooting tips for the vast majority of known issues.

 &nbsp; 

### Environment

* Camera:  ZWO  or  RPi - include the model  RPi HQ camera
* OS: Buster or Bullseye   Bullseye
* Allsky version:
     * The newest software includes a file called  ~/allsky/version   Latest version on the GIT Repository
     * If it does not exist, run:       grep Software /var/log/allsky.log
* Pi model and amount of memory (512 MB, 1, 2, 4, or 8 GB)  PI 4 4GB 

### Bug Description
 During the install process on my Raspberry Pi 4 I was cloning into Allsky.

It moseyed along for a while and then I received the error that is attached in the screenshot.
Please advise my next steps.

If so, please add a Discussion item by clicking on the Discussions link above;  do NOT create an Issue. 

* Include a clear and concise description of what the bug is.
* Can it be reproduced?  If so, how?
* Did anything change?  Any settings?  Most issues occur after a change was made.
* Please include the exact messages or a screenshot.
* If this is a configuration-related issue (or you are not sure), attach these files, appending ".txt" to their names first:
  * Allsky problems:
    * ~/allsky/config/config.sh
    * ~/allsky/config/settings.json
    * ~/allsky/config/ftp-settings.sh (only for upload-related problems)
  * Allsky Website problems:
    * ~/allsky/html/allsky/configuration.json (local Website)
    * ~/allsky/config/remote_configuration.json (remote Website)

If you have program output or multi-line messages to include,
add it like this so it formats correctly (note the lines with tildas):

~~~
output/message line 1
output/message line 2
~~~

### Log / configuration files
No log files as I didn't get past the GIT Clone process
[image]

Follow the instructions for Reporting Issues in the Wiki, then ATTACH a copy of the file(s) above.
Do NOT copy/paste them into this Issue.

EricClaeys, 2023-12-30:
@rdwhynot,  Since this happened before Allsky was installed, it's not an Allsky issue.  Sometimes GitHub hiccups or your network went down.  Please remove the "allsky" directory and try the `git clone` again.

Yesterday I had to try "git clone" several times on one of my test machines and had to reboot it to get it to work.  I assume the Pi was in a funky state.

rdwhynot, 2023-12-30:
You can close this issue.   I figured out the issue, my network connection
wasn't good.   I rebooted and changed the cable and the Git Clone worked
fine.

Thanks, and Happy New Year

EricClaeys, 2023-12-31:
@rdwhynot, glad you figured it out.
Hope you don't mind, but I'm using your screenshot in [a new Issue](https://github.com/AllskyTeam/allsky/issues/3264) to add a Troubleshooting documentation page for `git` problems.

closed 2023-12-31 by EricClaeys