AllskyTeam/allsky #3317: [BUG] If a field is outside the image nothing is indicated in overlay manager

EricClaeys, 2024-01-14:
There log file contains `ERROR: Field 'ZWO ' is outside of the image` but nothing in the Overlay Editor to indicate that **OR to fix it**.  The user has to know which .json file to fix and manually do so.

morrowwm, 2024-01-19:
I had this issue also. While I was learning how to use the overlay editor, I think I put ${CAMERA_TYPE} ${CAMERA_MODEL} at a huge "y" position. Or perhaps something put the item there by default?

The error log reported:
```
ERROR: Field 'RPi ArduCam_16_MP' is outside of the image
```
I found this configuration in `allsky/config/overlay/config/overlay.json`, with "y" value of 3008. I changed that to 200, then the item showed up in the overlay editor, where I could delete it.

You could test this by typing in an off-image value in the x or y Position fields.

I added the file to the error message. Is it only ever `_overlayConfigFile` which can have these items?
```
    652             if outOfBounds:
    653                 s.log(0, f"ERROR: Field '{fieldLabel}' is outside of the image. Check config in {self._overlayConfigFile}")  
```

EricClaeys, 2024-01-20:
@morrowwm, the default position for CAMERA_TYPE, CAMERA_MODEL is fixed in the next release.  The current location assumed it was on an RPi HQ camera or a high-resolution ZWO camera.

This Issue is to have the Overlay Editor tell you when a field is outside the image and to put it somewhere in the image so you can fix it or delete it.  That should eliminate the need to change the `outOfBounds` message, although I suppose it should say "Fix in the Overlay Editor".

morrowwm, 2024-01-20:
OK, good to know.

Alex-developer, 2024-01-22:
Fix committed to the next release branch