AllskyTeam/allsky #3327: [BUG] The environment variable 'DELAY' is not available the first time a perodic module is run

Adler6907, 2024-01-18:
During the first run of a periodic module, the environment variable 'DELAY' cannot be retrieved with `s.getEnvironmentVariable("DELAY")`.

The problem lies in 'periodic.sh'. The variable is described there after the module is called for the first time.

incorrect:
```
while :
do
	activate_python_venv
	python3 "${ALLSKY_SCRIPTS}/flow-runner.py" --event periodic
	deactivate_python_venv

    DELAY=$(jq ".periodictimer" "${ALLSKY_MODULES}/module-settings.json")
....
```

correct:
```
while :
do
    DELAY=$(jq ".periodictimer" "${ALLSKY_MODULES}/module-settings.json")
	activate_python_venv
	python3 "${ALLSKY_SCRIPTS}/flow-runner.py" --event periodic
	deactivate_python_venv
....
```

Alex-developer, 2024-01-19:
Thanks, I'll change it so the variable is available before the first periodic flow is run

Alex-developer, 2024-01-20:
This has been done and added to the next release

closed 2024-01-20 by Alex-developer