AllskyTeam/allsky #3336: Overlay Editor does not accept any changes[BUG]

4358haltern, 2024-01-22:
Hello
I've already made several attempts to place the text fields in a different place...
Unfortunately the settings are not saved.

[image]

Alex-developer, 2024-01-22:
Can you check the lighttpd error log in  /var/log/lighttpd/error.log and see if there are any errors logged when you try and save the overlay

4358haltern, 2024-01-24:
I need to sort out my network problems and check it out on the weekend

4358haltern, 2024-01-29:
`2024-01-28 00:00:19: (server.c.1848) logfiles cycled UID = 0 PID = 38833`

4358haltern, 2024-01-29:
[image]

4358haltern, 2024-01-29:
[image]

4358haltern, 2024-01-29:
I think I've solved it. I've switched the modules on/off, now it's running

EricClaeys, 2024-06-17:
Closing issue.  User resolved it.

closed 2024-06-17 by EricClaeys