AllskyTeam/allsky #3378: [BUG] terminate called after throwing an instance of 'cv::Exception'

EricClaeys, 2024-01-29:
### Discussed in https://github.com/AllskyTeam/allsky/discussions/3373

<div type='discussions-op-text'>

<sup>Originally posted by **ssagerian** January 24, 2024</sup>
So, each day, at the night to day transition and again at the day to night transition, I get the same crash.
Daytime bin is 1x1 and nighttime is 2x2.

capture_RPi is crashing at the day/night transitions when the image sizes change.

</div>

ssagerian, 2024-01-29:
ok, 1x1 bining is on for both day and night

ssagerian, 2024-01-29:
hum..I think you are on to something. Ok last night I disabled overlays. 
This morning I turned them on, the overlays looked normal sized as expected. 
Then,  when I read your email, I configured bining to 1x1 for both night and day, but THEN the overlays were shrunk and re-positioned in the upper left corner. 
When I set the bining to 2x2 the images returned to normal size..that might be a clue for Alex wrt to the overlay module work he is doing?

EricClaeys, 2024-01-29:
@ssagerian, I assume you didn't get the crash when the binnings were the same?

A future version of Allsky will allow multiple overlay configurations so you can have one for daytime with 1x1 bin and one for nighttime with 2x2 bin and the size and placement of overlay fields will be correct.

If you haven't notices, keograms and startrails won't work unless all the images in them are the same dimensions, so binning different during day and night will break keograms and startrails.  Not sure what it'll do to timelapse videos.
A more future version of Allsky will allow different sized images with keograms and startrails.

ssagerian, 2024-01-29:
all the overlay images need to be the same size? If you only have one image?

Looks like that change worked, no crash or shutdown this morning. Thanks for that!

EricClaeys, 2024-01-30:
Steve,
This file should fix the problem.  I could reproduce it with different bins and the fix works.

Please rename the file to `mode_mean.cpp` and put it in `~/allsky/src`, then run
~~~
cd ~/allsky/src
sudo systemctl stop allsky
make
cp capture_RPi ../bin
sudo truncate -s 0 /var/log/allsky.log
#  In the WebUI lower the Debug Level to 3 or whatever you normally use.  Then change night binning to 2 and press "Save changes"
~~~

Let me know how it goes.
FYI, the fix was to create a new auto-exposure mask every time the image size changes, which is the first image plus at every day/night transition.  If the image size and mask size are different it gives that exception.
[mode_mean.cpp.txt](https://github.com/AllskyTeam/allsky/files/14093244/mode_mean.cpp.txt)

closed 2024-05-12 by EricClaeys