AllskyTeam/allsky #3423: Documentation deadlinks

troublestarter, 2024-02-12:
Hi

I think there is some deadlink in the documentation

For exemple : on the main page:
https://htmlpreview.github.io/?https://raw.githubusercontent.com/thomasjacquin/allsky/master/html/documentation/index.html

When you click on installing or configuring
[image]

You arrive on such page:
[image]

Can you confirm me it should not do that so i will reference below all deadlink.

Thanks

EricClaeys, 2024-02-13:
Thanks for reporting it.  I fixed the Wiki page on GitHub which still pointed to the old GitHub page which no longer exists and caused all the links to still point there.

I'll leave this Issue open for a while in case you find more dead links.
FYI, we are considering getting rid of the Wiki so users would only be able to view the documentation via the "Allsky Documentation" link in the WebUI, so please use that when reviewing the documentation.  Thanks.

troublestarter, 2024-02-13:
So if i understand clearly, you want the users to be able to use "Allsky Documentation" link in the **local** WebUI from the Pi ?

But note that also:
On the main url : https://github.com/AllskyTeam/allsky
The link to the documentation point to the old documentation with deadlinks ;) - Too much confusing
[image]
So from here, INSTALLING and CONFIGURING always deadlinks.

troublestarter, 2024-02-13:
On the local documentation from webui, i have an error on this page:
http://192.168.0.155/documentation/settings/allskyWebsite.html

[image]
404 error

We should have the public url here because 404 error
[image]

We have no link on this one here:
[image]

EricClaeys, 2024-02-13:
@troublestarter,
Users can currently view the documentation via the "Allsky Documentation" link in the WebUI which is on the Pi, which means it is **local**.  They can see the same documentation via the `Wiki` link in GitHub.  It's the GitHub documentation we are considering getting rid of.

I will take a look at the dead links.

troublestarter, 2024-02-13:
> I will take a look at the dead links.

Sure. I'm waiting for your feedback

troublestarter, 2024-02-13:
**Overlay page**

http://192.168.0.155/documentation/overlays/overlays.html
In yellow : to verify if a link should be present
In red, 404 error
[image]

Here apperas has a link but not link provided
[image]

Here not linked to the external website, 404 error
[image]

Here not working link, site may be down or temporary unavailable, so it should be testted again:
[image]

troublestarter, 2024-02-13:
Moduke page
http://192.168.0.155/documentation/modules/modules.html

- Replace the link with the good direct link (https://github.com/AllskyTeam/allsky-modules)
[image]

-

troublestarter, 2024-02-13:
**Constellation overlay page**
http://192.168.0.155/documentation/explanations/constellationOverlay.html

- Link with 404 error
[image]

troublestarter, 2024-02-13:
**Other page:**
http://192.168.0.155/documentation/troubleshooting/other.html

- 404 link for Reducing wear on your SDcard
[image]

troublestarter, 2024-02-14:
FYI
https://github.com/AllskyTeam/allsky/discussions/3426

troublestarter, 2024-02-14:
Still in issue:

**- Overlay page**

[image]

then

[image]

troublestarter, 2024-03-10:
[https://github.com/AllskyTeam/allsky/discussions/3489](url)

EricClaeys, 2024-03-11:
@troublestarter, The link in your last post doesn't work.  I do this all the time, too.  A valid link in GitHub is
~~~
[click here](https://github.com/AllskyTeam/allsky/discussions/3489)
~~~
unfortunately, clicking on the chain link icon often does what it did for you:
~~~
[https://github.com/AllskyTeam/allsky/discussions/3489](url)
~~~
I think it works if you stand on your left foot and chant something LOL

EricClaeys, 2024-05-12:
I believe all the dead links are fixed in the next release, so I'm closing this Issue.

closed 2024-05-12 by EricClaeys

troublestarter, 2024-05-13:
Perfect