AllskyTeam/allsky #3452: [ENHANCEMENT] Backup/Restore Settings Option

EricClaeys, 2024-03-04:
### Discussed in https://github.com/AllskyTeam/allsky/discussions/3448

<div type='discussions-op-text'>

<sup>Originally posted by **custard2022** March  3, 2024</sup>
Hello! 
Would like to request to Export/Backup and then Import/Restore Allsky Settings option, along with saved local profiles that you could pick from depending on what you're doing with the system.

1. Backup/Export settings like:
    - Allsky Settings page
    - Overlays
    - Editor files
    - Module setup
    - WiFi networks
2. Restore/Import the settings.
3. Have option on what to export/import from the list above.
</div>

Alex-developer, 2024-03-10:
@EricClaeys I started work on this a little while ago. Please give me a shout before doing anything as I have quite a lot of it done.

EricClaeys, 2024-03-11:
Cool - I was assuming it was a post-v2024.xx.xx change so wasn't going to do anything.

EricClaeys, 2024-11-28:
@Alex-developer, should this go in v2024.12.06_01 or v2024.12.06_02 ?

Alex-developer, 2024-11-28:
I would put it in v2024.12.06_01