AllskyTeam/allsky #3495: Overlays vanished at random time and have not come back

Alex-developer, 2024-03-11:
Stop the runtime data being written to the module config file as it can lead to corruption, save it in a separate file instead

### Discussed in https://github.com/AllskyTeam/allsky/discussions/3454

<div type='discussions-op-text'>

<sup>Originally posted by **STLS-1** March  4, 2024</sup>
As in subject, My overlays simply vanished at around 12:35PM yesterday and have not come back. No clue why. I have not touched anything with overlays once I got them working. I rebooted and no change. I removed overlay items to see if there was one that suddenly went bad.. NO change. </div>

rmoravcik, 2024-03-21:
Same thing is happeing to me using V2024.xx.xx branch. For some reason overlay module is getting disabled by something.

[image]

rmoravcik, 2024-03-24:
It seems my issue of disapearing overlay is caused by some stuck of system:

`2024-03-24T02:15:09.077479+01:00 allsky allsky[885]: STARTING EXPOSURE at: 2024-03-24 02:15:09   @ 2.8 sec
2024-03-24T02:15:09.992652+01:00 allsky allsky[25652]: ERROR: Module allsky_overlay.py will be disabled, it took 33.54 seconds; max allowed is 30 seconds
2024-03-24T02:15:16.307289+01:00 allsky allsky[885]:   > Saving NIGHT image 'image-20240324021509.jpg'`

EricClaeys, 2024-03-24:
You can adjust the maximum allowed - see the documentation.  33 seconds is a long time, do you have a lot of text and/or images in your overlay, or have a slow Pi?

Alex-developer, 2024-07-26:
Closing as the watchdog has been removed

closed 2024-07-26 by Alex-developer